dresden-elektronik/deconz-rest-plugin #167: Philips Hue Entertainment

ebaauw, 2017-09-04:
See https://developers.meethue.com/entertainment-blog.

Functionality looks great, but another manufacturer-specific extension to the ZigBee standard.

I suppose it will only work with Hue lights, as it needs support by the lights firmware.  Not sure if we would want to support this in deCONZ.  However, the ability to have the Hue bridge and RaspBee/ConBee on the same ZigBee network, both able to control the lights, becomes more and more attractive.  Would also be a workaround for the Hue tap/ZGP setup (see #117).

I don't suppose the RaspBee will pair with new devices when setup as a router on the Hue bridge network.  Would it be possible to configure the RaspBee as gateway (so I can still connect my ZHA and ZigBee 3 devices), but with the same pan id, channel, and network key as the Hue bridge?  I would pair my Hue (compatible) lights and Hue taps with both the Hue bridge and deCONZ, but all sensors, switches and (obviously) ZHA and ZigBee 3 lights with deCONZ.

posixx, 2017-09-04:
@ebaauw we need ZGP proxy support within deconz to be able to receive the hue tap button presses on the web socket API. We don't use bridge rules preprogrammed, all events are managed externally through websocket events

ebaauw, 2017-09-04:
> we need ZGP proxy support within deconz to be able to receive the hue tap button presses on the web socket API. 

The gateway is already working as ZGP proxy (you receive the buttonevents when the Hue tap is in range of the gateway).  The issue is that you need to tell the Hue lights to act as ZGP proxy as well, for when the tap is outside the direct range of the gateway.  You would only pair the lights and Hue tap with the Hue bridge so it configures the ZGP proxy in the Hue lights.  After that, you could even power-off the Hue bridge: the deCONZ gateway should pickup the re-broadcast from the light and issue a web socket notification.

posixx, 2017-09-05:
As ZGP proxy / green power is a zigbee specification feature this should definitely  be included in the deCONZ software. To have all lights paired to the hue bridge only to enable this is way to much overhead. 

About hue entertainment; apart from the question if you want this behaviour / functionality in my opinion this shouldn't be implemented in deCONZ as it's not part of the zigbee standard. If someone wants to use hue entertainment; use a hue bridge.

mariusmotea, 2017-09-11:
Not exactly. Once the api will be released i will integrate this in my bridge emulator so the same raspberry pi can be used for both things

> If someone wants to use hue entertainment; use a hue bridge.

mariusmotea, 2018-07-30:
Is there any way to send rgb data to lights at fastest rates? Hue Entertainment works with 25 frames/second, currently i try to send only one request per second to Deconz using http api, but after a few seconds Deconz is overloaded and don't update the lights anymore.

manup, 2018-07-31:
Currently there is no implementation of the Hue Entertainment API nor the functions on MAC level. Also on ZigBee level there are no such interfaces beside the ones implemented currently. 

Anyway 1 request per second should not overload deCONZ, can you provide the payload of the request?

mariusmotea, 2018-07-31:
You can use the dtls server for entertainment from my repo write in C by one user.

closed 2018-10-21 by ebaauw