dresden-elektronik/deconz-rest-plugin #168: Xiaomi round push button / switch double click / triple click / long press

rtenklooster, 2017-09-04:
As I've already commented on issue #152 I found new events regarding single / double / triple / quad clicks. Therefore I did some extensive logging, removed all Xiaomi devices from my gateway and added the switch. 

All relevant packets are captured and pasted here. https://pastebin.com/X7wG5LVr

I hava analyzed the packages and wrote all relevant info to this Excel spreadsheet.
https://tenklooster-my.sharepoint.com/personal/richard_tenklooster_net/_layouts/15/guestaccess.aspx?docid=0745e883aa555494cb8885a2d8eb48f3e&authkey=AQ08dGXdAz2r02LBPoyxdMw

rtenklooster, 2017-11-16:
Hi,
As you know I have sniffed a couple of xiaomi sensors.
I would realy like to use those sensors right now.
Of course I understand there are more issues, with a higher priority.
Therefore it would be great if someone would help me to add support for those sensors myself.

Maybe provide me with a small example, what files are how modified. I can't imagine it's a lot of work.
So if someone could tell me what to do to support multiple clicks for the round switch, I will try to add the xiaomi outlet, xiaomi smoke detector and battery percentage for the motion sensors myself.

manup, 2017-11-17:
The one, double and triple click are now working with the latest commit.
However there seems to be no command sent to indicate a long press event by the switch.

manup, 2017-11-17:
> Maybe provide me with a small example, what files are how modified. I can't imagine it's a lot of work.
So if someone could tell me what to do to support multiple clicks for the round switch, I will try to add the xiaomi outlet, xiaomi smoke detector and battery percentage for the motion sensors myself.

That's sadly not that easy since the plugin is not very modular, like for example groovy device files of Smart Things. Some sensors are easy to integrate and only need a few lines of code added to certain places, others are more complicated and need to be supported on various places to work.

So currently the best way to get into how things work, is to look at sensor specific commits, many look similar but as usual devil is in the details :)

rtenklooster, 2017-11-27:
@manup thanks for the explaination. 

> The one, double and triple click are now working with the latest commit.
> However there seems to be no command sent to indicate a long press event by the switch.
I've updated deconz and the rest plugin to .93. The rest api is still only showing event 1000 en 1002. No double / triple press events. I've removed the butten and re-added it, but that doesn't help either.

manup, 2017-11-29:
Hmm strange I'll test it later today, haven't yet used the button since implementation, so many toys so little time :)

manup, 2017-11-29:
Seems that the *.aq2* prefix in the model id stopped the button map to be loaded.

rtenklooster, 2017-11-29:
Ok cool 👍 
Will give it a try when the new build has ben released. Thanks anyway!

rtenklooster, 2017-11-30:
Ok, tested .94 👍 
All working as it is supposed to :) Thanks a lot!

closed 2017-11-30 by rtenklooster

diderikfrom, 2017-12-16:
As commented above, the addition of double/triple/quadruple clicks **broke** the _button pressed_ (1000), now I only have _button released_ (1002) , as well as the multiple clicks (2002, 3002, 4002).
Multiple clicks is great, but I would much rather have _button pressed_ and _released_. That way, the switch works great as a push/pulse dimmer with a bit of scripting.

ebaauw, 2018-01-19:
See #349.