dresden-elektronik/deconz-rest-plugin #169: Invoking GET / reports rule inconsistently

grover, 2017-09-04:
Hi,

I have created the following rule using the REST API:

```
{
  "actions": [
    {
      "address": "/sensors/8/state",
      "body": {
        "flag": true
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/sensors/5/state/dark",
      "operator": "eq",
      "value": true
    },
    {
      "address": "/sensors/6/state/presence",
      "operator": "dx"
    },
    {
      "address": "/sensors/6/state/presence",
      "operator": "eq",
      "value": true
    },
    {
      "address": "/sensors/8/state/flag",
      "operator": "eq",
      "value": false
    }
  ],
  "name": "Enable bedlight on motion after dark",
  "status": "enabled"
}
```

(Some fields omitted as they're unaffected.)

If I query all data from the REST API by calling ebaauw's great ph.sh with:

`ph_get /`

The above rule is reported as:

```
  "rules": {
    "1": {
      "actions": [
        {
          "address": "/sensors/8/state",
          "body": {
            "flag": true
          },
          "method": "PUT"
        }
      ],
      "conditions": [
        {
          "address": "/sensors/5/state/dark",
          "operator": "eq",
          "value": true
        },
        {
          "address": "/sensors/6/state/presence",
          "operator": "dx",
          "value": null
        },
        {
          "address": "/sensors/6/state/presence",
          "operator": "eq",
          "value": true
        },
        {
          "address": "/sensors/8/state/flag",
          "operator": "eq"
        }
      ],
      "created": "2017-09-03T17:00:12",
      "etag": "25a8e778048f5dbe4be34da308aa5c6f",
      "lasttriggered": "none",
      "name": "Enable bedlight on motion after dark",
      "owner": "ACEB2492C5",
      "periodic": 0,
      "status": "enabled",
      "timestriggered": 15
    },
    ...
```    
    
Note the missing value field in the last condition. If I query the rule directly, e.g.

`ph_get /rules/1`

I get the rule as I created it in the first place. So the value seems to be there, but
not reported when calling GET /.

Also this rule is only executed once after restarting deCONZ, which caused me to look into it. I suppose that there's an issue around that too, but I'll file another if I figure that out.

ebaauw, 2017-09-04:
You need to specify the value for the `value` key as string when defining the rule, so `"true"` instead of `true`.  Same for numeric attributes: `"0"` instead of `0`.

grover, 2017-09-04:
Thanks @ebaauw - do I need to stringify them only in conditions or for actions too?

ebaauw, 2017-09-04:
Only in conditions.

grover, 2017-09-04:
Closing here and moving this to #160

closed 2017-09-04 by grover