dresden-elektronik/deconz-rest-plugin #170: Enable the use of the reachable attribute in rules

grover, 2017-09-04:
Hi,

I'd like to make sure that a lightstrip plus is off anytime the raspbee detects it as reachable. To do this I'm trying to create the following rule:

```
{ 
  "name": "Disable bedlight when it becomes reachable.",
  "actions": [ 
    { "address": "/sensors/8/state", "body": { "flag": false }, "method": "PUT" },
    { "address": "/lights/9/state", "body": { "on": false }, "method": "PUT" }
  ], 
  "conditions": [
    { "address": "/lights/9/state/reachable", "operator": "dx" },
    { "address": "/lights/9/state/reachable", "operator": "eq", "value": true } 
  ]
}
```

I know this isn't supported by Philips Hue and I know this could be done differently, but I'd like to script my entire network and rules and this is one of them that would get me consistent results no matter what.

Do you think that would be possible?

Thanks!

ebaauw, 2017-09-04:
deCONZ already issues websocket events for `state.reachable`, so technically it would be possible.  I think the issue is more that conditions on light states aren't yet supported, see #59.

> this is one of them that would get me consistent results no matter what

Whether `state.reachable` would actually reliably reflect whether the light is powered off or on is a different issue altogether.  deCONZ might not detect that the light is powered off for quite some time.   If you power it on again before that, deCONZ should pick up the announcement message, and hopefully issue a `state.reachable` event, even though `state.reachable` was still `true`.  With the `dx` condition, the rule would also fire when `state.reachable` changes from `true` to `true`, which might just be what you want.

grover, 2017-09-04:
> If you power it on again before that, deCONZ should pick up the announcement message, and hopefully issue a state.reachable event, even though state.reachable was still true. With the dx condition, [...]

That's exactly what I want - to configure the lightstrip to sane defaults when it regains power. I've only shown "on", but I also want to control brightness and color.