dresden-elektronik/deconz-rest-plugin #171: raspbee module not detected

mariusmotea, 2017-09-04:
Hi,

Today i received my raspbee module, plugged it in my Raspberry 2 as explained, parse all software installation steps from this README and when i start deconz it is unable to connect to the raspbee module. Is there a way to debug what is the problem here? 
I'm very good with linux and electronics but i don't know how this is working, so i'm helpless. The sd card was flashed with latest raspbian version, i had issues to compile de rest plugin (/usr/include/c++/6/cstdlib:75:25: fatal error: stdlib.h: No such file or directory) and in order to solve it i add ```qmake QMAKE_CFLAGS_ISYSTEM= && make -j2```. On boot the led on module is blinking green 2 times fast then remain red.

Thanks,
Marius.

mariusmotea, 2017-09-05:
I found the issue. Raspbee module is using the serial interface for communications with raspberry pi, and by default serial port is used by system for shell login/messages, and this cause a conflict. I run raspi-config and under serial interface i disable login shell access over serial interface ( leave hardware enabled). Please update README because new users will spend lot of time to solve same problems.

closed 2017-09-05 by mariusmotea

donnib, 2017-09-05:
Eu am folosit imaginea oficială de la Dresden si a funcționat ok dar bănuiesc că ai dreptate dacă folosești o imagine generica.