dresden-elektronik/deconz-rest-plugin #172: Please add support for the pending key in sensors

grover, 2017-09-04:
I've a hue motion sensor that's not reachable - I can update the configuration, e.g. sensitivity, but these can't be applied as the sensor is not reachable. As such I'd expect the sensitivity key to be listed when I query the sensor.

In practice this is not returned and as such I do not know if the configuration has been pushed or not, as the GET request already returns the new configuration.

This is also described by Philips at [General Sensor Resource](https://developers.meethue.com/documentation/supported-sensors#generalSensorResource)

There might be other keys missing too.

ebaauw, 2017-09-05:
Update of sensitivity is not yet implemented.  Same for ledindication and usertest.  There's also something wrong with duration.  See #20.

grover, 2017-09-05:
Ok that explains the erratic behavior I saw with the update duration.

ebaauw, 2017-09-10:
In v2.04.76.

Note that `config.duration` is still broken in that version.  Also note that attribute reporting for `config.ledindication` and `config.usertest` isn't setup, so setting these is only reflected in the resource you changed, but not in the other two resources.  The fixes are already available if you want to compile them yourself.

Krocko, 2017-09-11:
@ebaauw can you tell which value on `config.sensitivity` is most sensitive? `0` or `2`?

ebaauw, 2017-09-11:
2

wvuyk, 2017-09-11:
Using v2.04.76 here, but do not see the sensitivity updated in the REST api?

ebaauw, 2017-09-11:
Do you mean it remains "stuck" in pending?  Yes, I'm seeing the same.

Not exactly sure what's causing this (and why it appears to be only for sensitivity), but I made some improvements in PR #193 and don't see it anymore.

wvuyk, 2017-09-11:
Thanks, will check it with the next beta

wvuyk, 2017-09-17:
@ebaauw I am still not seeing the sensitivity update in the REST API. I see the change coming back with the Websocket, this is working correct, but if I do a REST api call after that, the "on" / "sensitivity" and "ledindication" values are still the same old values.  "pending" is empty.
So it looks like the values are not stored maybe?

ebaauw, 2017-09-17:
@wvuyk that's odd.  Are you on v2.04.77?  Can you check the log of `deCONZ --dbg-info=2`?

For me it works as expected:
```
$ ph_get /sensors/222
{
  "config": {
    "alert": "none",
    "battery": 100,
    "duration": 0,
    "ledindication": false,
    "on": true,
    "pending": [],
    "reachable": true,
    "sensitivity": 2,
    "sensitivitymax": 2,
    "usertest": false
  },
  "ep": 2,
  "etag": "28c02b6ce66d14e38afd2041de654ee5",
  "manufacturername": "Philips",
  "modelid": "SML001",
  "name": "Living Room Motion",
  "state": {
    "lastupdated": "2017-09-17T09:12:36",
    "presence": false
  },
  "swversion": "6.1.0.18912",
  "type": "ZHAPresence",
  "uniqueid": "00:17:88:01:02:xx:xx:xx-02-0406"
}
$ date ; ph_put /sensors/222/config '{"sensitivity":1,"ledindication":true}'
Sun Sep 17 11:16:50 CEST 2017
$ ph_get /sensors/222
{
  "config": {
    "alert": "none",
    "battery": 100,
    "duration": 0,
    "ledindication": true,
    "on": true,
    "pending": [],
    "reachable": true,
    "sensitivity": 1,
    "sensitivitymax": 2,
    "usertest": false
  },
  "ep": 2,
  "etag": "61d133ef2de00bd7282f06664ae4ab73",
  "manufacturername": "Philips",
  "modelid": "SML001",
  "name": "Living Room Motion",
  "state": {
    "lastupdated": "2017-09-17T09:12:36",
    "presence": false
  },
  "swversion": "6.1.0.18912",
  "type": "ZHAPresence",
  "uniqueid": "00:17:88:01:02:xx:xx:xx-02-0406"
}
```
I'm seeing the following events on the web socket:
```
Sun Sep 17 2017 11:16:50: /sensors/222/config: {"alert":"none","battery":100,"duration":0,"ledindication":true,"on":true,"reachable":true,"sensitivity":1,"sensitivitymax":2,"usertest":false}
Sun Sep 17 2017 11:16:53: /sensors/223/config: {"alert":"none","battery":100,"ledindication":true,"on":true,"reachable":true,"tholddark":12000,"tholdoffset":4000,"usertest":false}
Sun Sep 17 2017 11:16:53: /sensors/224/config: {"alert":"none","battery":100,"ledindication":true,"on":true,"reachable":true,"usertest":false}
```
The first notification is for the ZHAPresence resource, the others are for the ZHALightLevel and ZHATemperature resources.  The first is issued directly when processing the PUT; the other two are issued when deCONZ receives the report attributes from the motion sensor.

In the deCONZ log, I see:
```
11:16:50:906 discard sensor config push for config/sensitivity (already pushed)
...
11:16:51:604 handle pending ledindication for 0x0017880102xxxxxx
11:16:51:604 write manufacturer specific attribute of 0x0017880102xxxxxx cluster: 0x0000: 0x0033
11:16:51:604 handle pending sensitivity for 0x0017880102xxxxxx
11:16:51:604 write manufacturer specific attribute of 0x0017880102xxxxxx cluster: 0x0406: 0x0030
...
11:16:53:169 Node data 0x0017880102xxxxxx profileId: 0x0104, clusterId: 0x0000
11:16:53:171 Update Sensor 0x0017880102xxxxxx Basic Cluster
11:16:53:171 update ZCL value 0x0000/0x0032 for 0x0017880102xxxxxx after 161 s
11:16:53:171 update ZCL value 0x0000/0x0033 for 0x0017880102xxxxxx after 161 s
11:16:53:171 Update Sensor 0x0017880102xxxxxx Basic Cluster
11:16:53:172 update ZCL value 0x0000/0x0032 for 0x0017880102xxxxxx after 161 s
11:16:53:172 update ZCL value 0x0000/0x0033 for 0x0017880102xxxxxx after 161 s
11:16:53:172 Update Sensor 0x0017880102xxxxxx Basic Cluster
11:16:53:172 update ZCL value 0x0000/0x0032 for 0x0017880102xxxxxx after 161 s
11:16:53:172 update ZCL value 0x0000/0x0033 for 0x0017880102xxxxxx after 161 s
```
The first line is when I issue the PUT; the next lines are when deCONZ handles the `pending` changes, the last lines are when deCONZ handles the report attributes from the motion sensor.  I'd expect a similar message for 0x0406/0x0030, but it looks like attribute reporting for `sensitivity` still hasn't been setup.

ebaauw, 2017-09-17:
Sniffing the ZigBee traffic, I did find (and fixed) a bug that when sensitivity is written to the motion sensor, the passed value is always 0.  However, that's unrelated to `ledindication`.

Note that `on` is handled by deCONZ (or by the Hue bridge) internally (like `tholddark` and `tholdoffset`).  Also note that `on` doesn't do anything yet, see #90.

wvuyk, 2017-09-17:
I am testing this with my plugin and notice that I am getting a 400 back from deCONZ when I send the command({"config":{"ledindication":false}}). 
I am wondering now if this was working before with the Philips Hue bridge or not. Will have to dive deeper here first.

wvuyk, 2017-09-17:
Yup... it was my bad. Cannot test the Philips Hue bridge staight away, but I am guessing it never worked correctly from here. Sorry for that! Thanks.

ebaauw, 2017-09-17:
> Yup... it was my bad.

Not to worry.

I did find (and fix) another bug while sniffing, which caused setting up attribute reporting for sensitivity to fail.

ebaauw, 2017-10-08:
I think we can close this issue?

closed 2018-11-23 by manup