dresden-elektronik/deconz-rest-plugin #173: Updating tholddark and tholdoffset does not update dark/daylight until next attribute report

grover, 2017-09-05:
Changing the tholddark and tholdoffset parameters of a hue motion sensor does not cause a recalculation of dark/daylight keys immediately with the last stored light level value.

For me this only happens when either the sensor reports a new light level or I explicitly read the light level attribute in deCONZ.

ebaauw, 2017-09-05:
Yes, by design (for now).  I figured you only set the thresholds when calibrating the sensor, so this wouldn't be a big deal.  Does the Hue bridge update dark and daylight immediately?

grover, 2017-09-05:
I don't know if this is implemented in the Hue bridge. I don't have one.

It didn't match my expectation, which is why I wrote the issue.

ebaauw, 2017-09-06:
Solved in PR #175.

Note that `state.lastupdated` does _not_ change when `state.dark` and/or `state.daylight` change due to changing `config.tholddark` or `config.tholdoffset`.  I think this is correct, as the actual state of the sensor hasn't changed.  The web socket state change event for changed `state.dark` or `state.daylight` includes the old value of `state.lastupdated`:
```
Wed Sep 06 2017 20:29:25: /sensors/1/state: {"lastupdated":"2017-09-06T18:29:25","lightlevel":10000}
Wed Sep 06 2017 20:29:25: /sensors/1/state: {"lastupdated":"2017-09-06T18:29:25"}
Wed Sep 06 2017 20:29:38: /sensors/1/config: {"tholddark":5000}
Wed Sep 06 2017 20:29:38: /sensors/1/state: {"dark":false,"lastupdated":"2017-09-06T18:29:25"}
Wed Sep 06 2017 20:29:59: /sensors/1/config: {"tholddark":2000}
Wed Sep 06 2017 20:29:59: /sensors/1/state: {"daylight":true,"lastupdated":"2017-09-06T18:29:25"}
Wed Sep 06 2017 20:30:07: /sensors/1/config: {"tholddark":12000}
Wed Sep 06 2017 20:30:07: /sensors/1/state: {"dark":true,"lastupdated":"2017-09-06T18:29:25"}
Wed Sep 06 2017 20:30:07: /sensors/1/state: {"daylight":false,"lastupdated":"2017-09-06T18:29:25"}
```

This also means that a rule with:
```json
  "conditions": [
    {
      "address": "/sensors/1/state/dark",
      "operator": "eq",
      "value": "true"
    },
    {
      "address": "/sensors/1/state/lastupdated",
      "operator": "dx"
    }
  ]
```
will never execute because of changing `config.tholddark`.

grover, 2017-09-07:
@ebaauw Thanks. Will give it a shot with the next build based on this version.

ebaauw, 2017-10-08:
I think we can close this issue?

closed 2018-09-17 by manup