dresden-elektronik/deconz-rest-plugin #179: Manually set deconz listen port

mariusmotea, 2017-09-07:
Hi,

Currently deconz start on first generic free port. This ensure deconz will start almost in any cases, but cause issues to anybody that want to use deconz api and his application bind on port 80. In my example i need to get the websocket port from api and connect to it. If deconz start first, then will use port 80 and block my application to start. If my application start first then deconz will not be available to return the websocket port. I know this can be fixed with delays and checks, but i believe the proper fix is to give the option to manually specify the binding port (like 90% applications does). The option to disable upnp will be nice, as this enter in conflict with my application.

manup, 2017-09-07:
Currently you can use the `--http-port=<port>` command line option to specify the REST API server port. The Websocket port can't be changed via command line yet.

ebaauw, 2017-09-07:
>  The option to disable upnp will be nice, as this enter in conflict with my application.

Is this the same problem as #83 (about halfway down the conversation)?

mariusmotea, 2017-09-07:
Don't perform a deep analysis, i just saw that my python ssdp function will trow the error "address already in use" whet it try to listen to 1900/udp for M-SEARCH messages and this happen only when deconz is running.

ebaauw, 2017-09-07:
Yes that's it.  @manup, did you, amid all other changes, get a change to investigate further with QT5?

manup, 2017-09-08:
> Yes that's it. @manup, did you, amid all other changes, get a change to investigate further with QT5?

Sadly no, currently I'm focusing on the REST API issues like rules, schedules (localtime) and events system. Next week Friday I'm going to vacations for one week and I hope before that the high priority issues will be fixed.