dresden-elektronik/deconz-rest-plugin #180: Putting multiple switches into one group

perler, 2017-09-07:
After a little timeout at around version 2.0.53 I checked back and I am very happy to see that now everything IKEA is recognized and working. Great job!

So now try to switch to deconz and wonder: is there no way to put multiple switches into one group? Maybe my train of thought is on the wrong track here, but I was under the impression I can put all my lamps into one group, create scenes and then assign different scenes to different switches?

mariusmotea, 2017-09-08:
Today i will release a new version of my Hue emulator that will support raspbee/deconz. Will be possible to add as many switches as you want to control one room. Tradfri motion sensor will be added like a Hue Motion Sensor and will be configured from Hue application with original rules (i like the dimming with 30 seconds before the light is turned off).
Is the Tradfri dimmer switch is supported? I pair one and i don't have any output on websocket in order to create the default rules for it.

Krocko, 2017-09-08:
@mariusmotea That sounds very interesting. Can you explain, how to use it?

mariusmotea, 2017-09-08:
Now i'm working on README that will explain all the steps. I have in my house just Tradfri switches and sensors. My hue emulator will connect to deconz and will get all lights and sensors available, then monitor the websocket for devices events. For IKEA motion sensors will be created automatically an Hue Motion Sensor that can be configured from Official Hue application (i guess more applications can do the same) and will work like the original Hue Motion Sensor. For Tradfri Remote i have create an url where you can set every remote what room to control. The next plan is to allow also to emulate the Hue Dimmer Switch, because i have also rgb bulbs and i want to apply scenes. Because the buttons layout is different, i will put left/ right buttons to turn off the light and middle button to turn on. Up/Down will control the brightness.

Krocko, 2017-09-08:
Cool. Can’t wait. Possibly this is a solution for #142?

manup, 2017-09-08:
@perler 
> So now try to switch to deconz and wonder: is there no way to put multiple switches into one group? Maybe my train of thought is on the wrong track here, but I was under the impression I can put all my lamps into one group, create scenes and then assign different scenes to different switches?

It is possible for some but most switches including Hue dimmer switch and IKEA remote control do have their own group and won't use the same group.

It's only possible to forward commands received by one switch to be forwarded to the group of another switch by using rules in the REST API.

@mariusmotea 
>Is the Tradfri dimmer switch supported? I pair one and i don't have any output on websocket in order to create the default rules for it.

Not yet but it should be available soon, after other current bugs are fixed.

mariusmotea, 2017-09-08:
Not sure, don't have Alexa and i hear that Alexa try to discover Hue Bridge via remote api, that is not public. I believe you can confirm if this will work or not. In case is working fine with Alexa, then is very easy to change the values of any bulb in configuration to look like original Philips (ex: wifi bulbs are presented to applications like original Philips bulbs/strips)

mariusmotea, 2017-09-09:
@Krocko on my emulator you can change manually any value of any bulb (by default cheap rgb wifi bulbs are presented as Philips hue bulbs) and for sure this can be fixed. Problem is to work with Alexa. My emulator is very close to a real Hue Bridge, but i cannot test this as i don't have the Alexa device. I release the new version that works with deconz so you can try it.