dresden-elektronik/deconz-rest-plugin #181: Send unrecognized ZCL reports to websocket

Hector47, 2017-09-08:
I have several zigbee devices that are currently not supported in deCONZ.

I think that it should be nice to have the unrecognized ZCL reports (I don't know if I'm using the right term) forwarded to websocket: Energy Meter, on/off plugs, door sensor,...

I like the report treeview of Ubiqua we can see in [#152](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/152#issuecomment-326828973). If we can have something like that in json it shoud be really nice.

And maybe to be able to read/write attribut from the REST Api.

Thanks

ebaauw, 2017-09-08:
That's a bit of a catch 22: devices need to be supported by the REST API plugin, before it can send web socket events or enable reading or writing of attributes through the REST API.

You can interact with devices that are not (yet) supported by the REST API Plugin in the deCONZ GUI.  By clicking the (almost invisible) circle at the right of the node, a list of endpoints and clusters drops down.  When selecting a cluster, the cluster info panel shows the ZCL commands and attributes for that cluster.  By double-clicking on an attribute, you can write its value.  Note that these clusters and attributes need to be defined in the ZCLDB xml file.

You might want to post screenshots of the clusters and attributes for your unsupported devices, so we can assess whether we could support them.

mariusmotea, 2017-09-09:
I wanted to ask the same. Currently IKEA dimming switch is not supported, but for sure it send some values when the switch wheel is rotated. If raw values are sent, then we can see what values are important for us and write rules for them. Is there a chance to receive the raspbee UART api? i try to connect with a terminal with baud speed 38400, but when i press the button of a switch i saw just strange characters.

closed 2018-09-17 by manup