dresden-elektronik/deconz-rest-plugin #194: Duplicate events from Hue Motion sensor

donnib, 2017-09-13:
Hi,
I am not sure this is known already, i get following all the time from the Hue Motion sensor : 
```
{"e":"changed","id":"3","r":"sensors","state":{"lastupdated":"2017-09-13T14:15:23","presence":false},"t":"event"}
{"e":"changed","id":"3","r":"sensors","state":{"lastupdated":"2017-09-13T14:15:24","presence":false},"t":"event"}
```

wvuyk, 2017-09-14:
I am seeing this regularly too. For some reason the field lastupdated changed without obvious reason

`json {"e":"changed","id":"19","r":"sensors","state":{"lastupdated":"2017-09-14T07:32:59","presence":false},"t":"event"} 
json {"e":"changed","id":"19","r":"sensors","state":{"lastupdated":"2017-09-14T07:33:01","presence":false},"t":"event"} 
`
The sensor is reporting every period of time to show it is alive, that is the repetition we get. But why does lastupdated change? Maybe something changed that is not passed through the websocket?

manup, 2017-09-14:
Exactly, this should be the result of the reports send by the sensor. Lastupdated just tells when the value was updated the last time, hence the last report, which is correct since `lastupdated` doesn't automatically mean `lastchanged`. Maybe we should add an `lastchanged` to provide this information as well?

wvuyk, 2017-09-14:
Manup,
Not necessarily for me, maybe only an indication that the current websocket call is a report?
If I check the log of last night here, I see that sensors have been reporting once per 5 minutes. And last updated is adjusted accordingly. What is weird here is that both donnib and me are seeing this report in 2 seconds after each other. why was this not the "normal" 5 minutes

ebaauw, 2017-09-14:
I'm seeing the same, occasionally.  There's a second notification for the ZHAPresence sensor, 1 or 2 seconds after the regular report, where the sensor reports it's state every 5 minutes.  I suspect lastupdated is somehow updated wrongly, when the attribute for config.duration is read, which is now triggered when receiving a report from the sensor, to ensure the sensor is awake.  I need to anlyse my deCONZ log to be sure - I'm traveling right now, so this will have to wait until the weekend.

I don't think introducing lastchanged is a good idea - it's a characteristic of the attribute, not of the state as a whole.  You'd need four lastchange attributes for a LightLevel sensor: one for lightlevel, one for lux, one for dark, and one for daylight.

wvuyk, 2017-09-14:
I have just checked my log on the same sensor just now. It has been behaving all night, and started the doubles 2 hours ago, still doing the same though every 5 minuts 2 reports in 2 secondes indeed.

I am agreeing on not introducing lastchanged, a simple compare (or skip the attributes when not changed?) will do.

Not really a major issue, so keep on traveling! Hope it is for fun?

donnib, 2017-09-14:
But i still don't understand, if the state of the sensor has not changed why do we even get these events ? Is it because the sensor itself send this ? If so then it's not very clever. It could be fixed by knowing the state in deConz then only sending events if the state has changed from last.

wvuyk, 2017-09-14:
Look at it as a heartbeat. Your program could set its own state on the motion sensor if the report was not receviced in 5 minutes. I think I noticed that deCONZ uses a longer period before setting unreachable?

donnib, 2017-09-14:
I don't see how a heartbeat makes sense if no changes are there, there should be a event if it goes unreachable but not send an event that presence is false.

ebaauw, 2017-09-15:
> I'm seeing the same, occasionally.

It seems completely random.  Sometimes a report results in two web socket notifications, maybe for a couple of times in a row, and then it no longer does.
```
Fri Sep 15 2017 13:09:15: /sensors/312/state: {"lastupdated":"2017-09-15T11:09:15","presence":false}
Fri Sep 15 2017 13:14:15: /sensors/312/state: {"lastupdated":"2017-09-15T11:14:15","presence":false}
Fri Sep 15 2017 13:19:14: /sensors/312/state: {"lastupdated":"2017-09-15T11:19:14","presence":false}
Fri Sep 15 2017 13:24:14: /sensors/312/state: {"lastupdated":"2017-09-15T11:24:14","presence":false}
Fri Sep 15 2017 13:29:13: /sensors/312/state: {"lastupdated":"2017-09-15T11:29:13","presence":false}
Fri Sep 15 2017 13:34:13: /sensors/312/state: {"lastupdated":"2017-09-15T11:34:13","presence":false}
Fri Sep 15 2017 13:39:12: /sensors/312/state: {"lastupdated":"2017-09-15T11:39:12","presence":false}
Fri Sep 15 2017 13:44:12: /sensors/312/state: {"lastupdated":"2017-09-15T11:44:12","presence":false}
Fri Sep 15 2017 13:49:11: /sensors/312/state: {"lastupdated":"2017-09-15T11:49:11","presence":false}
Fri Sep 15 2017 13:54:11: /sensors/312/state: {"lastupdated":"2017-09-15T11:54:11","presence":false}
Fri Sep 15 2017 13:59:10: /sensors/312/state: {"lastupdated":"2017-09-15T11:59:10","presence":false}
Fri Sep 15 2017 13:59:12: /sensors/312/state: {"lastupdated":"2017-09-15T11:59:12","presence":false}
Fri Sep 15 2017 14:04:11: /sensors/312/state: {"lastupdated":"2017-09-15T12:04:11","presence":false}
Fri Sep 15 2017 14:09:10: /sensors/312/state: {"lastupdated":"2017-09-15T12:09:10","presence":false}
Fri Sep 15 2017 14:14:10: /sensors/312/state: {"lastupdated":"2017-09-15T12:14:10","presence":false}
Fri Sep 15 2017 14:14:11: /sensors/312/state: {"lastupdated":"2017-09-15T12:14:11","presence":false}
Fri Sep 15 2017 14:19:11: /sensors/312/state: {"lastupdated":"2017-09-15T12:19:11","presence":false}
Fri Sep 15 2017 14:24:11: /sensors/312/state: {"lastupdated":"2017-09-15T12:24:11","presence":false}
```
Looking at the deCONZ log, deCONZ seems to think it received two reports from the sensor:
```
14:14:10:350 Node data 0x001788010200d33d profileId: 0x0104, clusterId: 0x0406
14:14:10:353 update ZCL value 0x0406/0x0000 for 0x001788010200D33D after 299 s
14:14:10:355 rule event: /sensors/312 state/presence num (0 -> 0)
14:14:10:357 discard sensor state push for state/lastupdated (already pushed)
...
14:14:11:925 Node data 0x001788010200d33d profileId: 0x0104, clusterId: 0x0406
14:14:11:929 update ZCL value 0x0406/0x0000 for 0x001788010200D33D after 1 s
14:14:11:932 rule event: /sensors/312 state/presence num (0 -> 0)
14:14:11:934 discard sensor state push for state/lastupdated (already pushed)
```
As far as I can tell, it's only for the ZHAPresence sensor.

ebaauw, 2017-09-16:
> I don't see how a heartbeat makes sense if no changes are there, there should be a event if it goes unreachable but not send an event that presence is false.

I think I found the solution for this, by introducing a new config parameter, `/config/websocketnotifyall` (apologies for my lack of creativity), see PR #198.  When set to `true` (which is the default), all `state` or `config` attributes are included in the web socket `change` notification (as is the current behaviour); when set to `false`, only the changed attributes are included.  Note that you will still receive events when `state.lastupdated` has been changed, but they no longer include the `"presence": false` when `state.presence` hasn't changed.