dresden-elektronik/deconz-rest-plugin #195: Mesh problems with IKEA devices

donnib, 2017-09-15:
I see problems with IKEA meshing (i pressume that's the problem) together with other devices from other makers for example i see following cases quite often :

1. I can't control my osram lightpole lights even though they are very close to my IKEA lights which they mesh thru. If i click on it then use 0 they usually come back to life.
2. My Hue Dim switch lights up red when i click on the buttons, clicking 0 in deconz doesn't help, it usually comes back by itself after quite some time.

@manup do you have a suggestion how i could troubleshoot this further to find the root cause to this ?

staraxis, 2017-09-21:
I think I have a similar problem but only with my Osram lights. They seem to drop out of the mesh and behave erratically such as turning on less than a minute after being toggled off. This has been happening since updating to version 2.04.77. I have Wemo lights controlled by deconz and they still operate as normal.

If I restart the raspberry pi running deconz with the raspbee and conduct a power cycle of the lights, they will connect to the network but then drop out and not operate properly again after a short period.

donnib, 2017-09-21:
@staraxis yes there seem to be problems with mesh, it's unreliable at the moment, hopefully @manup will nail it down if he can find out what's going on. For me i can't go live until i trust they system is stable enough until i get my wife to use otherwise i'll get into trouble.

Also i have trouble in my 50 nodes setup that the RaspbeeGW sometimes hangs and i need to do a power cycle to get it running again (again @manup mentioned something with a que that fills up for come reason i don't know)

manup, 2017-09-21:
The filled up queues will be improved soon, by checking no more than one request to non responding devices is on air at a time (mainly sleeping end-devices). For the IKEA and OSRAM mesh problems I need more data to analyze, so any sniffer logs from problematic networks are very welcome.

donnib, 2017-09-21:
> so any sniffer logs from problematic networks are very welcome.

If/when i get the sniffer running can i set the sniffer up to sniff all traffic and save it then you can filter through it yourself or ?

staraxis, 2017-09-21:
@donnib No worries thanks for the response. My wife isn't too upset (yet), as it's only 2 lights that are behaving a little crazy so I can wear that for a little while. As to the fix, I'm happy to contribute just not sure what I require for sniffing zigbee packets.

ebaauw, 2017-09-21:
> The filled up queues will be improved soon

Yes, please.  I'm getting quite desperate, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/33#issuecomment-330633123.

I powered off on of my IKEA lights, and the network seems to be stable for over 24h now.  I "lost" one Hue lamp (showing red in deCONZ GUI; API returning error "resource, /lights/xx, not available", when PUTting the state).  Even pressing 0 wouldn't wake it, but cycling power did the trick.  I also "lost" the one remaining IKEA light.  Oddly, if I send a command to the group it's in, the light reacts, and even sends a report attribute command to the gateway, but it remains red.  The web socket sends a change event for the light, but not for the group.  GETting the group show `state.any_on` is still false.

donnib, 2017-09-24:
Yeah, i have all sort of problems :

1. Devices are shown in deconz in yellow but i can't control them, i can see websockets events comes from "them" but they don't react, i can go into deconz and click 0 multiple times then maybe they come alive.
2. My hue dim switch keep dropping out and comming back when it suits itself.
3. As mentioned earlier sometimes the whole system goes to a halt aka "filled up queues" so a reboot or the reset of the raspbeegw is needed.

regarding 1. i don't know if this may be caused that many times sombody by mistakes turns off like 10 ikea lights off where maybe some other devices was meshing thru them and even if i turn them back on the system is still in a weird state or at least some of the lights are in a weird state.

ebaauw, 2017-09-24:
@donnib, did you ever try and connect a large number of IKEA lights to a Hue bridge?  I would sort of expect it to choke as well.

manup, 2017-09-24:
> Yes, please. I'm getting quite desperate, see #33 (comment).

Version 2.04.78 now has prevention to fill up queues, it basically delays APS requests to a node if there are already unconfirmed requests in the queue. This is a very simple approach which might cause unexpected delays. If it turns out to be too restrictive I can raise allowed on the fly requests per node (currently 1).

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.78-qt5.deb

donnib, 2017-09-24:
@ebaauw why would it choke ? IKEA issue or ?

ebaauw, 2017-09-24:
> why would it choke ? IKEA issue or ?

Yes, that's what I'd expect.  If not, it might be interesting to sniff how the Hue bridge deals with discovering and polling the IKEA lights.

> Version 2.04.78 now has prevention to fill up queues, it basically delays APS requests to a node if there are already unconfirmed requests in the queue.

I only have two IKEA lights, and found that my network has been running stable for four days now that I've powered off one.  Feeling lucky, I powered in on again for 2.04.78.

donnib, 2017-09-24:
Imagine having 50 IKEA nodes on ;)

Ok so you are saying this is an issue with IKEA. I guess if sniffing is possible we could sniff the ikea gateway however i can't move to it since all nodes are paired with the raspbeegw. 

I'll give 2.04.78 a try.

donnib, 2017-09-25:
I am running 2.04.78 and at the moment there are no changes for the good for me. 

I still experience 1 and 2 VERY extensive and very often and as i mentioned i have a feeling it's getting worse or is caused by power cycle of IKEA lights. In my head i see something like the mesh network get's ruined then deconz doesn't get them back online or IKEA does something weird idk but it's very annoying.

markbeee, 2017-09-25:
Did you check whether the IKEA lights have the newest firmware installed? I had some problems with IKEA lights meshing and old firmware revisions. Anyway I have only 15 nodes of which IKEA are 8 of them.

manup, 2017-09-25:
Hey donni, we really need some sniffer logs to analyze what's happening here.

donnib, 2017-09-25:
@manup sure but i need to know what you want as i asked above https://github.com/dresden-elektronik/deconz-rest-plugin/issues/195#issuecomment-331092444

donnib, 2017-09-25:
@markbeee yes all (most) IKEA nodes have the newest firmware. I just added 5 nodes more where these probably does not have the newest firmware.

manup, 2017-09-25:
Sorry I missed that one.

> If/when i get the sniffer running can i set the sniffer up to sniff all traffic and save it then you can filter through it yourself or ?

Yes the sniffer can save the logs in a file and we can import and analyze them.

>@markbeee yes all (most) IKEA nodes have the newest firmware. I just added 5 nodes more where these probably does not have the newest firmware.

Would be interesting to test if the problems also happen when the network only contains newest firmware lights.

markbeee, 2017-09-25:
I (luckily) have no latency problems with my 5 IKEA lights, 1 remote and 2 motion sensors - all updated to the newest firmware. But I do have latency problems with my three Philips Hue lights - but this might be due to the polling technique of them. That is a clear argument to buy only IKEA/ Osram (where I have a plug running w/o problems) in the future.

ebaauw, 2017-09-25:
> That is a clear argument to buy only IKEA/ Osram (where I have a plug running w/o problems) in the future.

Despite ZigBee advertising themselves as a standard, all lights are not created equal.  Indeed the OSRAM and IKEA lights support attribute reporting, which is a big plus.  Also, the OSRAM lights allow setting the power-on default.  One of my OSRAM lights even supports measuring power consumption, but it looks like that’s been removed in the later firmware (somehow, this light won’t update, even on the OSRAM gateway).  And the OSRAM, IKEA, and innr lights allow setting the on level and default on/off transition time (which are hard-coded on the Philips lights).  Then again, the Philips lights support a broader range of colour temperatures, and can act as ZGP proxy (needed for the Hue tap).  And they’ve announced the low latancy entertainment option.  I have no experience with non-Philips colour lights, but I’m pretty sure each manufacturer supports different colour gamuts (Philips actually have three different gamuts for different types of lights).  Then, there’s a difference in how many groups and scenes the light can store, whether their firmware can be updated over the air (innr cannot), and whether they publish their firmware so we can use deCONZ for the update. Last, but certainly not least, there’s the price.  I’ve been thinking about setting up a buyer’s guide kind of Wiki to document these differences.

I have one OSRAM plug running without any issued as well, but the other one died on me.  I read that @manup also had a plug dying on him - could be coincidence, but still makes me think...

manup, 2017-09-25:
>  I’ve been thinking about setting up a buyer’s guide kind of Wiki to document these differences.

There is one at reddit, but it seems not to be maintained anymore, maybe this one can be picked up?

https://www.reddit.com/r/Hue/wiki/compatibility_chart

iConnectHue provides a nice overview as well:

http://iconnecthue.com/supported-devices/

> I have one OSRAM plug running without any issued as well, but the other one died on me. I read that @manup also had a plug dying on him - could be coincidence, but still makes me think...

I reckon mine died because I carried it around between work and home daily :)

grover, 2017-09-26:
Two things: A buyers guide would be great with the added benefit of also being able to track RaspBee compatibility.

Before I do more smart light purchases: Are there known issues with the OSRAM GU10 spots? Is this really an IKEA only issue?

donnib, 2017-09-26:
@manup here are two logs, 

[logs.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1334931/logs.zip)

log.dcf is just run without touching any lights or switches
log1.dcf i turned off/on couple of times and everything seemed normal.

I don't know if you can see anything unusual from this. Now that i got the sniffer un and running expect to get more when i see issues. Of course when i had the issues i didn't have the sniffer running.

manup, 2017-09-26:
@donnib thanks I'll look into them, but I'm afraid we more importantly need logs while issues happen, with the info which devices have the problems (by MAC or NWK address).

donnib, 2017-09-26:
@donnib yes of course, i'll provide as soon as i see problems

manup, 2017-09-26:
Cool, thanks in advance, I'm really curious what is causing the issues.

donnib, 2017-09-26:
I can only say the last two days the whole network went on it's knees, the lights were reacting like maybe 30min after i asked them something or maybe never reacted.

donnib, 2017-09-27:
@ I have reverted back to 77 since 78 was causing many problems with delays i mean unbearable like you anticipated.

manup, 2017-09-27:
Version 2.04.79 addresses issues with sleeping end devices causing queues to get stuck with lots of tasks added from the REST API plugin, also discovery of the network was optimized. My network of 85 lights gets discovered in under 3 minutes (no OSRAM and IKEA lights though).

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.79-qt5.deb

The version might help in @holli73 OSRAM network too: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/208

donnib, 2017-09-29:
@manup So for me here is the status. My network is VERY unstable. I am running with 2.04.79. As mentioned before i have 1 OSRAM, 3 Philips (2 hue dim and 1 motion sensor) and 50 IKEA nodes (remotes, bulbs, motion sensors).

Here is a log file made from last night till today. In this session the Osram light should have been off during the night (i have automation that calls rest api) but in the morning it was not. The Osram that should be off is NWK 0x3cb0. Another issue is that i pushed the IKEA remote NWK 0x0601 in the morning like 10 times which should have turned off the lights 0x0197 and 0x1fa4 which didn't happen. Usually problems are resolved with power cycle where some of the pushes i did before power cycle are done so the light may turn off on few times before it settles. 

Right now my network runs a moment then the next moment it may not work e.g extremely slow reactions. I presume all problems at the moment are queue related. Every-time i decide that i run a sniffer when a problem occurs it works so that's why i let the sniffer run so long because it's hard to catch in the sniffer. 

Sorry i can't give you more at the moment but i do hope you can see a pattern in the log file. 

[log2.dcf.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1345349/log2.dcf.zip)

manup, 2017-09-30:
The logs are *very* weird, your lights jump between panid (networks) very often and sometimes change their NWK address. Normally they should stay at one network with the assigned NWK address.

The log shows like 50 networks on the same channel.

I don't know why they do it but as quick guess this looks like an light firmware issue.

For example the light with mac address *0x000B57FFFE3A6BB9* slipped to 5 different networks. By looking at the PanIds you'll notice they look similar and 4 of them end with **5C**, this might be a memory corruption bug in the bulb firmware.

[image]

Is this particular light at latest firmware version? 

Lights which changed network (uncomplete list):

* 0x000B57FFFE3A6BB9
* 0x000B57FFFE446B5A
* 0x000B57FFFE362B58
* 0x000B57FFFE2FC119
* 0x000B57FFFE37DBC8
* 0x000B57FFFE10A7B7
* 0x000B57FFFE37BB67
* 0x000B57FFFE2B9976
* 0x000B57FFFE313C32
* 0x000B57FFFE3A3621
* 0x000B57FFFE37DC4B
* 0x000B57FFFE391456

You may check them for firmware version.

donnib, 2017-09-30:
@manup the bb9 lights is an E27 bulb opal 980lumen model running 1.2.217

manup, 2017-09-30:
That's the latest version, hmm bad, right now it looks like a bug in the firmware or at least I can explain why the lights behaving like this. I have no contacts to IKEA devs but the issue should be forwarded to them.

donnib, 2017-09-30:
@manup i'll try sending it to them see if they will look into it. 

Is any of this related that many times somebody power cycle (power off from the wall) the lights so the jumping is caused by that ? I did power cycle when they don't work so maybe that what's causing this but i have no idea how zigbee exactly works so it's just a wild guess. The NWK address is that something the light somes with and can't change or that is only the MAC and the NWK can change.

manup, 2017-09-30:
> Is any of this related that many times somebody power cycle (power off from the wall) the lights so the jumping is caused by that ?

It shouldn't the panid should stay fixed also after power cycle. The NWK address can change in rare conditions when address conflicts are detected, the address is randomly assigned by each device on initial network join and should then stay fixed also after power cycles.

It looks therefore like a bug, memory corruption due stack overflow or something like that. You can test if the power cycle is related to the problem by sniff the traffic for some hours where you don't do power cycle the lights.

donnib, 2017-10-03:
@manup I have send the information to my contact at IKEA, they will have a look at it. In the meantime i tried your suggestion, all nodes up (56 of them) all yellow in deConz. I let the sniffer running for 1h without controlling the lights in any way or doing any power cycles so the log should be as cleas as possible. Please have a look to see if you see same behaviour. Prior to starting the sniffer i turned lights on which were turned off from power. 

[log_without_power_cycle.dcf.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1351544/log_without_power_cycle.dcf.zip)

manup, 2017-10-04:
I'm also curious, hadn't have the time yet but will look into it later on. Stay tuned :)

manup, 2017-10-04:
This log also shows this behavior as well, albeit there are less networks.

I have seen two routing *hickups* which were caused by RaspBee firmware, not sure if it is related to the greater issue but the next firmware version will prevent it. The gateway needs to send a "I'am here" broadcast periodically so devices know their routes are ok, currently this is done every 2 minutes &mash; if the IKEA lights miss that or the gateway doesn't send the broadcast the lights will try to discover a route to the gateway via expensive broadcasts. And they did ... all at once, which I describe as hickup.

The OSRAM gateway sends this broadcast every 30 seconds, which is quite fast for large networks. I'll lower it from 120 seconds to 60 in the next version. Need to check how often the IKEA gateway does it.

Other than that I just saw a few lights which use an old nwkUpdateId of 0x01 instead of 0x04 (deCONZ network settings), but I don't think it's a problem (not sure though).

donnib, 2017-10-04:
@manup maybe the hickups could be related idk, the contact person i have at IKEA development told me that they always test with big network such as mine and they have not seen this behaviour.

> Other than that I just saw a few lights which use an old nwkUpdateId of 0x01 instead of 0x04 (deCONZ network settings), but I don't think it's a problem (not sure though).

So do i have to do something or have i done something wrong ?

manup, 2017-10-04:
@manup maybe the hickups could be related idk, the contact person i have at IKEA development told me that they always test with big network such as mine and they have not seen this behaviour.

Maybe, I've connected a IKEA light to their gateway now, seems that they also use a 2 minutes interval, But lowering it a bit should be fine for RaspBee.

I also see that the IKEA gateway doesn't query the lights beside initial query, everything works via attribute reporting like OSRAM does it. This is the main difference I see, deCONZ does various queries periodically like reading neighbor tables etc. I think this can be optimized so that after initial network setup queries will slow down and eventually stopped and only lights without reporting will be queried.

> So do i have to do something or have i done something wrong ?

I think the lights will figure it out on their own over time (mostly after power cycle).

donnib, 2017-10-05:
> I think this can be optimized so that after initial network setup queries will slow down and eventually stopped and only lights without reporting will be queried.

Hmm either IKEA is not testing and there is indeed a bug or we use the lights in a different way with deConz i guess. I don't think my network is particular big or small, it's probably what you would expect for a normal house. I would say it's on the small side since i have more to install but before i get this stable there is no point in doing so. I guess my network would go to about 60 nodes if i install the rest of the lights i have.

ebaauw, 2017-10-05:
I doubt whether it’s related to the size of the network.  I only have 40 lights, 70 nodes incl. sensors and switches.  When both my Trådfri bulbs are powered on, the queues eventually fill up, stopping the gateway from issuing any ZigBee commands, even in 2.04.79.  When I power off the white spectrum bulb, the queues don’t fill up.  It smells like superstition, but I’ve been running steady now for 4 days with the bulb powered off.  Typically the hang happens anywhere between a couple of hours to two days after resetting the RaspBee.

I’m thinking maybe deCONZ should whitelist lights (manufacturers?) that support attribute reporting and those that don’t.  Then, only include the lights that don’t report attribute reporting in the polling, and only setup attribute reporting for lights that support it.  After restart, I see a lot of log messages that deCONZ tries to setup attribute reporting on my Hue lights.

Note that the IKEA lights store the reporting configuration in volatile memory; it needs to be setup after each power-cycle.

manup, 2017-10-05:
Yes network size might not be the problem, could be anything. To proceed various problems will are under investigation and differences to IKEA gateway in regard to polling and reporting will be minimized to exclude that as problem source.

The recently added *poll manger* is related to that see: c41a60ed93665762fb0f5682ace3d3fe26042791 and a20b7a0bdfcf1e44327d2c8f6495a18c35d0c646

> I’m thinking maybe deCONZ should whitelist lights (manufacturers?) that support attribute reporting and those that don’t. Then, only include the lights that don’t report attribute reporting in the [...]

The poll manager is an generic approach to that, it won't poll attributes for which reports are received in a timely manner. The scenes and color reporting must be improved though.

Further the following will be optimized to lower requests:
* neighbor tables periodic polling (ZDO Mgmt_Lqi_req)
* binding table polling
* scene and group table polling

Normally polling of these should be no problem even at high frequency, but there seems to be issues in OSRAM and IKEA firmware.

> Note that the IKEA lights store the reporting configuration in volatile memory; it needs to be setup after each power-cycle.

deCONZ should take care of that and reconfigure on power cycle or when reports are not received in a certain time.

ebaauw, 2017-10-06:
> It smells like superstition, but I’ve been running steady now for 4 days with the bulb powered off.

And that jinxed it.  Had a hang last night.

manup, 2017-10-06:
Just came back from IKEA and now extending my network with 4 GU10 and and 1 shiny new color light. I hope to track down IKEA related issues better.

... opening these plastic package drives me crazy.

wvuyk, 2017-10-06:
It did cut my hand badly once... don’t understand why package it this way

donnib, 2017-10-06:
> Just came back from IKEA and now extending my network with 4 GU10 and and 1 shiny new color light. I hope to track down IKEA related issues better.

That sounds great, hope you get more acquainted with them. What kind of shiny new color light did you buy ? I guess you mean a color spectrum light like cold to warm ? IKEA does not have any RGB color lights afaik. I went to ikea yesterday and also ended up buying two things, the wireless dimmer and a E14 bulb of the chandelier type.

>... opening these plastic package drives me crazy.
Yeah i fully agree, i had to open over 50 packages and the best way i ended up doing that was with a scalpel.

manup, 2017-10-06:
> IKEA does not have any RGB color lights afaik.

It's new, haven't connect it to deCONZ yet. It's paired to the remote from the kit and shows, orange, blue and white via the side buttons.

http://www.ikea.com/us/en/catalog/products/20353289/

manup, 2017-10-06:
Update, the color bulb works very well, vivid colors and and smooth transitions. However it only supports  CIE XY color commands, color temperature (but can be derived from xy) and no hue and saturation commands are ignored.

ebaauw, 2017-10-06:
Cool.  What gamut does it support?  Does it do attribute reporting on Current X and Current Y?  What's the model?

How do you derive xy from ct?  That's been a long outstanding question on the Hue developer's forum where Philips support is very quiet.  Others have given some formulas, but I haven't tested these.

mariusmotea, 2017-10-06:
I made wifi esp8266 based lights firmware for my project that accept all xy,ct and hue parameters. All conversion formulas are in code, but from ct to xy directly i don't have, you will need to convert first to rgb

Marius.

manup, 2017-10-06:
> Cool. What gamut does it support?

Sadly it doesn't tell, the primary values of the color cluster return unsupported, but it sure is wider than the E27 hue color bulb since it goes deeper into green and blue.

> Does it do attribute reporting on Current X and Current Y? 

I guess yes, the IKEA gateway uses reporting for almost everything, need to write a bit more code to set it up.

> What's the model?

TRADFRI bulb E27 CWS opal 600lm

And it has ct min/max = 250/454.

> How do you derive xy from ct?

Color temperature is the black body curve in CIE xy.
I don't have a formula though.

[image]

mariusmotea, 2017-10-07:
Convert ct to rgb:
```
void convert_ct() {
  int hectemp = 10000 / ct;
  int r, g, b;
  if (hectemp <= 66) {
    r = 255;
    g = 99.4708025861 * log(hectemp) - 161.1195681661;
    b = hectemp <= 19 ? 0 : (138.5177312231 * log(hectemp - 10) - 305.0447927307);
  } else {
    r = 329.698727446 * pow(hectemp - 60, -0.1332047592);
    g = 288.1221695283 * pow(hectemp - 60, -0.0755148492);
    b = 255;
  }
  r = r > 255 ? 255 : r;
  g = g > 255 ? 255 : g;
  b = b > 255 ? 255 : b;
  rgb[0] = r * (bri / 255.0f); rgb[1] = g * (bri / 255.0f); rgb[2] = b * (bri / 255.0f);
}
```
then from rgb to xy:
```
Get the RGB values from your color object and convert them to be between 0 and 1. So the RGB color (255, 0, 100) becomes (1.0, 0.0, 0.39)
Apply a gamma correction to the RGB values, which makes the color more vivid and more the like the color displayed on the screen of your device. This gamma correction is also applied to the screen of your computer or phone, thus we need this to create a similar color on the light as on screen. This is done by the following formulas:

float red = (red > 0.04045f) ? pow((red + 0.055f) / (1.0f + 0.055f), 2.4f) : (red / 12.92f);
float green = (green > 0.04045f) ? pow((green + 0.055f) / (1.0f + 0.055f), 2.4f) : (green / 12.92f);
float blue = (blue > 0.04045f) ? pow((blue + 0.055f) / (1.0f + 0.055f), 2.4f) : (blue / 12.92f); 
Convert the RGB values to XYZ using the Wide RGB D65 conversion formula The formulas used:

float X = red * 0.664511f + green * 0.154324f + blue * 0.162028f;
float Y = red * 0.283881f + green * 0.668433f + blue * 0.047685f;
float Z = red * 0.000088f + green * 0.072310f + blue * 0.986039f;
Calculate the xy values from the XYZ values

float x = X / (X + Y + Z);
float y = Y / (X + Y + Z);
```
This is from hue api documentation [link](https://developers.meethue.com/documentation/color-conversions-rgb-xy)

ebaauw, 2017-10-07:
Thanks, @mariusmotea.  Where did you get the ct to rbg formula from?  Do you also have the inverse (rgb to ct)?

mariusmotea, 2017-10-07:
Sorry, i don't have the rgb to ct formula and I don't remember from where i get the one from ct to rgb, maybe google can help. I will try to help on this problem to develop an optimized ct to xy formula based on that curve.

ebaauw, 2017-10-07:
Couldn't resist and got myself an IKEA color bulb.  SW Build 1.3.002, Date Code 20170315.  Firmware image is 0x2801, but no file included in the IKEA downloads.

It's a ZLL Color Light, so it doesn't support colour temperature.  It reports Color capabilities (0x400A) as only CIE 1931 XY, so no hue/sat and no colorloop either.  A bit odd, but probably still compliant with ZLL?  Indeed, it reports Color temperature min (0x400B) and Color temperature max (0x400C), but that doesn't make sense.  My Hue color lights don't report these attributes.

The REST API reports the light as follows:
```json
{
  "etag": "99e349df49ba922027e4183bc522e8f5",
  "hascolor": true,
  "manufacturername": "IKEA of Sweden",
  "modelid": "TRADFRI bulb E27 CWS opal 600lm",
  "name": "Light 2",
  "state": {
    "alert": "none",
    "bri": 1,
    "colormode": "xy",
    "effect": "none",
    "hue": 0,
    "on": false,
    "reachable": true,
    "sat": 0,
    "xy": [
      9.19132e-05,
      9.19132e-05
    ]
  },
  "swversion": "1.3.002",
  "type": "Color light",
  "uniqueid": "00:0b:57:ff:fe:xx:xx:xx-01"
}
```
The REST API correctly doesn't expose `state.ct` (probably because it's a Color Light) but it should also suppress `state.hue`, `state.sat`, and `state.effect` (and `config.ctmin` and `config.ctmax`).  Also note the funny formatting of `state.xy`.

The bulb doesn't seem to report the actual Current X and Current Y values like the Hue lights do - it echoes the values last set.  It accepts the full range 1..65535.  When setting Current Y to 0 (which is invalid), the light shows blue-ish and flickers.  When changing brightness it flickers again.  The deCONZ REST API has a bug: it should set Current X and Current Y to 1 when setting `"xy": [0,0]`.  And there's still the issue that deCONZ maximises the values to 65279.  Also it uses this number to convert the ZigBee uint16 value to a JSON float, instead of 65535, see #30, point 6.

manup, 2017-10-07:
> The REST API correctly doesn't expose state.ct (probably because it's a Color Light) but it should also suppress state.hue, state.sat, and state.effect (and config.ctmin and config.ctmax). Also note the funny formatting of state.xy.

The non existing values will be filtered out soon, I'm working on using the `color capabilities` attribute for that. I think the ctmin/ctmax might be ok since the lights does provide these values.

> The deCONZ REST API has a bug: it should set Current X and Current Y to 1 when setting "xy": [0,0].

Why 1?

> And there's still the issue that deCONZ maximises the values to 65279. Also it uses this number to convert the ZigBee uint16 value to a JSON float, instead of 65535, see #30, point 6.

There was a reason for that number, albeit I'd forgot which one, need to look in the spec and properly some ZLL firmware.

Forecast: There is an upcoming commit for the color cluster attribute reporting which will include xy and color mode.

manup, 2017-10-07:
Here we go:
http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.80-qt5.deb

The version is experimental, after startup phase polling of IKEA and OSRAM will be dropped smoothly in favor for reporting. Further neighbor table requests will be slowed down after startup phase.

In contrast to that, lights which don't support attribute reporting will be polled much more frequently. There is still much room to optimize certain requests, which will be addressed in the next versions.

holli73, 2017-10-07:
@manup,

i did compile on x64 and after restart of deconz it did abort during lights logon

```
[913888.052943] deCONZ[850]: segfault at fa200420000 ip 00000fa200420000 sp 00007fff5663c178 error 14
[913920.796795] deCONZ[9900]: segfault at 18 ip 00007fc5025d14e3 sp 00007fffca2a5cd8 error 4 in libdeCONZ.so.1.2.0[7fc5025b6000+4b000]
```
next start is still running and all 61 devices are available.

best regards
holli

manup, 2017-10-07:
Thanks for feedback, might be related to the updated core. I'll release 2.04.80 for x64 with dev-packages later today.

[46 more replies]

closed 2019-03-30 by stale[bot]