dresden-elektronik/deconz-rest-plugin #197: v2.04.77 doesn't run on port 80

ebaauw, 2017-09-15:
v2.04.77 runs the REST API plugin on port 8080 instead of port 80.  This is because deCONZ-autostart.sh no longer calls setcap to allow deCONZ to run on port 80 (unless you use the SD card image, if I read the script correctly).  I worked around this by manually running
```bash
sudo setcap cap_net_bind_service=+ep /usr/bin/deCONZ
```

manup, 2017-09-24:
Version 2.04.78 now contains a debian package **preinst** script which will be executed by `dpkg -i`, this script will do the `setcap` and deCONZ should be allowed to start with port 80. 

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.78-qt5.deb

I want to get rid of most sudo calls in deCONZ-autostart.sh since it's not the proper place to setup stuff.

ebaauw, 2017-09-24:
Indeed, installed 2.04.78 and web server runs on port 80.

ebaauw, 2017-12-15:
In .99 deCONZ-autostart checks for cap_sys_time in addition to cap_net_bind_service (line 205).  If deCONZ has cap_net_bind_service, but not cap_sys_time, it starts on port 8080.

closed 2017-12-20 by ebaauw