dresden-elektronik/deconz-rest-plugin #199: Nanoleaf support?

wvuyk, 2017-09-17:
I was pointed to a nice looking set of lights that is available now. They claim to support Zigbee and WIFI (probably depending on the model). 

> The Nanoleaf Smarter Series includes all of our innovative, connected LED lighting products. Whether you want purely brilliant white or color-changing light, we’ve got a bulb (or a panel) for you. Smarter Series products come in both Zigbee and WiFi versions with support for Apple HomeKit and Amazon Alexa.

I cannot find conformation yet if they really support Zigbee 3, but it might interest you?

https://nanoleaf.me/en/

ebaauw, 2017-09-17:
Their [tech specs](https://nanoleaf.me/en/consumer-led-lighting/for-owners/get-support/knowledge-base/technical-specifications/) state the Smarter series use ZigBee HA 1.2.  They should work with deCONZ.

wvuyk, 2017-09-17:
Thanks, sounds like great news! Could not find that on their site...  Will check if I can get someone to test it.

closed 2017-09-17 by wvuyk