dresden-elektronik/deconz-rest-plugin #200: Conditions based on sunset/sunrise

Krocko, 2017-09-17:
Is it possible, to use conditions based on sunset/sunrise?

ebaauw, 2017-09-17:
Not yet, as deCONZ doesn't yet have a Daylight sensor, see #73.

As workaround, I use a CLIPGenericFlag sensor exposed to HomeKit by homebridge-hue, which I set and clear from HomeKit.  Note that you can only use the iOS builtin Home app to define automations based on sunrise and sunset.

Krocko, 2017-09-17:
With iOS 11 sunset and sunrise is build in the Home App.

ebaauw, 2017-09-17:
> With iOS 11 sunset and sunrise is build in the Home App.

This is already the case with the Home app on iOS 10.

> You can use it in Automations.

For some reason Apple decided to keep this functionality exclusive to the Home app.  It's not included in the HomeKit API, used by other HomeKit apps, even though the Automations run on the AppleTV (or iPad when used as a HomeKit Hub).

Krocko, 2017-09-17:
It is also in the EVE App.

[image]

ebaauw, 2017-09-17:
That's a condition, not a trigger.

Hypfer, 2017-09-18:
You could simply use the websocket to script that. Thats what I'm doing.

Krocko, 2017-09-19:
@Hypfer can you please describe this in more detail?

Hypfer, 2017-09-25:
You can use the REST API of DeConz to trigger a Scene/do basically whatever you want to do. I for example use it to set the color temperature according to the current position of the sun.

https://github.com/mourner/suncalc I'm using that library to calculate sunrise/sunset.

Hypfer, 2017-09-25:
To do that I'm running node-red on that deconz rpi. I've exported the relevant flow for you. You can find it in this gist:
https://gist.github.com/Hypfer/c7fa63a4f6d99e0fd856247adf81fae2

You do however need to change a few things:
URL_OF_DECONZ
APIKEY
Group ID
LAT
LONG

Maybe you'll also need some node-red plugins. I'm not sure since I've made that a few months ago at 3am. It still works fine, though.

Krocko, 2017-09-26:
Thank you. Will give it a try.

closed 2018-09-17 by manup