dresden-elektronik/deconz-rest-plugin #201: Firmware update fails on Ikea Trådfri Bulb E27 opal 1000lm

labanx, 2017-09-17:
Everytime i try to update this bulb it gets stucks and aborts on 0.02%.

What i find strange is that the gui shows it tries to use the firmware 117C-2101-12214572.zigbee and not the ikea firmware downloaded with the python tool.

[image]

> otau $ ls
> 1014-0005-14010400.zigbee
> 1135-0000-2010009B-FLS-PP3_RGBW.zigbee
> 1135-0000-2010009B-FLS-PP3_RGBW.zigbee.md5
> 1135-0000-201000A0-FLS-PP3_RGBW.zigbee
> 1135-0000-201000A0-FLS-PP3_RGBW.zigbee.md5
> 1135-0006-20100021-Ribag.zigbee
> 1135-0006-20100021-Ribag.zigbee.md5
> 117C-0002-165801BC.zigbee
> 117C-11C1-12214572.zigbee
> 117C-11C2-12248572.zigbee
> 117C-11C4-12214572.zigbee
> 117C-2101-12214572.zigbee
> 117C-2201-12217572.zigbee
> 117C-2202-12217572.zigbee
> 117C-2203-12217572.zigbee
> 117C-4201-12217572.zigbee
> 117C-4202-12217572.zigbee
> 14010400_0x0005.zigbee
> 14010400_0x0005.zigbee.md5
> 159693-TRADFRI-bulb-ws-1.2.217.ota.ota.signed
> 159694-TRADFRI-bulb-ws-gu10-1.2.217.ota.ota.signed
> 159695-TRADFRI-bulb-ws-1000lm-1.2.217.ota.ota.signed
> 159696-TRADFRI-bulb-w-1000lm-1.2.214.ota.ota.signed
> 159697-TRADFRI-driver-hp-1.2.217.ota.ota.signed
> 159698-TRADFRI-driver-lp-1.2.217.ota.ota.signed
> 159699-TRADFRI-remote-control-1.2.214.ota.ota.signed
> 159700-TRADFRI-motion-sensor-1.2.214.ota.ota.signed
> 159701-TRADFRI-wireless-dimmer-1.2.248.ota.ota.signed
> 190578-TRADFRI-gateway-1.1.0015.p.elf.sig.ota.signed
> 190579-ncp572b444.ebl.ota.ota.signed
> pi@raspberrypi:~/otau $
> 

Also tried to select firmware manually without any luck.

What am i doing wrong ?

ebaauw, 2017-09-18:
It's the same file - the OTAU plugin copies the downloaded files to vendor-deviceid-version.zigbee.

Are you on the latest beta?  There was a similar issue before v2.04.75, see #166.

Otherwise, try moving the bulb and the RaspBee closer to each other and power off other lamps while attempting the firmware update.

svenove, 2017-11-16:
I'm having the same problem - freezes at 0:58 (0,04%, but that's probably because I try with newer firmware, with a smaller filesize).

I'm running on 2.04.86.

svenove, 2017-11-16:
Solved it by upgrading the ConBee-firmware to 0x261a0500, as described in issue #230.