dresden-elektronik/deconz-rest-plugin #202: Bitcatcher on Windows 10 with ConBee

donnib, 2017-09-19:
Hi @manup ,
I received the ConBee yesterday, thx for that. I did following :

1. Installed deConz for windows `2.04.77`
2. Flashed the ConBee with `BitCatcher_Firmware_V2_07_03.GCF`
3. Installed BitCather from https://digital.luxoft.com/bitcatcher/
4. Launched sniffer.exe -> no device detected

I tried reboot, start sniffer.exe with `Run as administrator` but still it can't find it. Any ideas what to do ?

/donnib

manup, 2017-09-19:
Unfortunately Windows 10 is not supported by the BitCatcher sniffer, you can use it with Windows 7 and Linux (for example in a VM).

https://www.dresden-elektronik.de/funktechnik/solutions/wireless-light-control/gateways/conbee/?L=1

We are working on Wireshark integration which will also support Windows 10.

donnib, 2017-09-19:
Ohh i see then it makes sense it didn't work for me ;) I'll make a VM to try this out thx.

closed 2017-09-19 by donnib