dresden-elektronik/deconz-rest-plugin #203: Websocket does not return brightness in particular case

androme13, 2017-09-19:
Hi, when a bulb is in a philips hue dimmer group and when the hue dimmer changes brightness with buttons, the new value of brightness is not returned via websocket.

ebaauw, 2017-09-19:
Actually changes to brightness (and other light state attributes) are reported through the web socket just fine.  The issue you're facing is that the gateway (deCONZ, or any other ZigBee gateway, like the Hue bridge) needs to update it's cache of the light state, before the web socket event is pushed.  This cache is updated through three mechanisms:
1. The gateway tries to guess what the effect to the light state will be of commands it issues to the light.  Normally this works pretty well, except for longer transitiontimes and relative commands (like bri_inc - which is not yet implemented, or the commands send from the dimmer switch to the group);
2. The gateway polls the light states periodically.  The more lights you have, the longer this takes.  The delayed notification you see is the result of the light being polled;
3. The light might send a push notification to the gateway when its state changes.  In ZigBee speak, this is called attribute reporting.  In all their wisdom, Philips decided not to support this in their lights firmware.

So in order to get the notifications you want, either (3) change light bulbs (both Osram and Ikea support attribute reporting, Philips and innr don't) or (1) don't use the group to change the brightness, but use gateway rules instead (this is what the Hue bridge does).

androme13, 2017-09-19:
Thank you for your reply.

closed 2017-09-19 by androme13