dresden-elektronik/deconz-rest-plugin #204: Xiaomi Door Sensor

mtornblad, 2017-09-19:
I got a Xiaomi door sensor that i managed to join into deCONZ, but it do not show up as an sensor so i can react on the changes from it. I can see in the deCONZ application that the ON/OFF value changes correctly! What is needed to have it visible as an sensor in the API?

[screenshot from 2017-09-19 17-30-14](https://user-images.githubusercontent.com/592869/30600950-ff645bfe-9d60-11e7-937d-04c7f061663d.png)

ebaauw, 2017-09-19:
Open the network and move the magnet (to wake the sensor), and the resource should be created.  Worse case, while the network is open, reset the sensor (pressing the reset button until the light blinks) and keep moving with the magnet (to keep the sensor awake) until the resource has been created.

mtornblad, 2017-09-19:
Thanks,

Open and close it while network was opened seemed to do the trick! I thought that the process of joining it was finished when it was joined,

closed 2017-09-19 by mtornblad

abmantis, 2017-09-22:
I'm having the same issue, but moving the magnet is not solving it. Also tried reseting, pressing the button, etc, without success.
The door sensor shows on deCONZ app, but not on the REST plugin

jonatanolofsson, 2017-11-24:
For me, the node count in deCONZ top right corner goes up one when I connect mine (also one of the rounded, "older" ones), but nothing shows up in the network...

ebaauw, 2017-11-24:
Does it appear in the _Node List_ panel?   I find sometimes new nodes are hidden behind existing nodes in the GUI.

jonatanolofsson, 2017-11-24:
@ebaauw You were spot on, thank you =)

jonatanolofsson, 2017-11-24:
The sensor obviously works now, and changes are seen in the cluster info tab. However, no updates are published on the websocket. Is this a general setting for the websocket I'm missing, or is this an error to this sensor?

ebaauw, 2017-11-24:
> The sensor obviously works now, and changes are seen in the cluster info tab.

That's a good start, the sensor talks to deCONZ.  However, the websocket notifications are pushed by the REST API plugin.  If the sensor isn't yet shown there, you won't see anything on the websocket either.  Make sure the Manufacturer and Model attributes in the Basic cluster are filled (else try a brief press on the reset button and the sensor should report them) and then open the network through the web app (_not_ from the GUI).  You should get a websocket notification when the sensor resource is created.

jonatanolofsson, 2017-11-25:
Indeed, it seems it needs to be first deleted from the GUI, then added in the webapp, then - while the search in the webapp is still active - a short press on the reset to send. The issue seems to be that cluster 0 is only registered in the initial pairing, and cluster 6 is only registered on the short press. I also changed 

```    
//else if (node->nodeDescriptor().manufacturerCode() == VENDOR_JENNIC && modelId.startsWith(QLatin1String("lumi")))
else if (modelId.startsWith(QLatin1String("lumi")))
```
on line 3104 in de_web_plugin.cpp but I'm not sure this actually does anything. If it does, I suppose its a better condition, as it will catch even if the manufacturer is missing (which it is in my case) and "lumi" is still in the model name..

The sensor now shows in the websocket (which is what I needed), but not on the webpage - I'm not sure if it's supposed to do that? 

Thank you for your assistance and your work on this!

manup, 2017-11-25:
> The sensor now shows in the websocket (which is what I needed), but not on the webpage - I'm not sure if it's supposed to do that?

It should be visible in the Phoscon beta app.

Just send you a invitation to the repo if you want to check it out.

[image]

tseel, 2017-12-05:
Are you talking about the MCCGQ01LM or MCCGQ11LM? Tried to pair the older MCCGQ01LM with 2.04.94 just using the rest api without success.

manup, 2017-12-05:
This one:
https://www.gearbest.com/access-control/pp_626703.html

tseel, 2017-12-05:
Ok, that's the newer Aqara sensor. 
Should I file a new issues for the old one or do you reopen this one? How can I support the bugfix, try pairing and send debug log?

jonatanolofsson, 2017-12-05:
@tseel : Mine is of the older type. The steps that worked for me:
 - Delete the sensor from the gui
 - Open the network through the webapp.
 - Initial join (press reset button on sensor for ~4s until it starts blinking)
 - Short presses on the reset button to trigger the cluster message
 - (possibly keep the sensor awake by triggering with magnet)

It sometimes takes a few tries before it works.

abmantis, 2017-12-05:
@tseel open the Zigbee network and simply reset the sensor multiple times. I've managed to pair the old model correctly that way.

tseel, 2017-12-05:
Thank you for the help. After a lot of tries the sensor got paired.
Why is the pairing so unreliable? Can't the pairing get improved from the deconz side to work the 1st or 2nd time? As, at least for the Xiaomi devices and the Tradfri Remote, the pairing was never painless, it's also hard to tell if you did something wrong, the device is broken or it's a deconz issue, assuming you at least know the device should be supported.

tomfritz1, 2018-02-27:
The Xiaomi door sensor is perfectly visible as sensor in Phoscon. But is there any “action” I can do with it?

manup, 2018-02-27:
Not yet but soon, it is currently in active development and will be visible in the sensor control settings which can be attached to groups.

sveip, 2018-05-07:
I am only able to add one sensor per type of the Aqara ones. Only the first motion sensor and door sensor got paired, the rest I tried won't nomatter what I try. Any ideas @manup?

GoSpursGoNL, 2018-05-08:
I followed the above suggestions many times, but am unable to add my two Xiaomi (v2) sensors via the Phoscon webapp. I am using Raspbian. I have also tried it in Windows, also no luck there. However, in Windows via the deConz app, both sensors appear immediately. So the problem seems to be with the webinterfaces, not with the Conbee nor the sensors.

manup, 2018-05-09:
I'll do some tests, with multiple sensors of same type. Not sure what goes wrong since there are already many installations out there wich are running multiple sensors.

sveip, 2018-05-09:
I did some more tests, and it seems its some issue with opening the
network, and keeping it open. If I open it from the legacy web app, it
closes by itself. However, if I use Phoscon, it stays open.

Also I think there is a bug, if you set open time in the settings to 60
minutes, its set to 60 seconds (1 minute.)

Peter

GoSpursGoNL, 2018-05-14:
In the meantime I received 2 additional sensors, and see the same behavior.

manup, 2018-05-14:
> I did some more tests, and it seems its some issue with opening the
network, and keeping it open. If I open it from the legacy web app, it
closes by itself. However, if I use Phoscon, it stays open.

The Phoscon App also closes the permit join when sensor search is stopped.

> Also I think there is a bug, if you set open time in the settings to 60
minutes, its set to 60 seconds (1 minute.)

There should be a timer running which keeps the total open duration to the configured value.

I have now tested to integrate two Xiaomi Aqara motion sensors via Phoscon App. Both joined normally and are listed and update their values.

I've read in another forum that Xiaomi sensors don't like ZigBee channel 25, I haven't checked this out (my network is on channel 15), but out of interest what is your ZigBee channel? The channel is shown in *Phoscon App > Gateway Settings*.

ebaauw, 2018-05-14:
My production network is on channel 25 with plenty of Xiaomi Aqara sensors.

sveip, 2018-05-15:
I am on channel 15 as well, but if I only use Phoscon I'm able to add the
devices. About the timer, I think it's a magnitude error somewhere, try
setting "open network duration" to 60 minutes from the web-app (legacy or
Phoscon) and you'll see what I mean.

Kind regards,
Peter