dresden-elektronik/deconz-rest-plugin #205: config.battery not sent by websocket (2.04.77)

markbeee, 2017-09-20:
Since using 2.04.77 I get no battery status by websockets any more from my IKEA motion sensors (firmware: 1.2.214).
I also get a lot more event notifications when the sensor detetcts motion now. Before there was only one event for detecting motion.

manup, 2017-09-20:
Normally the battery status should be updated after some time, after starting deCONZ it may take a while. If it isn't updated even after an hour the binding seems to be inactive. I'll check if this is handled in deCONZ, so the binding will be recreated if necessary.
 
The increased event count are due the changes that all incoming commands are forwared, not only if attributes (here presence) change. The IKEA sensor sends multiple presence packets which cause these events.

@ebaauw has created a new config flag, so that the behaviour can be changed, see PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/198

This will be available in 2.04.78

ebaauw, 2017-09-20:
> Since using 2.04.77 I get no battery status by websockets any more from my IKEA motion sensors (firmware: 1.2.214).

Indeed, checking my logs, I see plenty of battery notifications from the IKEA motion sensor to the gateway, but no corresponding web socket notifications.  This is unrelated to the IKEA motion sensor firmware (or even the IKEA motion sensor - I see the same for the IKEA remote, the Hue motion sensor, and the Hue dimmer switch).  It was broken (unintentionally, I think) in v2.04.76, when combining all config or state attributes into one web socket notification.  Now, it only sends a web socket notification when the value has changed (as before v2.04.65).  I'll try and fix this coming weekend.

> I also get a lot more event notifications when the sensor detetcts motion now. Before there was only one event for detecting motion.

You get a notification on the web socket when the IKEA motion sensor sends a ZigBee message that it has detected motion.  With the new firmware, it seems to do this more often, every 2-3 seconds while there is motion.  
Note that the IKEA motion sensor does _not_ send a ZigBee message when it no longer detects motion (unlike the Hue motion sensor).  deCONZ automatically sets `state.presence` to `false` (without updating `state.lastupdated`), 60 seconds (`config.duration`) after motion was last reported (regardless of the setting of the timer dial).  Also, deCONZ resets `config.duration` to 60 seconds shortly after you change it.  I mean to fix this when I find the time (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/23#issuecomment-325131934).

markbeee, 2017-09-21:
Thanks for the answers & contributions to this awesome project!
Before I got a config.battery event about every 47 minutes and I also did some measurements (under load) when the value changed:

| Fresh | no load | 3.32V |
|-------|---------|-------|
| 100%  | load    | 3.27V |
| 87%   | load    | 3.14V |
| 74%   | load    | 3.05V |
| 60%   | load    | 2.96V |

When dropping to/ below 47% the PIR got unstable (flase alarms) as it is not unusual for this part having low voltages.

An update of an IKEA motion sensor took ~105 minutes and dropped the battery by about 30%.

manup, 2018-09-17:
Should work now. Closing the oldest issues for know to tidy up the tracker and duplicates in newer issues.

closed 2018-09-17 by manup