dresden-elektronik/deconz-rest-plugin #209: Ecozy radiator valve

twansevriens, 2017-09-27:
Does anyone have any experience with the zigbee ecozy (ecozy.de) thermostat?

mshli, 2018-02-25:
I have the same issue. I am using deconz with a raspbee module and i want to connect some eCozy valves. Does anybody have a solution?

ebaauw, 2018-02-25:
From their manual:
> eCozy is compatible with ZigBee home automation profile 1.2, and can therefore, operate with smart home platforms like Qivicon or Almond and many others (please contact  support@ecozy.de  for more information).

You could mail them and ask for technical documentation (which endpoints/clusters does the thermostat support?).

mshli, 2018-02-28:
I send the email to ecozy, but i‘m still waiting for an answer. 
I found this (https://github.com/ckpt-martin/SmartThings) here on github. It‘s a plugin to integrate ecozy thermostate valves into an Samsung smartThings Hub. Maybe it helps for developing.

mshli, 2018-03-01:
I got something from eCozy. I hope, it's the right thing. With the implementation i definitly need your help.

[eCozy_thermostat_List_of_clusters_and_attributes.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1772919/eCozy_thermostat_List_of_clusters_and_attributes.pdf)

mshli, 2018-03-04:
How do i add a new device?

rygel, 2018-03-07:
I would also be interested in support for these Thermostatic Radiator Valves. They are currently available in Germany for little money.

ebaauw, 2018-03-10:
That will be a huge change, impacting at least de_web_plugin.cpp, sensors.cpp, rest_sensors.cpp, and database.cpp.  Probably also bindings.cpp and various .h files.  There’s currently no support in the REST API for the 0x0200 and 0x0204 clusters.  As it’s battery powered, you probably need to use `config.pending` or something similar to collect changes until the device is awake.

Not even sure if we’d want to expose the radiator valve as a `/sensors` or a `/lights` resource.  See also the discussion on Keen vents, #269 and the Heiman Siren, #404.

mshli, 2018-03-12:
OK. Thanks for the information. What do you think how long does it takes to implement de ecozy's? Do you need weeks, months or years?

mshli, 2018-06-25:
Did you find a solution for integrating the radiator valves? Any advances?

Jasonthefirst, 2018-09-29:
I would be interested in this too. 
I am not so good in integrating it myself.

ethologus, 2018-12-16:
Is there a way to use ecozy with deconz?

closed 2019-04-22 by stale[bot]

simonbove, 2020-01-07:
Anything new about this or other radiator valve?