dresden-elektronik/deconz-rest-plugin #210: Osram Lightify Bulb turns on randomly

wlma, 2017-09-27:
Hi everybody,

I'm facing the problem that my Lightify bulb turns on randomly, and i've no clue why. 
My Setup: Rasperry PI->ConBee->Osram Bulb (Classic A60 RGBW)
deConz is at 2.04.78 (this happened also with 2.04.77); Firmware Version 0x26190500

with "--dbg-info=1", this happens when the log says DeviceAnnce of .... 

> 19:29:24:576 send permit join, duration: 0
19:31:49:269 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x8418260000C9DA21
19:31:49:270 skip configure report for cluster: 0x0008 attr: 0x0000 of node 0x8418260000C9DA21
19:31:49:848 DeviceAnnce of LightNode: 0x8418260000c9da21 Permit Join: 0
19:31:53:539 verified group capacity: 7 and group count: 1 of LightNode 0x8418260000c9da21
19:31:53:543 0x8418260000c9da21 found group 0x0001
19:31:53:889 0x8418260000C9DA21 get scene membership response capacity 16, groupId 0x0001, count 0
19:32:05:505 send permit join, duration: 0
19:32:59:132 dev /dev/ttyAMA0 (/dev/ttyAMA0)
19:32:59:193 GW firmware version: 0x26190500

Did somebody face the same problem?

manup, 2017-09-28:
Device announce indicates that the light just startet up (or rebooted). I've seen the same for a OSRAM A60, since it was quite annoying at night I just pulled it out :) 

Haven't looked deeper into the issue but will reattach it to the network and sniff a bit, maybe the bulp expects some "heartbeat" from the gateway and will restart if it isn't received for some time (need to sniff the traffic with the OSRAM gateway for that).

By the way which version does your A60 show in the basic cluster?

staraxis, 2017-09-28:
This is the exact same issue I am having with my 2 OSRAM A60 bulbs. I'm having to use the physical switch for them. I was on 2.04.64 for a while and everything was fine. I updated to 2.04.77 and I am having this issue. I could try updating to the 2.04.79 beta and see if that works. I'm not sure how to get the version from the basic cluster.

manup, 2017-09-28:
>. I'm not sure how to get the version from the basic cluster.

The basic cluster is available in the *cluster info* panel.

[image]

wlma, 2017-09-28:
Thanks for your replies. here's my version info:
[image]

If I can provide additional Info, feel free to ask, but be warned: I'm a Zigbee Noob ;-)

holli73, 2017-09-28:
hello,

i have the same issue - one bulb keep's going on A60-RGBW - 2a/v1.03.07.
in my network - i have 3 of this bulb's - all report the same version - but only one keep's going on.

is there any firmware update possible through deconz for osram devices or must i buy a osram gw and re-attach all 60 devices for such a need?

thanks
holli

manup, 2017-09-28:
Mine has also the version V1.03.07, Datecode 20140331

Further in OTAU cluster it reports *Current File Version: 0x01020412*
We might compare this as well, I've seen OSRAM bulbs with same basic cluster version but different OTAU file version the past :)

[image]

I'll check if the OSRAM home gateway provides a update for the bulb now..

holli73, 2017-09-28:
hello,

my one reports 0x01020314 and all 3 report the same version.

best regards
holli

manup, 2017-09-28:
Took 20 minutes to login in the Lightify App ... 

It reports
* version 01020412 installed
* version 01020510 available

I'll sniff around a bit before updating

staraxis, 2017-09-29:
@wlma I have the same Build ID and Data Code as you.

wlma, 2017-09-29:
For completeness: my file version is also 0x01020307 (like @holli73  ones). 
So looks like we should update the bulbs first to 0120510 or 01020412?

manup, 2017-09-30:
Having the light with new firmware version back on the deCONZ network, it seems that OSRAM removed touchlink reset functionality, I've had to use the reset sequence. The sniffer is active to observe if the light will turn on on it's own again.

manup, 2017-09-30:
After roughly 3 hours the light did a reboot on it's own, so there seems to be a bug in the firmware.
deCONZ did the usual periodic queries, not sure if that is related, but it shouldn't cause the bulb to reboot.

etique57, 2017-10-02:
I have the same with one of my Osram bulb, on the Osram gateway.

So I guess it's purely a glitch in the bulb itself.

staraxis, 2017-10-03:
I dusted off my smartthings hub and connected my two OSRAM lights to that. 2 days in and I have had no issues at all. No flickering, disconnecting or turning on randomly. It has been smooth so far with the smartthings hub but I would really like to move back to deCONZ. If there is any other info I can provide let me know.

manup, 2017-10-06:
**Update** Found a bug in OSRAM firmware: The lights resets when searching for OTA server and receiving two responses instead of just (the expected) one. The upcoming deCONZ version 2.04.80 takes care to only send one response to not let the bug trigger.

Not sure if the above problem is solved with this, but very likely: time will tell.

kgorszczyk, 2017-10-10:
It seems, that even with Version 2.04.80 the above problem still exists. I have two A60 lights and they still reset randomly.

MasterRay2, 2017-10-12:
Still exists also with Version 2.04.81 (A60, V1.03.07).

manup, 2017-10-12:
>Still exists also with Version 2.04.81 (A60, V1.03.07).

Mine hasn't turn on in the last two days, did before 2.04.81 up to 3 times per day.
However version is newer (A60, V1.05.10).

kgorszczyk, 2017-10-12:
> Still exists also with Version 2.04.81 (A60, V1.03.07). 

I can confirm this. Updated to Version 2.04.81 this morning and my A60 light just turned on again.

> Mine hasn't turn on in the last two days, did before 2.04.81 up to 3 times per day. However version is newer (A60, V1.05.10).

Is there a way to update to Version 1.05.10 without an Osram Lightify Gateway?

ebaauw, 2017-10-12:
I just realised I have the same issue, v2.04.81.  My Classic B40 TW - LIGHTIFY with the V1.03.20 firmware has been turning on randomly, since I enabled the OTAU server to get the firmware Image and Version (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/212#issuecomment-335212537).

The other B40 with V1.04.12 hasn't turned on randomly, so I assume the newer firmware has fixed the bug.  I still don't understand why the Lightify gateway hasn't updated the firmware on the other bulb.  @manup did you have any luck extracting the firmware?

manup, 2017-10-12:
> Is there a way to update to Version 1.05.10 without an Osram Lightify Gateway?

> The other B40 with V1.04.12 hasn't turned on randomly, so I assume the newer firmware has fixed the bug. I still don't understand why the Lightify gateway hasn't updated the firmware on the other bulb. @manup did you have any luck extracting the firmware?

Yes I have the A60, will do the B40 TW and other models on weekend. Since I'm not so keen to publish the raw files for legal reasons, I'll figure something out to provide them encrypted. Before updating please be aware that this might void warranty if the lights take damage so it's on own risk.

> I still don't understand why the Lightify gateway hasn't updated the firmware on the other bulb. 

Maybe it's an bug or perhaps the gateway did not update on purpose, since it might know that the update may fail for certain devices, like a faulty firmware which has a bug in OTA service itself (not OSRAM but I saw cases like this).

lcorsini, 2017-10-15:
@manup do you happen to have also a smart plug?
I have one, always used fine with hue, which started to get labeled unreachable, despite the fact that I'm still able to turn it on and off from hue app, but only ON from a xiaomi button thru home-assistant.
I tought that it could be a bug, but not having the lightify gateway I'm not able to update it

manup, 2017-10-15:
Yes I have the Smart+ plug from OSRAM, which is identical to the former Lightify labeled ones. It reports SW Build Id V1.04.12 in basic cluster and file version 0x01020412 in OTAU cluster.

lcorsini, 2017-10-15:
I've checked and Hue app reports the same firmware version, so I don't really know what could mean that the plug is marked as unreachable, maybe I'll try do a factory reset and add it again, thank you

ebaauw, 2017-10-15:
> Yes I have the Smart+ plug from OSRAM, which is identical to the former Lightify labeled ones. It reports SW Build Id V1.04.12 in basic cluster and file version 0x01020412 in OTAU cluster.

The Smart+ plug is _not_ identical to the old Lightify plug.  They have different firmware images, 0x000A for the Lightify vs 0x0027 for the Smart+ plug).

lcorsini, 2017-10-15:
@ebaauw I haven't tried to make in join the raspbee (got it working only 2 days ago) so I don't know which image is, mine is the lightify one, and came with the build id said by @manup

manup, 2017-10-16:
>The Smart+ plug is not identical to the old Lightify plug. They have different firmware images, 0x000A for the Lightify vs 0x0027 for the Smart+ plug.

Interesting haven't seen that, but as from the functionality they seem to be identical.

mysteryan, 2017-10-16:
I do have an OSRAM gateway.  From a legal standpoint, couldnt I just provide the links from OSRAM of the firmware that my gateway receives for the bulbs since I'm not actually redistributing the files.  Deconz doesn't seem to be able to accept them or I can't figure it out though.  Could be because they're encrypted.  With that in mind, here are some from OSRAM provides:
LIGHTIFY A19 ON/OFF/DIM,01020492,http://s3-eu-west-1.amazonaws.com/lightify-osram-firmware/firmware/release/Cloud/NAFTA/NAFTA_A19_OOD_0x01020492_encrypt.bin
LIGHTIFY A19 RGBW,01020492,http://s3-eu-west-1.amazonaws.com/lightify-osram-firmware/firmware/release/Cloud/NAFTA/NAFTA_A19Rgbw_0x01020492_encrypt.bin
LIGHTIFY A19 Tunable White,01020492,http://s3-eu-west-1.amazonaws.com/lightify-osram-firmware/firmware/release/Cloud/NAFTA/NAFTA_A19_TW_0x01020492_encrypt.bin

oleg77s, 2017-10-16:
This is the exact same issue I am having with my two (ZLL EU) OSRAM A60 RGBW lights with different sw (V1.04.12 and V1.05.10), but there are no issues with other Lightify lights: A60 W, A60 TW and B40 TW

ebaauw, 2017-10-16:
> I do have an OSRAM gateway.

Is that the Lightify Home gateway?

> With that in mind, here are some from OSRAM provides

Where did you find these?  Did you sniff the network traffic or did I miss some setting in the Lightify app?

manup, 2017-10-16:
>I do have an OSRAM gateway. From a legal standpoint, couldnt I just provide the links from OSRAM of the firmware that my gateway receives for the bulbs since I'm not actually redistributing the files. 

Thats what we do with IKEA downloads, when files are available public it should be safe to use them.

>Deconz doesn't seem to be able to accept them or I can't figure it out though. Could be because they're encrypted. With that in mind, here are some from OSRAM provides:
LIGHTIFY A19 ON/OFF/DIM,01020492,http://s3-eu-west-1.amazonaws.com/lightify-osram-firmware/firmware/release/Cloud/NAFTA/NAFTA_A19_OOD_0x01020492_encrypt.bin
LIGHTIFY A19 RGBW,01020492,http://s3-eu-west-1.amazonaws.com/lightify-osram-firmware/firmware/release/Cloud/NAFTA/NAFTA_A19Rgbw_0x01020492_encrypt.bin
LIGHTIFY A19 Tunable White,01020492,http://s3-eu-west-1.amazonaws.com/lightify-osram-firmware/firmware/release/Cloud/NAFTA/NAFTA_A19_TW_0x01020492_encrypt.bin

Interesting, but sadly as the name suggests and by looking in the content, the files are encrypted so they aren't usable without knowing the encryption key and method.

mysteryan, 2017-10-16:
> Is that the Lightify Home gateway?
It's labelled as Lightify Home v2 Gateway in thrir database.
> Where did you find these? Did you sniff the network traffic or did I miss some setting in the Lightify app?
Found these in the Lightify app on iOS.  It caches the data for the gateway in a sqlite database called "ArrayentSDKDB.sqlite3".  Locations for resources for devices I don't own as well as gateway are also given in the database.  The database is located within the application's data sandbox folder in the "Documents".  Not sure if it contains encrytion key since I wouldnt know how to decrypt it anyways as well, but keys for the cloud service they use are also available.

donnib, 2017-10-25:
I think i have a Osram with such a problem but it's another model :

[image]

I don't have a Osram gateway, anyone know if there is a new firmware for it ?

dkmh, 2017-11-25:
Ran into the same problem with OSRAM A60 RGBW. Is this fixed in a newer bulb firmware or is it a HW problem? Is there any solution without an osram gateway?

ulf4711, 2017-11-29:
I‘ve just updated all bulbs to the latest OSRAM Firmware 1.05.10 via Lightify Gateway. But I’ve still this issue with one bulb. I‘m using Raspbee GW version  2.04.93. Before I used the latest official Raspbee GW version 2.04.35 without this issue. I think I will do a rollback to this version.

ebaauw, 2017-12-20:
Got a new version of the OSRAM Lightify app through the app store.  Decided to dust off my Lightify Gateway and plug it in.  Indeed, new firmware versions for the gateway, the E14 _Classic B40 TW - LIGHTIFY_ (V1.05.10), and the (new) _Plug 01_ (V1.05.09).  No new firmware for the old _Plug - Lightify_ (still on V1.03.21).

My E14 stuck on V1.03.20 was also upgraded to V1.05.10.  Let's see if the random turning on stops...

swttt, 2018-01-05:
Got the same issue.

For now i have two Classic A60 RGBWs paired, but only one of them is sporadic turning on.
The thing is, it turns on pretty quickly after it's turned off. (within 20 min or so)
Both with the same (old) firmware  V1.03.07

theendisnye, 2018-01-15:
I have had three OSRAM A60s installed on a Hue Hub for around a year running the original  v1.03.07 firmware. Once a month one light would randomly come on usually at night. 

I moved these lights over to Conbee/ DeCONZ and they have all repeatedly randomly come on, multiple times a day,  all used the same firmware v1.03.07. After a week of seeing this I upgraded the firmware (v1.05.10) and still see continual random switch ons. Switching OTAU on/off also makes no change.

Steve

squirrel289, 2018-03-23:
> Yes I have the A60, will do the B40 TW and other models on weekend. Since I'm not so keen to publish the raw files for legal reasons, I'll figure something out to provide them encrypted. Before updating please be aware that this might void warranty if the lights take damage so it's on own risk.

@manup, Any chance you could point me to where I can "find" the firmware extractor plugin?  Or even just the firmware for the RT 5/6 RGBW?  I have some that are way old and not working as repeaters.  I've heard the latest FW fixes that.

squirrel289, 2019-01-04:
@manup, any luck/update?  I'm still on the same, pathetically old firmware and would really appreciate any official or unofficial help you can offer.

manup, 2019-01-04:
My Osram E27 bulb is running since over week now and didn't turn on by it's own (Note that latest firmware of the bulb didn't fix the issue).

2.05.54 contains several fixes to address this very issue, you may give it a try:

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_54

rmorriss, 2019-02-15:
Hi @manup Hoping you can shed some light I would really appreciate it..  WAF is extremely low at the moment 👎 

Fighting this issue forever now.. Total of; yes I know it's more than I should have, 67 OSRAM lightify RT 5/6 US (firmware, latest in US, 01020492).  These lights work flawlessly 99.5% of the time.  Now running 2.05.54, thinking of going to 2.05.57 just incase.  Except for one really nasty issue.

Out of the lights there are several groups.  Not ALL groups, but ONE light in a group ALWAYS turns on by itself.  It's got something to do with proximity.  I think it's a repeater of some sort.  It's a definite pattern, but not always the same light.   The pattern: Always a light (sometimes two) which is extremely close, or extremely far away from the DECONZ.  The DECONZ is rock solid.  Not one issue.  However, this "auto turn on" problem is a daily regular.  Always at 11PM for example in my bedroom.  3AM in the child's room.  I have tried everything.  ALL lights run the same firmware.  Oddly, one day about a year ago, the problem simply vanished.  It was after a power outage.  Next power outage wasn't for 2 months.  the problem was not there.  Nothing changed.  No firmware, settings.. nothing ... Power outage again, problem returns.  Sometimes, after an outage, it's a different light that falls under this spell.  Sometimes, one the "really annoying ones" decides it's a buddy on the other side of the house is now annoying enough..and begins to behave.  

The latest attempt was to simply uncheck OTA and reboot.  Didn't seem to have an effect. Others report newer firmware, but not in the US.  I dusted off the Osram Gateway and it's up to date, 01020492... Except for ONE light.  Updating that one now incase he is the cause.  Any help you can provide would be greatly appreciated, sort of on the fence about just taking them all out, but this is very time consuming and expensive.  Ideas??

holli73, 2019-02-15:
@rmorriss ,

i do have the same situation sometimes after a power outage - on only a few of them (i have more than 70 osram devices on my network).

i figured out that sometimes after a power outage this devices lose their group information!
so within deconz these devices are still part of the group but the light itself is not anymore or even sometimes part of a different group.

i'm not sure if this is a bug within the osram firmware or deconz but if i remember correctly the group information is stored on the device.

this is easy to resolve - just reset the device - remove it from the group and add it again.
in 100% the cases this always solves the issue - sometimes it is enough it the light shows up in deconz as active - remove it from the group and re-add it - but sometimes the device is grey so this is not possible.

hope this helps.

holli

rmorriss, 2019-02-26:
Hi @holli73 !  Thanks for the advice.  I was on vacation for a week just after I sent this message.  Updating the software has seemed to make things much worse...  So, I did the following, let's see what happens:

Making sure it's not interference, as just before I left I also paired a bunch of Nest Smoke detectors.  Thinking "what changed?"..  Software to version 2.05.59, and the addition of 7 new wifi devices which happen to be on the ceiling with the lights.  California makes me have more smoke detectors than required. :)

1. Moved all Wifi access points from Channel 6 to Channel 1.  Of course, we have ~30 wifi devices, we live in a microwave it seems.
2. Moved from Channel 15 to Channel 25 on Zigbee.

Channel 6 and Zigbee 15 overlap - but not 1 and 25.

Basically, there is new behaviour after the update.  What happens now is not only do I have the random light issue, 80% of the lights completely stop working via ZigBee control.  Resetting the power to all the lights in that group immediately makes them come back to life.  They ALL work again for a day or so, then it's back to having to reset.  This is totally NEW after the update.

So, Taking your advice, I've removed each device from the group and hit apply. I then went back and added all devices which belong to that group back, and hit apply.  

1 hour after this, everything is working as expected, which is always the case.  The problems appear after time.  We will see tonight if the "closet light 03", which is as predictable as a Swiss train, decides to go rogue.

rmorriss, 2019-02-26:
@holli73 any idea how to enable logging whereas if I need support from Dresden, they will have the items then need off the bat?

rmorriss, 2019-02-27:
Well, seems like version .54 or .59 causes all kinds of problems above and beyond the problem with one random light going on. I'd say after 8 hours, 75% of the lights stop responding.  I have to reset them (they are grouped on a switch) and they come back to life for another 8 hours.  Back to 2.04.99, let's see if the one light issue come back.  this version is stable regarding the lights remaining active and responsive. @manup perhaps you have some ideas?

czebulski, 2019-03-06:
I can confrim the problem, regardless of used firmeware, either on deconz or osram!

closed 2019-07-11 by stale[bot]