dresden-elektronik/deconz-rest-plugin #211: Qestion, getting last state of the hue switch (on or off)

TimothyFehr, 2017-10-01:
Hello, 

how can I get the last state of the Hue Dimmer switch ? it seams not realy to change much when i run the switch with your dock (with a small exception, which is the last seen timestamp). No i do not have hue lights to directly connect to.

And is there a more efficent way (other than a custom plugin in c++) to request state changes from deCONZ than the rest api ? (since light switches should be nearly instant and i rather not force pull the data over the rest api (if possible).

thanks in advance
Tim

ebaauw, 2017-10-01:
> how can I get the last state of the Hue Dimmer switch ?

The last button pressed is encoded in `state.buttonevent`.  For the meaning of the values, see https://developers.meethue.com/documentation/supported-sensors.

> And is there a more efficent way (other than a custom plugin in c++) to request state changes from deCONZ than the rest api ?

Yes there is, using the web socket notifications.  See http://dresden-elektronik.github.io/deconz-rest-doc/websocket or https://github.com/ebaauw/dc_eventlog for an example programme (in NodeJS).

manup, 2017-10-06:
I close this for now since the question is answered.

closed 2017-10-06 by manup