dresden-elektronik/deconz-rest-plugin #212: with latest deconz (2.04.79) i have sporadic group command not working

holli73, 2017-10-03:
i have a room (kitchen) with 8 osram gu10 tw spots and since the latest deconz update one light will not turn on or off through the group command - so i removed the light from the group and added it again - this does  not help the light will not perform group actions - if within the room i do action on this single spot - it is working but not on the group where it is located.

any idea how i can troubleshot what is going on and why this spot is not reacting on group but on direct command?

thanks
holli

manup, 2017-10-04:
Should not be related to the deCONZ version. Is the light also in other groups, maybe the group table is full (I think OSRAM lights can be in up to 8 groups)? 

You can check this in the group cluster by *exec* the `Get group membership` command, capacity should be greater 0.

[image]

holli73, 2017-10-04:
hello,

cap: 7
grp cnt: 1
grp list: 0x0008

the light is only in this one group further the other lights report the same information.

this is the group information
```
    "8": {
        "action": {
            "bri": 255,
            "colormode": "ct",
            "ct": 153,
            "effect": "none",
            "hue": 0,
            "on": true,
            "sat": 0,
            "xy": [
                0.312704,
                0.328988
            ]
        },
        "devicemembership": [],
        "etag": "cc1cca0d3f1e0c3464456377dcba3a45",
        "hidden": false,
        "id": "8",
        "lights": [
            "19",
            "46",
            "17",
            "14",
            "16",
            "13",
            "15",
            "22"
        ],
        "lightsequence": [
            "19",
            "17",
            "46",
            "14",
            "22",
            "16",
            "13",
            "15",
            ""
        ],
        "multideviceids": [],
        "name": "kueche",
        "scenes": [],
        "state": {
            "any_on": true
        },
        "type": "LightGroup"
    },
```

so all looks ok - but this light is not reacting on the group but on single commands...

thanks
holli

manup, 2017-10-04:
>so all looks ok - but this light is not reacting on the group but on single commands...

Yes indeed from ZigBee view it should work, group commands are quite simple and all lights which are in the group should accept them.

Given that it looks like an light firmware issue, next thing that can try:

* power cycle the light, turn it off for a few seconds (I guess you may already did that), this is quite simple but often helps for weird behaving devices
* reset the light with OSRAM reset sequence and rejoin it

holli73, 2017-10-08:
@manup 
i'm now on .80 but still there is a problem - i do have lights which do work after a real power cycle but after some time the do not react on group command - within the deconz all looks ok (yellow) but the lights are not reacting - if i do a power cycle they start reacting - i only have osram lights - but it looks like onyl the rgb stripes and e27 rgbw bulbs are effected more often the gu10 spots and surface tw do not show this that often but do not perform as stable as before .70...

so maybe the new polling makes thinks worse - looks like lights are ok from deonz point of view but not connected/reacting on group command.

i further saw that on some lights i do have negative TEMP values within the webinterface - i did never see this before .80

any idea?

thanks
holli

manup, 2017-10-08:
The biggest change in newer deCONZ versions is the setup of ZCL attribute reporting for OSRAM lights. My E27 OSRAM color light and GU10 with latest versions support this very well.

However maybe there are issues for some specific lights or older firmware versions. I'll put a global switch for reporting in the next deCONZ version, so that the behavior can switched back to polling only.

Since the OSRAM gateway heavily uses reporting I figure there might be firmware issues in older lights. Note to myself : find a way to provide OSRAM OTA updates via deCONZ.

manup, 2017-10-08:
Also, if you can provide the output from `$ deCONZ --dbg-info=1 --dbg-zdp=1` when the issues occur, this would be helpful, thanks.

holli73, 2017-10-08:
hello,

i did run with the needed dbg info - i had to turn off the esszimmer twice - the light surface did go off but the stripes not:

if the tail catched it correct this was ok for turn off again:

```
23:11:26:033 delay sending request 128 - type: 14 to group 0x0009
23:11:26:133 delay sending request 128 - type: 14 to group 0x0009
23:11:26:233 delay sending request 128 - type: 14 to group 0x0009
```

the previous turn off did only turn off the surface but not the stripes.

hope this helps - there were also some tries before where it was not reacting as expected but could not catch this in the tail - too many messages but the example above should be fine - hopefully.

best regards
holli
[deconz80.dbg.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1367220/deconz80.dbg.zip)

holli73, 2017-10-08:
hello,

i have 3 lights where i did give them a name xx times already but the loose the name after deconz restart why?

2x osram
1x dresden led strip

best regards
holli

[image]

holli73, 2017-10-09:
hello,

i had to move back to .67 - because my wife got crazy - lights do not get on/off just not reacting after some time (even with .80 and latest commit's from yesterday).

i try to get an osram gateway to update my lights to the latest firmware since this lights are in place for almost 2 years now and without any update (i'm sure the osram devices did get some fixes as well because back than this was all brand new even for osram...)

is there any chance to update the osram devices through deconz or must i reset all 61 devices and connect to the lightify gw?

or can i just update one device from each brand (gu10/e27 rgb/surface tw/led stripe) and somehow copy the fw to the other ones?

thanks
holli

manup, 2017-10-09:
Oh sorry to hear that, from the logs I see that the group commands are executed with a way too large delay, the delays were added to prevent filling up queues, I'll lower them again for group commands in the next version.

> is there any chance to update the osram devices through deconz or must i reset all 61 devices and connect to the lightify gw?

Inofficial yes, I should be able to download OTA firmware from any gateway with an internal tool, I can this and provide the OTA files to you for your lights via Email. Can you make a screenshot from the `Plugins > STD OTAU Plugin` panel, I need the *Version* and *Image* info to retrieve files from the OSRAM gateway.

ebaauw, 2017-10-09:
> I should be able to download OTA firmware from any gateway with an internal tool

You're not, by change, looking for any private beta testers for this tool?

Reading their forum, it would seem that Smartthings allows OTA updating the OSRAM firmware, but I don't know where they get it from.

holli73, 2017-10-09:
@manup,

email is send - hopefully on the correct one took the one from the git commit history.
maybe i could have written in german because i'm located in austria **gg**

i could not provide a lot because this ota list was not updated after i queried the cluster/ota info.
i further could not get for all devices this information on the read request.

thanks a lot for any help.

best regards
holli

manup, 2017-10-09:
>You're not, by change, looking for any private beta testers for this tool?

When it is a bit more user friendly I can provide you the code for personal usage (it's another deCONZ plugin). But I'm afraid it will (should) never go public since some vendors would get very nervous and hackers very happy ;) some vendors protect and encrypt their firmware, some not so much.

I'm thinking of maybe at some point provide "all the firmware" but in a secured and encrypted way via deCONZ. The madness of not being able to update devices is not really acceptable, albeit I understand that for vendor legal departments the question is not that easy (who is responsable if gateway XY destroys our devices?).

> Reading their forum, it would seem that Smartthings allows OTA updating the OSRAM firmware, but I don't know where they get it from.

I heard that too but couldn't find more information about it, would be nice.

manup, 2017-10-09:
> I would help if the OTAU plugin would support sorting on the colums, or at least on Address. And, of course, if it would take over the node selection on the main screen.

Good points, I put them on the todo list it looks quite chaotic in larger networks :)

donnib, 2017-10-10:
> i had to move back to .67 - because my wife got crazy - lights do not get on/off just not reacting after some time (even with .80 and latest commit's from yesterday).

@holli73 i have same issues as yours both the lights and the wife. She is about to cut my head soon if i don't fix it so for the moment we just power cycle the lights and i don't use deConz because it's so unreliable in my network that i can't use it. I am surprised that others that does not have IKEA lights also have problems. It may then not just be an IKEA problem after all. Does 2.04.67 work for you ? See my issue here : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/195#issuecomment-335261655

holli73, 2017-10-10:
@donnib,

no .67 did not work - i tried because my own house app was changed to use the new websocket feature - so i do not need to poll - but luckely i did not remove the old code.

i had to move back to 'deconz-2.04.35.deb' - this is far the stablest version - i can recommend - sadly no websocket functionality but all GROUP commands are done really fast and stable.
further no light's are switching on during the night as well.

most important - my wife is not complaining anymore (since at least 48hours...)

best regards
holli

manup, 2017-10-10:
Version 2.04.81 for RPi and Ubuntu is up, it fixes issues related to OSRAM and IKEA and should help better to to find issues in donnis network.

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.04.81-qt5.deb

holli73, 2017-10-10:
@manup,

i tried .81 it is better than before but still some group command are not executed.
f.ex. in my kitchen i have one surface tw and 2 led flex stripes.
so i call on for the surface group and indirect flex - fine - i call off - one flex is still on...

and it looks like the color code is changed as well - my blue is still blue on the flex but cold white is now yellow white?

i moved back to .35 - because in the morning when my wife uses the lights - they should work and if the switch is pushed the light's must be on/off.

even in my big bathroom with 19 spots in 3 groups the on works almost instant but the off is not instant anymore and some light's turn off later...

if some traces are need i'm sure i can provide what ever is need - i did upgrade all my osram devices to the latest firmware as well (maybe the firmware is the problem for the new colors...)

best regards
holli

PS: with .35 the color is changed as well so it must be something with the new firmware - i have to check what new color code i need to send.

ebaauw, 2017-10-12:
Still seeing issues in .81 (stock version w/o the post-.81 commits):
- One IKEA bulb turned red and would only revive after power cycling the bulb;
- Sometimes a delayed reaction to group and/or scene commands - sometimes commands even seem to get lost.
- Huge variations over time in the mesh density: sometimes the deCONZ GUI shows only a few lines, sometimes it show very many lines.

manup, 2017-10-12:
> Sometimes a delayed reaction to group and/or scene commands - sometimes commands even seem to get lost.

I'm doing some tests with group commands to minimize delay, in the old implementation the commands were faster but totally unchecked and may cause queues to get stuck. The current delays are due sanity but can be improved a lot.

> Huge variations over time in the mesh density: sometimes the deCONZ GUI shows only a few lines, sometimes it show very many lines.

That's a result of the IKEA, OSRAM investigation, after setup stage deCONZ will slow down query of neighbor tables dramatically. Works but I have not yet updated the visual timeout of the mesh lines, it will follow in the next release.

> Motion sensors and dimmer switches losing connection to their parent (blinking red), but finding a new parent.

This can be related to the missing post .81 commits. I'll watch it in my setup, usually if Philips end devices reconnect it is due the gateway doesn't send the ZCL default response after a few reports. Which is most likely due a full queue.

closed 2018-09-17 by manup