dresden-elektronik/deconz-rest-plugin #213: Gibt es eine Chance auf Support in deutsch?

Krocko, 2017-10-03:
Hallo @manup. Besteht die Möglichkeit, auf Support von dir auf deutsch? Hatte vor einer Woche eine E-Mail an Dresden Elektronik geschrieben, aber leider keine Antwort erhalten. 
Ich habe einige Dinge festgestellt, welche nicht so richtig funktionieren, oder wo sich noch fragen aufgetan haben. Da meine englisch ehr schlecht ist (leider nur Schulenglisch mit etwas Google Übersetzer), würde ich das gerne auf deutsch erklären/beschreiben.

manup, 2017-10-04:
Hallo @Krocko, ja per Support-Mail ist das kein Problem. Üblicherweise antwortet unser Support sehr flink, da ist wohl noch was in der Warteschleife.

Welches Datum war die Mail? Da kann ich nochmal nachfragen.

Krocko, 2017-10-04:
Das war letzten Mittwoch. 
Kann man dir auch persönlich eine Mail zukommen lassen?
Da du ja der Entwickler bist, wäre die Fehlerbeschreibung gleich bei der richtigen Person.

manup, 2017-10-04:
Da scheint was im System untergegangen zu sein, zumindest die Mails vom Mittwoch haben alle eine Antwort.

>Kann man dir auch persönlich eine Mail zukommen lassen? 

Zumindest indirekt :) komplexere Fragen bzgl. deCONZ, RaspBee usw. landen i.d.R. schnell bei den Entwicklern. Entwicklung und Support ist bei uns im ständigen Austausch.

Daher am einfachsten bitte die Mail nochmal an support@dresden-elektronik.de schicken, mit dem Zusatz "bitte an MPI weiterleiten".

Krocko, 2017-10-08:
Erledigt. E-Mail ist raus.

closed 2017-10-08 by Krocko

manup, 2017-10-08:
Ok danke, schon gelesen, ich werde morgen die Punkte durchgehen und antworten — sind ja einige :)

Krocko, 2017-10-10:
Guten morgen Manuel. Hattest du dich schon um meine Mail gekümmert?
Bei mir hat sich die Situation seit der .80 erheblich verschlechtert.