dresden-elektronik/deconz-rest-plugin #214: Check RaspBee firmware and update to the latest version

Moskus, 2017-10-07:
I got a RaspBee board, installed deConz on a RPi3 and set it up. I even managed to update deConz to the lastest firmware. However, I'm not able to add an Ikea Trådfri bulb, which makes me think I need to update the firmware on the RaspBee board to the latest beta.

However, I can't figure out how to do that. I found there's a update tool on the Linux image already, but not where to find the firmware files.

I am obviously missing something, but what?
(And why isn't there an easy way to do this from the web interface? ;) )

manup, 2017-10-07:
If you have installed the latest deCONZ beta as described in steps 1&ndash;3 on the following page, you will get asked in the webapp to update the firmware (if needed) and just need to press the update button.

https://github.com/dresden-elektronik/deconz-rest-plugin#introduction

To add IKEA lights you must be reset them first as shown in this short video https://www.youtube.com/watch?v=mJm9YpPrGzk

Moskus, 2017-10-07:
Hmm... I did press update after updating the webapp. But I found the problem: The bulb was completely new and I'm guessing the firmware wasn't new enough. Took a bulb already connected to the Trådfri gateway, reset it, and it connected without problems. :)

closed 2017-10-07 by Moskus