dresden-elektronik/deconz-rest-plugin #215: Raspbee not updating

alexsporn, 2017-10-07:
I nuked my old Raspbian installation and wanted to start with a fresh deCONZ.

Now I have a problem updating the RaspBee firmware to be able to use the latest beta.
I have tried running the update from the webapp and manually with GCFFlasher_internal on a Pi2 and on a Pi3 using both Raspbian jessie and stretch. It always got stuck while transfering the bytes (the green led also stopped blinking).

Now I can not even get the transfer started. I always get:
```pi@raspberrypi:~ $ sudo GCFFlasher_internal -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26190500.bin.GCF 
GCFFlasher V2_10 (c) dresden elektronik ingenieurtechnik gmbh 2016/09/26
no device specified use RaspBee (/dev/ttyAMA0)
using firmware file: /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26190500.bin.GCF 
reset target 

error: bootloader not responding
device used by other application?```

Resetting the Raspbee using GCFFlasher_internal -r gives no error but also does not fix anything.
Now the Raspbee led is blinking with a very dimmed red color.

Any hints? Is my RaspBee bricked?

manup, 2017-10-07:
> Any hints? Is my RaspBee bricked?

Would be the first one :)

A few things which you may check:
* deCONZ must be closed while using GCFFlasher
* on Stretch a recent beta must be used (seems to be not the problem here)
* serial must be configured:
`$ sudo raspi-config` > Interfacing Options > Serial > Select **No** for login shell to be accessible over serial.

Since you have an Pi2 Bluetooth should not be an issue.

alexsporn, 2017-10-07:
Before reading your response I remembered there was an issue with Pi3, Bluetooth and UART, so I disabled bluetooth and after 2 tries it worked :-)
Now moving the RaspBee back to the Pi2 and see if I can finally use the latest deCONZ

alexsporn, 2017-10-07:
It's working now. Thanks!

closed 2017-10-07 by alexsporn

buckettea, 2018-07-11:
I'm now facing the same error, and it seems my raspbee is almost certainly bricked. I had it working fine on a RPi 2 model B+ and upgraded to a RPi3. All I did was change out the Rpi and noticed that the Pi didn't light up at first (known good pi). a couple of jacks with the power and it came up but with the continuous 'upgrade firmware' loop. 

i had made various other changes to the O/S, but mainly around pi/root passwords, ssh, etc. and i doubt this contributed but thought i'd mention it. 

i tried various versions and firmware with both the flasher and deCONZ and keep getting this error below. I even went back to the original Pi 2 and a fresh image that came with the device (i.e. exactly as i unpacked and installed).

i noticed the module felt a bit hot, but i don't know if this is normal? it still has an LED and doesn't give any errors on a reset and is listed with the flasher -l flag as /dev/ttyAMA0
The firmware reported in the GUI is "unknown" while in the web tool is "0000000".

Are there any other diagnostic tools i can use to verify if the Raspbee is damaged? 

error: bootloader not responding
device used by other application?

i can see /dev/ttyA

manup, 2018-07-12:
The RaspBee can only be damaged by connecting it to the wrong pins, otherwise bricking is unlikely.

It is very important that the serial port is configured as mentioned above. For a quick check I would recommend that you try the dresden elektronik sd-card image, it should work out of the box. At least the connection to RaspBee should work, ZigBee network configuration should then be the next thing to check.

https://www.dresden-elektronik.de/funktechnik/solutions/wireless-light-control/raspbee-gw-sd-card-image/?L=1

buckettea, 2018-07-25:
thanks for your responses manup.

same result, i guess it's damaged. one thing i noticed was that in the webapp "your gateway is on channel 0 and has the network ID of 0x0000"

before, it was on channel 11. so, i don't know if this means it's not communicating at all or if it's stuck in some weird state. 

i have ordered another module, but waiting for the mailman.

is there anything i can do at the serial port level to interrogate the device?

buckettea, 2018-07-26:
I got this via tcpdump from a pc to the webapp - i notice this part, and have some vague recollection about access points (that this would have a profound effect): "wifitype":"accesspoint"

is it possible it's stuck in some mode (AP) stopping it from taking a new image?

TP/1.1.200.OK..A
	0x00f0:  6363 6573 732d 436f 6e74 726f 6c2d 416c  ccess-Control-Al
	0x0100:  6c6f 772d 4f72 6967 696e 3a20 2a0d 0a43  low-Origin:.*..C
	0x0110:  6f6e 7465 6e74 2d54 7970 653a 2061 7070  ontent-Type:.app
	0x0120:  6c69 6361 7469 6f6e 2f6a 736f 6e3b 2063  lication/json;.c
	0x0130:  6861 7273 6574 3d75 7466 2d38 0d0a 436f  harset=utf-8..Co
	0x0140:  6e74 656e 742d 4c65 6e67 7468 3a31 3236  ntent-Length:126
	0x0150:  0d0a 0d0a 7b22 7769 6669 223a 226e 6f74  ....{"wifi":"not
	0x0160:  2d63 6f6e 6669 6775 7265 6422 2c22 7769  -configured","wi
	0x0170:  6669 6170 7077 223a 2222 2c22 7769 6669  fiappw":"","wifi
	0x0180:  6368 616e 6e65 6c22 3a22 3122 2c22 7769  channel":"1","wi
	0x0190:  6669 6970 223a 2231 3932 2e31 3638 2e38  fiip":"192.168.8
	0x01a0:  2e31 222c 2277 6966 696e 616d 6522 3a22  .1","wifiname":"
	0x01b0:  4e6f 7420 7365 7422 2c22 7769 6669 7479  Not.set","wifity
	0x01c0:  7065 223a 2261 6363 6573 7370 6f69 6e74  pe":"accesspoint

manup, 2018-07-27:
>is there anything i can do at the serial port level to interrogate the device?

The most easy test if the communication is working, is using GCFFlasher to program some firmware. On the official sd-card image this must always work.

>is it possible it's stuck in some mode (AP) stopping it from taking a new image?

No shield communication is done on serial port only and doesn't interfere with WiFi.