dresden-elektronik/deconz-rest-plugin #216: IKEA Trådfri Dimmer

ebaauw, 2017-10-08:
Got myself a Trådfri dimmer switch as well - it was lying next to the colour bulb at Ikea screaming "Buy me!".

Did the firmware update to 1.2.248 using deCONZ on my test RaspBee.  Had some issues connecting it to my production RaspBee running v2.04.80 afterwards.  I would show in the deCONZ GUI, but the descriptor wouldn't be read.  Had a similar issue with the new remote that came with the colour bulb: button presses would register in deCONZ, but reading the Basic cluster from the GUI would fail (blinking red in deCONZ GUI).  Sniffing showed no Read Attribute commands being sent at all?!  Further sniffing revealed it was connected to a Philips Hue light.  I powered off the Philips Hue lights in the room, forcing both switches to connect to the RaspBee directly.  After that, the descriptor and Basic cluster were read without issues and the REST API plugin created the resource for the dimmer.

I expected the dimmer would be using the magnetic contact, but it actually works without the base unit: it seems to contain a gyroscope or something similar.  It's extremely verbose when attached to the base unit, especially in a vertical position (as if the magnet is interfering with the gyroscope) .  It will probably help it I fixate the base unit to a wall (or fridge door or whatever), but I doubt whether it can be used in practice.

Sniffing, I find that it actually uses two discrete values for rate with the Move and Move with On/Off commands: 0x46 for a small move, and 0xC3 for a larger move.  Also, there seems to be a fixed 1:1 relation between Move (with On/Off) and Stop.  I would suggest to simply ignore the Stop and maybe use the rate to distinguish between release (after press) and release after hold.  Also, I would use additional buttons for On and Off (based on the Move to Level with On/Off commands when turning the dimmer abruptly).  There's a bug in the REST API plugin that both an abrupt right turn as well as an abrupt left turn result in buttonevent 1000.  Probably because the check for cmd 0x04 (Move to Level with On/Off) is missing from https://github.com/dresden-elektronik/deconz-rest-plugin/blob/0ebbae77a3144218832af1eb8917c517aee872de/de_web_plugin.cpp#L2163.  Finally, the DimUp and DimDown buttons are reversed - for the other switches DimUp comes before DimDown.  

So I suggest the buttonmap would be (similar to the Hue dimmer):

cmd 0x04, level 0xFF -> 1002
cmd 0x05, direction up (0x00), rate 0x46 -> 2002
cmd 0x05, direction up (0x00), rate 0xC3 -> 2003
cmd 0x01, direction down (0x01), rate 0x46 -> 3002
cmd 0x01, direction down (0x01), rate 0xC3 -> 3003
cmd 0x04, level 0x00 -> 4002

Not sure if all apps would be happy with a x003 buttonevent without a corresponding x001.  Maybe better to ignore the rate and only issue 2002 and 3002.

I'm happy to create a PR if you agree, @manup.

manup, 2017-10-08:
I have to admit the current implementation is just a very quick and dirty shot to provide *some* API to the dimmer, so any PR which improves behavior is very welcome :)

I can test it then with the older firmware, hopefully they didn't change too much.

ebaauw, 2017-10-08:
Hmm, something is wrong on my RaspBerry.  Not sure if it's related to Raspbian Stretch or if I forgot some dependency when re-installing the RaspBerry from scratch. 
```
$ qmake
$ make
make -f Makefile.Release
make[1]: Entering directory '/home/pi/GitHub/deconz-rest-plugin'
g++ -c -pipe -Wno-attributes -Wall -Wno-attributes -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DARCH_ARM -DARCH_ARMV7 -DUSE_WEBSOCKETS -DGW_SW_VERSION=\"2.04.80\" -DGW_API_VERSION=\"1.0.4\" -DGIT_COMMMIT=\"8395bdf9edf93189505bb9e1965167ec9a017570\" -DGW_MIN_RPI_FW_VERSION=0x26190500 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"deCONZ-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include -I../.. -I../../common -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o release/authentification.o authentification.cpp
In file included from /usr/include/c++/6/bits/stl_algo.h:59:0,
                 from /usr/include/c++/6/algorithm:62,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/qglobal.h:94,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/qatomic.h:41,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/qrefcount.h:43,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/qbytearray.h:44,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/qcryptographichash.h:44,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/QCryptographicHash:1,
                 from authentification.cpp:11:
/usr/include/c++/6/cstdlib:75:25: fatal error: stdlib.h: No such file or directory
 #include_next <stdlib.h>
                         ^
compilation terminated.
Makefile.Release:421: recipe for target 'release/authentification.o' failed
make[1]: *** [release/authentification.o] Error 1
make[1]: Leaving directory '/home/pi/GitHub/deconz-rest-plugin'
Makefile:38: recipe for target 'release' failed
make: *** [release] Error 2
```
Oddly, stdlib.h is available:
```
$ find / -name stdlib.h 2>/dev/null
/usr/include/c++/6/stdlib.h
/usr/include/c++/6/tr1/stdlib.h
/usr/include/arm-linux-gnueabihf/bits/stdlib.h
/usr/include/stdlib.h
/opt/Wolfram/WolframEngine/11.0/SystemFiles/Links/ArduinoLink/Resources/CSource/avr-libc/1.8.1/avr/include/stdlib.h
```

ebaauw, 2017-10-08:
Patching `Makefile.Release`, changing `-isystem /usr/include` with `-I /usr/include` seems to workaround the issue.

manup, 2017-10-08:
Hm strange, I haven't compiled on stretch yet, build server is still on jessy. I'll put up a second build pi for that the next days.

kernehed, 2017-10-12:
@ebaauw 

How did you make it work? Im total noob on this. I got the same "fatal error" as you on Raspbian stretch.

How do you do this:

_"Patching Makefile.Release after running qmake, changing -isystem /usr/include to -I /usr/include seems to workaround the issue."_

ebaauw, 2017-10-12:
With `vi`, but your favourite text editor will work just as well.

kernehed, 2017-10-12:
Sorry, what file is it I should change? :(

manup, 2017-10-12:
Are you on stretch? On fresh stretch installations the serial must be configured via `raspi-config`.

kernehed, 2017-10-12:
Hi @manup I did enable serial but still not connected..

Is there anything else I need to know about? The Raspbee has steady light of red...

ebaauw, 2017-10-12:
> Do you mean edit Makefile.Release and make the change there?

Yes, indeed.

> I did enable serial but still not connected

The hardware for the serial port needs to be enabled, but the login shell on it should be disabled.

Does GCFFlasher find the RaspBee?  It should report something like:
```
$ sudo GCFFlasher_internal -l
GCFFlasher V2_10 (c) dresden elektronik ingenieurtechnik gmbh 2016/09/26
RaspBee: /dev/ttyS0 (default)

no FTDI devices found
```

kernehed, 2017-10-12:
GCFFlasher V2_10 (c) dresden elektronik ingenieurtechnik gmbh 2016/09/26
RaspBee: /dev/ttyAMA0 (default)

no FTDI devices found

ebaauw, 2017-10-12:
The error messages appear when you first run GCFFlasher - just ignore them.  Are you on a pi2?  On the pi3, `/dev/ttyAMA0` is the Bluetooth chip.  Are you sure you disabled the login shell on the serial port in raspi-config?

kernehed, 2017-10-12:
Im on a pi2.

I disabled the login shell now. I will report back if it still doesn't work.

kernehed, 2017-10-12:
Well, now it works.

I want to thank you both, real heroes!

Now I just gonna implement it in HA! 👍

ebaauw, 2017-10-12:
The PR I made is included in v2.04.81, so I'm closing the issue.

closed 2017-10-12 by ebaauw