dresden-elektronik/deconz-rest-plugin #218: IKEA Colour bulb

ebaauw, 2017-10-08:


mariusmotea, 2017-10-09:
Hi!

Today i also buy some IKEA color bulbs. Everything is working fine, except when i apply hue scenes only the color of the bulb is changed, not also the brightness that come default in the scene

I have the following output:

```
scene "Savanna sunset":
LightRequest: http://127.0.0.1:8080/api/283145a4e198cc6535/lights/7/state
 {"e":"changed","id":"7","r":"lights","state":{"bri":203},"t":"event"}
{"e":"changed","id":"7","r":"lights","state":{"x":24827},"t":"event"}
{"e":"changed","id":"7","r":"lights","state":{"y":26102},"t":"event"}
```
but this is not correct, as the bulb was setup previously on brightness 13 and remain with this brightness. After some time i see a new notification with correct brightness:

```
{"e":"changed","id":"7","r":"lights","state":{"bri":13},"t":"event"}
```
This can be a bulb firmware issue, but maybe there is a workaround that can be implemented in deconz (send brightness command with a small delay?)

Official Hue application support this bulb very nice, despite is not original Philips bulb. When i select to change the state it will show me only the color palette and from there is very easy to choose cold white/warm white or any other color. I really don't feel the lack of color temperature option.

ebaauw, 2017-10-09:
> This can be a bulb firmware issue

I don't think so.  When I recall a scene, both colour and brightness are restored correctly.

You're probably having an issue with the scene not being stored correctly on the bulb.  Note that the first events are issued from the scene as stored on the gateway; the event after a while is issued when the light reports its attributes (or when the gateway polls the light).

manup, 2017-10-09:
>I don't think so. When I recall a scene, both colour and brightness are restored correctly.

Same here I've created 4 scenes with various colors and brightness settings, on scene recall these get restored as expected.

Reporting should work for both, color and brightness (with deCONZ 2.04.80).

mariusmotea, 2017-10-10:
This is something new for me, did not know the scenes are stored in bulbs, i only sent all scene attributes via PUT request thru REST API.
In this case how i can create the scenes in bulbs and call them via REST API? Same procedure like hue api protocol?

I will close #187 because i guess it was happening only when scenes are applied.

ebaauw, 2017-10-12:
Scenes are actually stored on the bulbs in the 0x0005 cluster.  When recalling a scene, the gateway only broadcasts group ID and scene ID and each bulb restores the state it previously stored for that group/scene.  When creating a scene using the REST API, the gateway instructs the bulb to store its current state under the group/scene.  Also, the REST API stores the (cached) state of each light with the scene, so it can update its cache copy of the light states (on which the web socket notifications are based) when the scene is recalled.  I find there's plenty of scenarios where the state for a certain scene as stored on the bulb differs from the cached state in the REST API, especially when storing the scene while the lights are still changing (e.g. because of a longer transition time, or when the bulb's state has changed since it was last polled).  Also, there's some bugs in storing scenes (see e.g. #53).

You cannot see the scene state stored in a light, other than recalling the scene.  You can store the current light state as a scene using the deCONZ GUI, in the cluster info panel for the Scenes cluster 0x0005, but you need to do some decimal to hex conversions for the group ID.

The Hue API uses a different syntax for scenes, `/scenes/id` instead of `/groups/id/scene/id`.  Under the hood, the Hue bridge creates a regular ZigBee group, to store the scenes.  Each light is made a member of this group, when first connected to the Hue bridge.  This group is exposed as `/groups/0`, but it's corresponding ZigBee group ID is not 0.  This differs from the `/groups/0` exposed by deCONZ, which is actually the ZigBee group 0.  Otherwise, the Hue bridge works the same (for v2 scenes), incl. the occasional mismatch between the state stored on the bulb vs the cached state on the bridge.

mariusmotea, 2017-10-12:
@ebaauw Thanks for this detailed description, it help me to understand how scenes are working. I still have a question. When default scenes are created on Hue Bridge for a particular room, the lights play all the scenes at fast speed in order to inform the bulbs to store is current state? I ask because i don't have a hue bridge and all i can see from application side is the scene POST request with scene details. Is there a limit of scenes that can be stored in one bulb, because i saw Hue application create a new scenes for every switch and sensor like there are plenty available?

Marius.

ebaauw, 2017-10-12:
> When default scenes are created on Hue Bridge for a particular room, the lights play all the scenes at fast speed in order to inform the bulbs to store is current state?

Not sure, actually.  I never sniffed that part.  Or maybe the scene gets created in the bulb only when it's first recalled.

> Is there a limit of scenes that can be stored in one bulb

Yes, and in the wonderful world of ZigBee standardisation, different manufacturers use different limits, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/29#issuecomment-297516311.

lcorsini, 2017-10-17:
I bought one of those too, there is no OTA update, but work with hue bridge out of the box, but it doesn't seems to like hue scenes at all and when turned on it appears as CT lights until you change intensity. strange behaviour

DOliana, 2017-11-05:
I just discovered this (especially point 2 of the original comment) on my own and I suppose, this is the reason for openhab not being able to set the color (even though I do not see, which method it uses). Adding to the comment: when setting the hue/sat values, the API returns success - I think this should also be changed to something like "not supported".

closed 2018-06-12 by ebaauw