dresden-elektronik/deconz-rest-plugin #219: Getting Mesh information via REST or other means, without frontend

tgroeneveldlg, 2017-10-09:
Hi Manup,

Can you tell me if you have planned to make the information available within the UI, to build a mesh webbased UI, available through REST ?

I have found just the frontend for this, and would be great if I could access all information via REST/json.

Thanks !

Kind regards, Tom.

ebaauw, 2017-10-12:
You might want to checkout the undocumented `/lights/xx/connectivity2` resource.  There's also an undocumented `/lights/xx/connectivity`, but GETting that one just makes deCONZ hang (even the debug output stops).  Here's what it does for me:
```
$ ph_get /lights/221/connectivity2
{
  "connectivity": 246,
  "extAddress": "0x0017880100xxxxxx",
  "name": "Living Room 1",
  "neighbours": {
    "111": {
      "connectivity": 243,
      "name": "Entrance 1",
      "reachable": true
    },
    "222": {
      "connectivity": 247,
      "name": "Living Room 2",
      "reachable": true
    },
    "224": {
      "connectivity": 247,
      "name": "Living Room 4",
      "reachable": true
    },
    "225": {
      "connectivity": 247,
      "name": "Living Room 5",
      "reachable": true
    },
    "227": {
      "connectivity": 247,
      "name": "Living Room Left",
      "reachable": true
    },
    "228": {
      "connectivity": 247,
      "name": "Living Room Right",
      "reachable": true
    },
    "229": {
      "connectivity": 247,
      "name": "Living Room Desk",
      "reachable": true
    },
    "231": {
      "connectivity": 247,
      "name": "Kitchen 1",
      "reachable": true
    },
    "233": {
      "connectivity": 247,
      "name": "Kitchen 3",
      "reachable": true
    },
    "234": {
      "connectivity": 252,
      "name": "Kitchen Table",
      "reachable": true
    },
    "241": {
      "connectivity": 247,
      "name": "Kitchen Sink Left",
      "reachable": true
    },
    "242": {
      "connectivity": 231,
      "name": "Kitchen Sink Right",
      "reachable": true
    },
    "251": {
      "connectivity": 252,
      "name": "Washroom 1",
      "reachable": true
    },
    "261": {
      "connectivity": 247,
      "name": "Toilet 1",
      "reachable": true
    },
    "272": {
      "connectivity": 247,
      "name": "Jacco 2",
      "reachable": true
    },
    "273": {
      "connectivity": 247,
      "name": "Jacco 3",
      "reachable": true
    },
    "321": {
      "connectivity": 247,
      "name": "Bedroom 1",
      "reachable": true
    },
    "322": {
      "connectivity": 247,
      "name": "Bedroom 2",
      "reachable": true
    },
    "323": {
      "connectivity": 247,
      "name": "Bedroom 3",
      "reachable": true
    }
  },
  "reachable": true,
  "routesToGateway": 0
}
```

manup, 2017-10-12:
> You might want to checkout the undocumented /lights/xx/connectivity2 resource.

The connectivity resource hasn't been updated in years, so be aware it might fail badly :)

> Can you tell me if you have planned to make the information available within the UI, to build a mesh webbased UI, available through REST ?

It's not planned to forward the raw ZigBee functionality via REST/Websocket, however it could help a lot to get more details of not yet supported devices by capturing the raw APS packets and forward them via websocket. This could be added pupsub manner to the websocket interface so that it can be enabled on request. No ETA though.

tgroeneveldlg, 2017-10-13:
@manup > It's not planned to forward the raw ZigBee functionality via REST/Websocket
That is too bad, would be very nice to remove the full frontend, or make that optional to run in the future. I use a pure headless solution, and though I like the view of information, I can not run the UI constantly (not at all actually currently).

Porting this functionality to REST, might trigger people like myself, to create a very cool frontend so this information would be visual within a webpage like a sub page inside Home Assistant i.e.

@ebaauw Thanks for the info. I do not believe that this information is rich enough. Also, a specifically designed REST API call that would efficiently provide the mesh information, would be most desirable over pulling this information all the time from all machines manually.

Kind regards, Tom.

manup, 2018-09-17:
Closing the oldest issues for know to tidy up the tracker and duplicates in newer issues.

closed 2018-09-17 by manup

zoic21, 2019-03-01:
hello,

I try connectivity2 on sensors but it does not work... Are they any global rest endpoint to get all network topology ?

ebaauw, 2019-03-01:
Only Zigbee routers maintain a neighbour table (end-devices only connect to a single router, their parent).  As mentioned above, this is an ancient API endpoint.  It still assumes that all routers are lights, all lights are routers, all end-devices are sensors, and all sensors are end-devices.  I suppose this needs to be fixed in API v2, introducing the `/devices` endpoint.

imortkz, 2019-03-29:
Will the connectivity2 resource show signal strength for a sensor when it connected to the light router?

I wonder is there any way to check a sensor signal w/o GUI app?
Logs parsing maybe?