dresden-elektronik/deconz-rest-plugin #220: ConBee not found inside docker container

tgroeneveldlg, 2017-10-09:
Hi,

I am running deConz headless at the moment. Directly on Ubuntu it works like a charm, and the Conbee is found perfectly.

Now, I have tried various ways of pushing the stick to a docker container, but all have failed.

Tried using 'device' mapping, tried using 'volume' mapping, tried both with privileged set to true.

None have worked so far. The deConz web ui sets the channel to 0  and nothing happens after that. I can not control anything previously discovered (using the same DB.

So I reverted back to using deConz on the local OS again, but this is hardly desirable.

Any help would be appreciated, kind regards, Tom.

abmantis, 2017-10-14:
I've mapped it with "device" and it works, but you must map to /dev/ttyAMA0
For example: 
/dev/ttyUSB2:/dev/ttyAMA0 
or
/dev/serial/by-id/usb-FTDI_FT230X_Basic_UART_DJ00Q66M-if00-port0:/dev/ttyAMA0

jkinghed, 2017-10-17:
I'm also experiencing the same issue as OP. Spent hours on this, back and forth with different options and ways to map the Conbee stick. Mapping to ttyAMA0 suggested by @abmantis unfortunately doesn't work. 

Running checks from with in the container, I can see it in /dev

```
root@9becab556460:~# ls -l /dev/tty*
crw-rw-rw- 1 root root      5, 0 Oct 17 22:31 /dev/tty
crw-rw---- 1 root dialout 188, 0 Oct 17 22:31 /dev/ttyAMA0
```
And running GCFFlasher_internal -l it shows up in the list (along with some errors) as RaspBee. It should be said that if I map as ttyUSB0, nothing shows up in the list using this command, so at least I get a little bit closer...

```
root@9becab556460:~# GCFFlasher_internal -l
/usr/bin/GCFFlasher_internal: 9: /usr/bin/GCFFlasher_internal: rmmod: not found
/usr/bin/GCFFlasher_internal: 10: /usr/bin/GCFFlasher_internal: rmmod: not found
GCFFlasher V2_10 (c) dresden elektronik ingenieurtechnik gmbh 2016/09/26
RaspBee: /dev/ttyAMA0 (default)
/usr/bin/GCFFlasher_internal: 14: /usr/bin/GCFFlasher_internal: modprobe: not found
/usr/bin/GCFFlasher_internal: 15: /usr/bin/GCFFlasher_internal: modprobe: not found
```

lsusb shows all USB-devices, even though they aren't available in the container:

```
root@9becab556460:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 8087:0a2b Intel Corp.
Bus 001 Device 004: ID 0658:0200 Sigma Designs, Inc.
Bus 001 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Adding extra logging in startup of deCONZ: 

`DECONZ_OPTS="-platform minimal --auto-connect=1 --http-port=8080 --dbg-error=1 --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1" deCONZ-autostart.sh` gives a constant loop of actual USB devices connected in the host machine. 

```
2017-10-17T21:11:20.292336305Z 23:11:20:292 COM: /dev/ttyUSB0 : 
2017-10-17T21:11:20.292377706Z 23:11:20:292 COM: /dev/ttyACM0 : 
2017-10-17T21:11:20.292385161Z 23:11:20:292 COM: /dev/ttyS4 : 
2017-10-17T21:11:21.272185225Z 23:11:21:272 COM: /dev/ttyUSB0 : 
2017-10-17T21:11:21.272208109Z 23:11:21:272 COM: /dev/ttyACM0 : 
2017-10-17T21:11:21.272213615Z 23:11:21:272 COM: /dev/ttyS4 : 
```

Really not sure where to go from here. It feels like I'm missing something obvious somewhere... A silly mistake somewhere. Any help or suggestion would be appreciated!

Thanks.

tgroeneveldlg, 2017-10-20:
I have found the same thing indeed, it does not matter as what I map the device to. I have tried (I think) it all !

tgroeneveldlg, 2017-10-23:
@jo-me how did you get docker to see it ? or is it different on DockerPi ?

abmantis, 2017-10-23:
I'm using the following Dockerfile and docker-compose.yml to run the container: https://gist.github.com/abmantis/09df711ece2e9f5b5156fea193a05862

Try creating the container with that Dockerfile since you may be missing some dependency.

jo-me, 2017-10-24:
Hi, 
as far as I understood, ConBee needs to be referenced as /dev/ttyUSB* (also in the container) because otherwise deConz thinks that a RaspBee is connected (since the presence of "/dev/ttyAMA0" is checked for presence of a RaspBee) . 
If you map the ConBee to the RaspBee device name it still somewhat works in deCONZ itself, but the firmware update mechanism is broken then (see my other thread).

So, in order to really get it running as it is intended to, deconz needs to find /dev/ttyUSB0 (or other number) when it is running in the container. Only then it wuill be able to use its features and check firmware, etc. 

The question is why deCONZ is not seeing /dev/ttyUSB0 . Seems to be related to how QT is looking for serial ports maybe..

abmantis, 2017-10-24:
@jo-me When I tried it deCONZ wouldn't connect to the device until I mapped it to AMA0. But yeah, the firmware update wouldn't work.

doctorjames, 2018-02-02:
My [comment](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/236#issuecomment-361907590) on #236 might be useful to followers of this issue regarding successfully getting deConz working successfully in a container.

manup, 2018-02-24:
Closing the issue for now since topic is also discussed in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/236

closed 2018-02-24 by manup