dresden-elektronik/deconz-rest-plugin #221: Trust Remote control switch

vamanea, 2017-10-10:
Has anybody gotten the Trust Remote Control switch(model ZYCT-202) to join the network. As far as I can tell from their website it should be perfectly compatible yet no matter what I tried it does not join the raspbee network.

Does anybody have better experiences?

manup, 2017-10-10:
No, they are not included yet, I wonder why they look *very* similar to the Philips dimmer switch.

vamanea, 2017-10-11:
I'm sure it's just a _coincidence_ that it has exactly the same shape and it even fits in the Philips switch holder.
Is there something I can do to help adding support? I noticed on the back of the PCB they have some test pads marked as SPI which I suppose would be useful to dump the flash.

andrejem, 2017-10-17:
I have bought a Trust motions sensor and Window OpenClose sensor, but i am not able to get them work with deconz for now.

vamanea, 2018-01-03:
I commented on the other thread as well:

I can confirm I got the remote ZYCT-202 to join the network and through my deConz -> mqtt bridge I can see some events from the remote.

I have no yet tried the REST API but things are looking much better, thanks for your hard work.

mat1990dj, 2018-09-16:
I also bought some remotes and motion sensors and are unable to see the info on the rest API. There are a couple threads already

closed 2018-09-17 by manup