dresden-elektronik/deconz-rest-plugin #222: Hue Apps don‘t find my Radpbee Gateway

Krocko, 2017-10-11:
I have testet various Hue Apps, but no one can find my Raspbee Gateway. Must i change some Settings?

ebaauw, 2017-10-11:
Obviously, the deCONZ gateway doesn’t announce itself to the meethue portal.  Also, it announces itself to UPnP differently than the Hue bridge, so Hue apps won’t find it without (minor) changes.  You might try and configure the IP address in the app instead of doing discovery (if the Hue app supports that).  I only tried the official Hue app, but that doesn’t like deCONZ, even when I hard-configured the IP address.

Krocko, 2017-10-11:
Mhhh, ok. But i heard from some German Users, that they control the Raspbee with an Hue App. Also on the German Support side is written, that we can use 3rd Party Apps.

mark-tech, 2017-10-11:
I control the Raspbee with Hue App, hueManiac and Alexa using an hue emulator from github as proxy. Spent some time to setup because lack of documentation but in the end the result is awesome.

Krocko, 2017-10-11:
@mark-tech: Sounds good. Can you please explain it?

mark-tech, 2017-10-12:
i use one rpi with diyhue on port 80 and deconz on port 8080. I encored some issues that are not documented but in the end i figured out by myself what was the problem and once my setup was working everything was fine. I remember no hue application detect the emulator if deconz was running in the same time. Need to stop deconz and restart the emulator for every new device you want to pair, then start again deconz and restart again the emulator. You will not find this in documentation.

ebaauw, 2017-10-13:
With PR #225, the Hue app connects to the deCONZ gateway as if it were a Hue bridge.  Note that there's still differences between the Hue API and the deCONZ API, so expect some issues, notably around scenes.

ebaauw, 2017-10-13:
And the Hue app crashes on my production gateway, probably because I have lights, groups and sensors with IDs > 64.  Or more than 200 rules.

mariusmotea, 2017-10-14:
@mark-tech feel free to open issues for any problems you have. Regarding ssdp discover, deconz also use this service to announce is presence on the network which enter in conflict with hue emulator. It cannot be disabled so the only workaround is to start the hue emulator with deconz stopped. I will update the wiki to cover also this aspect.

Krocko, 2017-10-15:
>And the Hue app crashes on my production gateway, probably because I have lights, groups and sensors with IDs > 64. Or more than 200 rules.

The Hue app crashes for me too. I have 25 nodes and 16 rules.
I think it is because of the unsupported devices. I have several Xiaomi Accessories.

ebaauw, 2017-10-15:
I’ll do some more testing on a rainy Sunday afternoon.  The weather is too good for that right now...

oleg77s, 2017-10-15:
Lightbow app can connect to Raspbee Gateway (deConz GW)

to0b, 2017-10-23:
My hue app can connect to the deCONZ gateway but i‘m not able to save a new room with some tradfri lights. The room disappear after a few seconds. Any suggestions?

ebaauw, 2017-10-23:
deCONZ doesn’t support Room groups.

manup, 2017-10-29:
The next version (2.04.85) does support these groups, not testet with the hue app yet though.

Commit https://github.com/dresden-elektronik/deconz-rest-plugin/commit/93f0a1303f456214b57bc5190fb2ecdcf0001f23

ebaauw, 2017-10-29:
Did you also implement the check that a light can only belong to a single room?

manup, 2017-10-29:
> Did you also implement the check that a light can only belong to a single room?

Nope, haven't looked that deep into the topic yet, was just playing around with iConnectHue :)

OleUrgast, 2017-11-05:
The Android App "All4Hue" works very well with RaspBee. You could even reprogram the colourtemperature buttons on a Tradfri remote.
You can also use "Disco" function.
But the scenes are not supported.

FrodoKenny, 2018-02-23:
With the official Hue app I'm able to find the RaspBee. If I connect it asks me to press the bridge button. I then unlock the gateway, but nothing happens and it times out. Am I doing something wrong?

FrodoKenny, 2018-03-05:
Just as an FYI, in case someone else runs into this. It turns out that the Hue app had connected to the Raspbee the first time I tried and then crashed. After that I reset the Raspbee gateway, but the Hue app caches the username and tries to use that after the gateway reset as well, therefore never getting acces by requesting a new username. Fixed by looking in the logs which username the Hue app uses, and manually adding it to the gateway with a POST /api/ with that username.

closed 2019-03-30 by stale[bot]

vandalon, 2019-09-09:
> > Did you also implement the check that a light can only belong to a single room?
> 
> Nope, haven't looked that deep into the topic yet, was just playing around with iConnectHue :)

I loved iConnectHue , but I can not get it working with deconz. Do you still use this app with deconz? :)