dresden-elektronik/deconz-rest-plugin #223: deCONZ versioning scheme incompatible with semver

ebaauw, 2017-10-13:
The deCONZ versioning scheme doesn't use valid Semantic Versioning ([semver](http://semver.org)) versions, as there's a leading zero in the the minor version (e.g. 2.**0**4.81).

Would it be possible to ditch the leading zero?  I'm reluctant to change this because of the comment https://github.com/dresden-elektronik/deconz-rest-plugin/blob/4e943ff403920c68e2aa1c17486e8e1e221e7867/de_web.pro#L59

Alternatively, could we use the API version to match the swversion, but as a valid semver string?

manup, 2017-10-13:
>Alternatively, could we use the API version to match the swversion, but as a valid semver string?

The API version was supposed to reflect the hue API version, I know some apps like iConnectHue will check this attribute, so maybe keeping (and note to myself: *maintaining it*) would be better.

I'll check the scripts and webapp "version compare functions" if they would break when the leading zero is removed from `swversion`attribute. Breaking things won't be an option since a lot of installations out there are only updated via WebApp.

Changing `swversion` could be done for api response only, so that scripts and defines stay at current format.

ebaauw, 2017-10-13:
> Changing swversion could be done for api response only, so that scripts and defines stay at current format.

That sounds like a good plan.

> The API version was supposed to reflect the hue API version, I know some apps like iConnectHue will check this attribute, so maybe keeping (and note to myself: _maintaining it_) would be better.

We discussed this earlier (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/20#issuecomment-290807573) - that won't be easy.

closed 2017-10-14 by ebaauw