dresden-elektronik/deconz-rest-plugin #224: Empty action.body when creating rules in v2.04.81

ebaauw, 2017-10-13:
When creating rules under v2.04.81 (with the latest commits), `action.body` seems to be ignored and remains empty.  It isn't stored to the database either.  It seems to happen to all rules, irrespective to the type of action (group, sensor, light).

It's been a while since I last updated my rules, so I'm not sure if this is issue is introduced by v2.04.81 or by an earlier change.  Existing rules had been working without any issues.  When I patch the database, adding the body, the rule works as expected.

For example see the following transcript:
```
$ ph_rules_boottime 107 103
pi2: debug: gateway command: curl -s -X POST -H "Content-Type: application/json" -d '{
    "name": "Boot Time",
    "status": "enabled",
    "conditions": [
    {
        "address": "/sensors/103/state/lastupdated",
        "operator": "dx"
      },
    {
      "address": "/sensors/107/state/flag",
      "operator": "eq",
      "value": "false"
    }
  ],
    "actions": [
    {
        "address": "/sensors/107/state",
        "method": "PUT",
        "body": {"flag": true}
      },
    {
        "address": "/groups/0/action",
        "method": "PUT",
        "body": {"on": false}
      }
  ]
  }' "http://pi2/api/xxxxxxxx/rules"
pi2: debug: gateway response: [{"success":{"id":"1"}}]
pi2: /rules/1: "Boot Time"
$ ph_get /rules/1
pi2: debug: gateway command: curl -s -H "Content-Type: application/json" "http://pi2/api/xxxxxxxx/rules/1"
pi2: debug: gateway response: {"actions":[{"address":"/sensors/107/state","body":{},"method":"PUT"},{"address":"/groups/0/action","body":{},"method":"PUT"}],"conditions":[{"address":"/sensors/103/state/lastupdated","operator":"dx"},{"address":"/sensors/107/state/flag","operator":"eq","value":"false"}],"created":"2017-10-13T13:19:08","etag":"c31af637d0b1faa3a6dad2dcfdc158bf","lasttriggered":"none","name":"Boot Time","owner":"xxxxxxxx","periodic":0,"status":"enabled","timestriggered":0}
{
  "actions": [
    {
      "address": "/sensors/107/state",
      "body": {},
      "method": "PUT"
    },
    {
      "address": "/groups/0/action",
      "body": {},
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/sensors/103/state/lastupdated",
      "operator": "dx"
    },
    {
      "address": "/sensors/107/state/flag",
      "operator": "eq",
      "value": "false"
    }
  ],
  "created": "2017-10-13T13:19:08",
  "etag": "c31af637d0b1faa3a6dad2dcfdc158bf",
  "lasttriggered": "none",
  "name": "Boot Time",
  "owner": "xxxxxxxx",
  "periodic": 0,
  "status": "enabled",
  "timestriggered": 0
}
```

As you can see from the deCONZ log, the body is received by deCONZ, but not returned:
```
15:19:08:865 HTTP client hdr detected:
POST /api/xxxxxxxx/rules HTTP/1.1
Host: pi2
User-Agent: curl/7.54.0
Accept: */*
Content-Type: application/json
Content-Length: 375

15:19:08:866 HTTP API POST /api/xxxxxxxx/rules - 192.168.xxx.xxx
15:19:08:866 Text Data: 	{ "name": "Boot Time", "status": "enabled", "conditions": [ { "address": "/sensors/103/state/lastupdated", "operator": "dx" }, { "address": "/sensors/107/state/flag", "operator": "eq", "value": "false" } ], "actions": [ { "address": "/sensors/107/state", "method": "PUT", "body": {"flag": true} }, { "address": "/groups/0/action", "method": "PUT", "body": {"on": false} } ] }
15:19:08:867 [{"success":{"id":"1"}}]
15:19:08:873 index resource items for rules, handle: 1 (Boot Time)
15:19:08:873 	/sensors/103/state/lastupdated op: dx
15:19:08:873 	/sensors/107/state/flag op: eq
15:19:08:873 	state/lastupdated (trigger)

15:19:13:209 HTTP client hdr detected:
GET /api/xxxxxxxx/rules/1 HTTP/1.1
Host: pi2
User-Agent: curl/7.54.0
Accept: */*
Content-Type: application/json

15:19:13:210 HTTP API GET /api/philipshue/rules/1 - 192.168.xxx.xxx
15:19:13:211 {"actions":[{"address":"/sensors/107/state","body":{},"method":"PUT"},{"address":"/groups/0/action","body":{},"method":"PUT"}],"conditions":[{"address":"/sensors/103/state/lastupdated","operator":"dx"},{"address":"/sensors/107/state/flag","operator":"eq","value":"false"}],"created":"2017-10-13T13:19:08","etag":"c31af637d0b1faa3a6dad2dcfdc158bf","lasttriggered":"none","name":"Boot Time","owner":"xxxxxxxx","periodic":0,"status":"enabled","timestriggered":0}
```

Attached the full log.
[deCONZ.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1382732/deCONZ.log)

ebaauw, 2017-10-13:
When doing a PUT to an existing rule, updating only the "actions", the body gets updated alright.

snozzlebert, 2017-10-13:
I've the same issue and I'm still on .79
PUT works also when updating conditions and actions in the same request.
Creating rules with POST works if you specify `action.body` as a string (serialized JSON) instead of a JSON object, e.g. `body: "{\"on\": true}"` instead of `body: {"on": true}`.
It is a workaround and this should be fixed.

ebaauw, 2017-10-14:
Thanks, @snozzlebert, that's insightful.  Found the bug and fixed it.  Waiting for PR #225...

closed 2017-10-14 by ebaauw