dresden-elektronik/deconz-rest-plugin #227: Serial com disconnected, reason: 4

abmantis, 2017-10-14:
For some reason, deCONZ cannot connect with ConBee anymore, giving me the "Serial com disconnected, reason: 4" error.
It was working correctly before I mistakenly ran it with sudo. 

Any tips? Reason 4 is a little vague :(

manup, 2017-10-15:
Which platform are you using, Raspbian or Ubuntu?

The permissions of the USB device */dev/ttyUSB0* should be met by the user running deCONZ, therefore check that the user `pi` is in group `dialout`:

``` bash
    # shows which groups current user is in
    $ groups 

    # if the user is not in group dialout, add it (reboot required)
    $ sudo gpasswd -a pi dialout
```

sophof, 2018-02-25:
I have the same problem and my user is in the dialout group and the permissions on /dev/ttyUSB0 are:
"crw-rw-rw- 1 root dialout"

So apparently this isn't the issue? Does anyone know what reason 4 acutally is?

manup, 2018-02-25:
Can you provide more detail?
OS?
Also after adding the user to dialout, a reboot/relogin is needed before settings are valid.

sophof, 2018-02-25:
I'm using Ubuntu server and used the latest deconz version (can't check
right now). Weird thing is that another user can access the USB, but no
devices show up in deconz. I've been using a zwave stick for ages, so the
user has been in the dialout group from the start.

I would like to keep using the original user though, since the other user
(the one that works) has too many rights on the machine.

manup, 2018-02-25:
Hmm I don't know what it is.
Can you please start deCONZ with more verbose error logging, this will print more detail for this particular error.

    $ deCONZ --dbg-error=2 --dbg-info=1 -platform minimal

Also make sure that only one instance is running (I've seen this issue often recently).

    $ ps ax | grep deCONZ

sophof, 2018-02-26:
Hmm, I get a "COM failed to open /dev/ttyUSB0: Permission error while
locking the device" error, so it does seem to be a permission error. I
double checked but the permissions do seem fine though. I even tried
forcing ownership of the tty to the user, but even then I get the same
error.

I have also confirmed there's no other deconz instance running, just these
two:
 1037 ?        Ss     0:00 /bin/bash /usr/bin/deCONZ-init.sh
 1038 ?        Ss     0:37 /bin/bash /usr/bin/deCONZ-update2.sh

manup, 2018-02-28:
Hmm to be honest I don't know where the problem is.

Maybe `$ sudo journalctl -xb -f` provides more information when the error occurs?

tumik, 2018-10-05:
I had the same issue on my Debian 9 server, appeared after updating deCONZ and accidentally running it with a different user (after systemd files were overwritten).
First I got the error:
`Serial com disconnected, reason: 4`

When running with:
`$ deCONZ --dbg-error=2 --dbg-info=1 -platform minimal`
I got the following error:
`COM failed to open /dev/ttyUSB0: Permission error while locking the device`

**The problem was a stale lock file** at /run/lock/LCK..ttyUSB0 where only root had permissions to delete it.
I made sure no deCONZ instances are running, deleted the lock file and re-started deconz. This fixed the issue.

I believe this issue can now be closed.

closed 2019-01-17 by manup

cmuellner, 2020-12-13:
In case you find yourself here, because you run into this problem with podman and you are sure your user is in group ``dialout``, then add the flag ``--annotation run.oci.keep_original_groups=1`` (see also https://github.com/containers/podman/issues/4477).