dresden-elektronik/deconz-rest-plugin #228: BUG (w/ Workaround):  Losing Lights and/or improper discovery

mysteryan, 2017-10-15:
For over a month, I noticed that certain nodes would keep dissapearing from the database.  The workaround I've been doing is to simply edit zll.db and manually edit those entries back in.  Everytime I do a restart, the same entries would dissapear so I needed to copy the manually edited database everytime.  This would happen to 1/4 my OSRAM nodes that are on ZHA.  The problem never existed on my Phillips Hue, Cree, or GE Link nodes.  The same exact nodes would appear in the GUI, but get delayed (or sometimes not appear) on the web ui.  If I didnt manually edit the database every time, it would get a new generic name ("Light ##") and receive a new ID based on the first one to report.  Makes it a nuisance to utilize the rest api based light IDs when they keep getting shuffled.

WORKAROUND:
I finally realized theres a common characteristic with the nodes that keep getting deleted.  They all have a MAC address that's all numerical.  You can append another endpoint id to the node that includes a letter, but that would probably cause more issues.  Instead, I found the seamless solution is to use an "x".  Specifically, edit your node list in your database to notate the address in 0x000000... format instead of 00:00:00:00... format it does by default. 

After altering the offending nodes that were all numerical to 0x notation, deconz runs properly and all nodes report properly even after a reboot.  It looks like theres a problem with how the database is being stored or written.  That or its an issue with QT5 since the problem doesnt exist on 2.0.35.  I'm running the beta branch on Pi3/Stretch w/ a ConBee.

manup, 2017-10-15:
> After altering the offending nodes that were all numerical to 0x notation, deconz runs properly and all nodes report properly even after a reboot. It looks like theres a problem with how the [...]

Which OSRAM lights show the problems, A60? Seems like they are not stored in the database after discovery, which is strange since the process is the same for all lights. The 0x00000000... format is the old one which will be converted to the new format `00:11:22:33:44:55:66:77-EP` where EP is the endpoint of related light clusters.

mysteryan, 2017-10-15:
This happens on all my OSRAM models.  They're US based bulbs which utilizes a numerical only ZHA "03" endpoint.  The model names I use are: "LIGHTIFY A19 ON/OFF/DIM", "LIGHTIFY BR Tunable White", and "LIGHTIFY RT Tunable White".  I'll disclose the mac of one of my bulbs.  My A19 bulb kept getting deleted unless it was offline that was stored as: 84:18:26:00:00:04:55:15-03.  When changed to 0x8418260000045515, it was fine.

manup, 2017-10-15:
Should be fixed in 2.04.83

mysteryan, 2017-10-16:
Looks like that issue was resolved by your commit

closed 2017-10-16 by mysteryan