dresden-elektronik/deconz-rest-plugin #230: v2.04.82 won't let ubisys dimmer join network

ebaauw, 2017-10-15:
v2.04.82 seems more stable than previous versions, but somehow my ubysis D1 dimmer dropped off the network.  This had happened before, with earlier versions, typically when the network would lock up.  After a factory reset the dimmer would re-pair.  However, the dimmer won't pair with v2.04.82.  It makes the connected lights blink steadily when power is applied (meaning it's searching for a network to join, according to the reference manual).  When opening the network in the deCONZ web UI, the blink pattern changes, like it's trying to join the network and failing.  After closing the network, the steady blink pattern is resumed.

I don't see the dimmer's MAC address in the deCONZ log with --dbg-info=2.  What logging would I need to enable to see what's going on?

According to its documentation, the dimmer supports up to 78 entries in its neighbour table and up to 96 entries in its routing table.  Could this be causing buffer overflows in deCONZ or in the RaspBee firmware?

manup, 2017-10-16:
>v2.04.82 seems more stable than previous versions, but somehow my ubysis D1 dimmer dropped off the network. This had happened before, with earlier versions, typically when the network would lock up. After a factory reset the dimmer would re-pair. 

Shouldn't a powercycle be sufficient to get the device back?

>I don't see the dimmer's MAC address in the deCONZ log with --dbg-info=2. What logging would I need to enable to see what's going on?

To see APS and ZCL traffic `--dbg-aps=1` can be used, it's quite noisy.

>According to its documentation, the dimmer supports up to 78 entries in its neighbour table and up to 96 entries in its routing table. Could this be causing buffer overflows in deCONZ or in the RaspBee firmware?

Shouldn't be a problem, the RaspBee firmware doesn't process these and deCONZ doesn't have any limits. Also for the RaspBee the local neighbour table size is >150 :)

ebaauw, 2017-10-16:
Apparently the dimmer got stuck so badly, that it would no longer accept the power cycle sequence to reset.  After opening the wall switch and resetting the dimmer using the hardware button (hole), it paired with v2.04.82.

manup, 2017-10-16:
I've powered my D1 now, will report back if it also got stuck. I still wonder why it happens, because since 2.04.82 deCONZ won't poll lights when they support reporting.

Just found this GitHub link on ubisys homepage https://github.com/ubisys/SmartThings might be handy for further rintegration.

> As a side node, I would really like to be able to update the dimmer's firmware using deCONZ...

I've downloaded latest 2017 D1 OTA files from their homepage will do a test later on after doing some email stuff. Curios to get it working :)

manup, 2017-10-16:
The Update gets stuck at 0.76%, I've looked in older emails and found the problem was already investigated with ubisys. Problem is the ubisys device requests a larger chunk of data from OTA server than it can send (28 vs. 56 bytes), normally this shouldn't be an issue since the spec allows to send smaller chunks, but the (old) firmware doesn't support that.

There were RaspBee firmware versions recently which could send larger packages but these were replaced in favor of source routing firmware for IKEA and OSRAM.

I'll check to find an older firmware and if the updates work with it during the days.

ebaauw, 2017-10-20:
Been running .82 for some days now.  The GUI showed an occasional red node, but seemed to recover automatically.  Only now, my ubisys dimmer is lost again.  It shows red in the GUI, won't revive on pressing 0, won't revive on power-cycling the dimmer.  When powered, the connected lights are on, but blink off/on once every five seconds.  The log shows (filtered on 0x001f):
```
16:31:16:215 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 248 s
16:31:51:255 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 283 s
16:32:12:299 ZCL attribute report 0x001FEE0000001963 for cluster 0x0006
16:32:12:300 Node data 0x001fee0000001963 profileId: 0x0104, clusterId: 0x0006
16:32:12:310 update ZCL value 0x0006/0x0000 for 0x001FEE0000001963 after 306 s
16:32:13:701 ZCL attribute report 0x001FEE0000001963 for cluster 0x0008
16:32:13:701 Node data 0x001fee0000001963 profileId: 0x0104, clusterId: 0x0008
16:32:13:702 update ZCL value 0x0008/0x0000 for 0x001FEE0000001963 after 306 s
16:32:26:295 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 12 s
16:33:01:335 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 47 s
16:33:36:375 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 82 s
16:34:11:415 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 117 s
16:34:46:455 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 152 s
16:35:21:495 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 187 s
16:35:56:615 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 222 s
16:36:16:731 binding for cluster 0x0006 of 0x001FEE0000001963 exists (verified by reporting)
16:36:16:732 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x001FEE0000001963 (seems to be active)
16:36:16:732 binding for cluster 0x0008 of 0x001FEE0000001963 exists (verified by reporting)
16:36:16:732 skip configure report for cluster: 0x0008 attr: 0x0000 of node 0x001FEE0000001963 (seems to be active)
16:36:31:655 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 4 s
16:37:06:694 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 2 s
16:37:18:534 ZCL attribute report 0x001FEE0000001963 for cluster 0x0006
16:37:18:535 Node data 0x001fee0000001963 profileId: 0x0104, clusterId: 0x0006
16:37:18:540 update ZCL value 0x0006/0x0000 for 0x001FEE0000001963 after 306 s
16:37:19:970 ZCL attribute report 0x001FEE0000001963 for cluster 0x0008
16:37:19:971 Node data 0x001fee0000001963 profileId: 0x0104, clusterId: 0x0008
16:37:19:971 update ZCL value 0x0008/0x0000 for 0x001FEE0000001963 after 306 s
16:37:30:587 Poll APS request to 0x001FEE0000001963 cluster: 0x0006 dropped, values are fresh enough
16:37:41:737 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 11 s
16:38:16:775 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 46 s
16:38:51:903 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 81 s
16:39:26:943 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 116 s
16:40:02:079 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 151 s
16:40:24:731 binding for cluster 0x0006 of 0x001FEE0000001963 exists (verified by reporting)
16:40:24:732 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x001FEE0000001963 (seems to be active)
16:40:24:732 binding for cluster 0x0008 of 0x001FEE0000001963 exists (verified by reporting)
16:40:24:732 skip configure report for cluster: 0x0008 attr: 0x0000 of node 0x001FEE0000001963 (seems to be active)
16:40:37:118 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 186 s
16:41:12:158 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 221 s
16:41:47:199 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 256 s
16:42:18:406 Poll APS request to 0x001FEE0000001963 cluster: 0x0006 dropped, values are fresh enough
16:42:22:238 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 291 s
16:42:26:908 ZCL attribute report 0x001FEE0000001963 for cluster 0x0006
16:42:26:909 Node data 0x001fee0000001963 profileId: 0x0104, clusterId: 0x0006
16:42:26:919 0x001FEE0000001963 onOff 0 --> 1
16:42:26:919 update ZCL value 0x0006/0x0000 for 0x001FEE0000001963 after 308 s
16:42:27:623 ZCL attribute report 0x001FEE0000001963 for cluster 0x0008
16:42:27:623 Node data 0x001fee0000001963 profileId: 0x0104, clusterId: 0x0008
16:42:27:624 update ZCL value 0x0008/0x0000 for 0x001FEE0000001963 after 307 s
16:42:57:279 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 16 s
16:43:32:319 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 51 s
16:44:07:359 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 25 s
16:44:42:590 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 60 s
16:45:17:631 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 95 s
16:45:52:670 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 130 s
16:46:27:711 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 165 s
16:47:02:751 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 200 s
16:47:37:895 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 235 s
16:48:12:935 Node 0x001FEE0000001963 is known by 29 neighbors, last seen 270 s
16:48:25:737 Poll APS request to 0x001FEE0000001963 cluster: 0x0006 dropped, values are fresh enough
16:48:25:832 Poll APS request to 0x001FEE0000001963 cluster: 0x0008 dropped, values are fresh enough
16:48:47:975 Node 0x001FEE0000001963 is known by 27 neighbors, last seen 305 s
[... cut ...]
16:58:46:518 Node 0x001FEE0000001963 is known by 0 neighbors, last seen 0 s
16:59:21:558 Node 0x001FEE0000001963 is known by 0 neighbors, last seen 0 s
16:59:56:599 Node 0x001FEE0000001963 is known by 0 neighbors, last seen 0 s
17:00:31:638 Node 0x001FEE0000001963 is known by 0 neighbors, last seen 0 s
```

I don't understand where the max transmit errors come from.  I only control the dimmer through group commands: recalling a scene to switch the lights on, and setting `action.on` to `false` to turn if off.

manup, 2017-10-20:
>I don't understand where the max transmit errors come from.

The dimmer is polled by deCONZ core  and poll manager but at a very slow rate since attribute reporting is working.

    16:42:27:623 ZCL attribute report 0x001FEE0000001963 for cluster 0x0008

I don't know why the dimmer passed out, I don't think it's due polling since this should happen at a much slower rate than older deCONZ versions.

Maybe the source routing feature added for OSRAM and IKEA is problematic, perhaps the newer ubisys firmware will fix the problem (on it..)

manup, 2017-10-20:
New firmware in testing, ubisys dimmer OTA update runs but is very slow — 3 % in 5 minutes.

ebaauw, 2017-10-24:
> Apparently the dimmer got stuck so badly, that it would no longer accept the power cycle sequence to reset.

> I don't know why the dimmer passed out

It would seem my dimmer has serious issues.  This time it won't even react to the hardware button (hole).  When powered, the connected lights (and the LED next to the hole) are on for five seconds, then switch off briefly, then on again for five seconds... ad nauseatum.  I tried leaving the dimmer disconnected completely (including the N wire) overnight, but no change.

This sequence is not described in the technical reference - I'll try and contact ubisys support.

manup, 2017-10-29:
Did they have a solution, maybe reset the dimmer? I've also managed to update my D1-R  dimmer now, if you wanne give it a shot RaspBee/ConBee requires a different firmware version.

```bash
$ sudo GCFFlasher_internal -f deCONZ_Rpi_0x261a0500.bin.GCF
```

[deCONZ_Rpi_0x261a0500.bin.GCF.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1425255/deCONZ_Rpi_0x261a0500.bin.GCF.zip)

My dimmer didn't show any problems, but it's also not actively used other than being a permanent member of the network :)

ebaauw, 2017-10-29:
ubisys support have been very responsive, unfortunately unlike the dimmer itself.  I returned it for them to have a look at it.

manup, 2017-10-31:
Would be interesting to know what caused the state and if it can be prevented from the gateway side.

alexsporn, 2017-11-05:
@manup I created a network to update a friends Tradfri bulb to be Hue compatible and it always got stuck at 0.04% (using 0x26190500 firmware and 2.04.84 deCONZ). Tried it multiple times, power-cycled both Raspbee and the bulb and nothing helped. Then I tried the 0x261a0500 firmware you posted here and now it's updating blazingly fast on the first try!

manup, 2017-11-05:
Yes 0x261a0500 allows greater messages to be send which is especially useful for OTA updates. 0x26190500 will be replaced in the next versions. :)

DOliana, 2017-11-12:
Is there also a firmware version for the ConBee stick? I am trying to update a Tradfri remote and it's been stuck at 0,04% for hours...

manup, 2017-11-12:
You can use the firmware mentioned above `deCONZ_Rpi_0x261a0500.bin.GCF.zip` to workaround the problem. It will be included in the next deCONZ versions.

DOliana, 2017-11-12:
Thanks! I tried it already, but it didn't work because I tried to use the GCFFlasher_internal tool - I had to use the GCFFlasher tool and it worked immediately. I will now try the update again.

manup, 2017-11-12:
GCFFlasher_internal should work, note it is important that deCONZ is not running while updating the firmware.

DOliana, 2017-11-12:
Hmm - deCONZ was not running. It told me that it couldn't find the device. Also "-l" showed no devices, whereas GCFFlasher worked immediately without any issue.

snozzlebert, 2017-11-13:
@DOliana did you manage to update the firmware and upgrade your IKEA remote?
The OTA update of my Tradfri motion sensor is stuck on 0.04% (still on Conbee 0x26190500)

manup, 2017-11-13:
Sorry update was messed up for 0x26190500 (packages are too large which is only supported with firware 0x261a0500 from above).

Will be fixed in next release 2.04.88 so that 0x26190500 does work again.

DOliana, 2017-11-14:
@snozzlebert yes. I updated the firmware using GCFFlasher instead of GCFFlasher_internal. After that it took some time before the update started (had to press update in deCONZ / on the remote a few times). As soon as it started it was done in about 45 minutes.

ebaauw, 2017-11-14:
> Did they have a solution, maybe reset the dimmer? I've also managed to update my D1-R dimmer now, if you wanne give it a shot RaspBee/ConBee requires a different firmware version.

Just got the dimmer back with a cryptic note that they reset and “slightly refurbished” the dimmer under warrenty.  Haven’t installed in yet.

manup, 2017-11-14:
I guess they might have erased memory and programmed the firmware again. If no hardware defects are present this is the most simple approach to get a device alive again, but sadly delivers not much debug data.

snozzlebert, 2017-11-18:
> The OTA update of my Tradfri motion sensor is stuck on 0.04% (still on Conbee 0x26190500)

After updating to 0x261a0500 updating the Tradfri motion sensor succeeded after 45 minutes.

ebaauw, 2017-11-18:
Installed the dimmer and paired it to deCONZ (v2.04.89).  It still has the same MAC address.  They did update the firmware.  The Date Code is now `20171108-DE-FB0`, still no SW Build ID.  The OTAU plugin reports version 0x010E015E and image 0x7B01.  This is the version they published (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/96#issuecomment-321924364).

I had some problems recreating the groups and scenes, but it took multiple attempts before as well, storing the scenes from the deCONZ GUI instead of from the REST API.  Otherwise, it seems to work as before.  Fingers crossed.

ebaauw, 2017-12-15:
Haven't seen any issues with the Ubisys dimmer for almost a month (knock on wood).  Closing the issue...

ebaauw, 2017-12-17:
And that jinxed it.  Last night, my ZigBee network broke down again.  The RaspBee was completely incommunicado, having lost all routers (except the ubisys) and most end devices.  Restarting deCONZ, resetting the RaspBee, power-cycling the Raspberry didn't do anything.  The only remedy is powering down all routers, re-starting deCONZ, powering up the routers in small batches.  All routers and end devices are back, except the ubisys.

manup, 2017-12-17:
Damn I wonder what happens here :/ if the ubisys is still missing would it be possible to sniff the other channels for > 30 seconds to see if it is on a different channel?

Relevant channels: 11, 15, 20, 25.

After a power-cycle devices scan all channels to see if the network has wandered usually indicated by higher `network update id`.

Later on it would be possible for coordinator to go over all channels and see if devices are wrongly there, bringing them back. Still I wonder why such things happen in first place.

CDRX2, 2018-01-20:
Hi everyone,

Been following this thread these past few weeks and have recently decided to take the plunge: bought a D1 and C4 from Ubisys. Seeing the option of adding Ubisys switches in the (old) webapp helped as well ;-)

Just installed the C4 today and have two issues. Though the C4 shows up in deCONz (currently 2.04.99), the webapp does not find it when I search for it. Tried resetting it, still the same. Figured I would try to do a firmware update, got the file, update is stuck at various values below 0.77%. Remembered your conversation here, downloaded the firmware version from @manup earlier in this thread, not working either.

@ebaauw Did you add the "switch" of the D1 via the webapp or via REST? I'm not exactly proficient neither with code nor with deCONz, so I would appreciate any help possible.

ebaauw, 2018-01-20:
The C4 is not yet whitelisted, so it won’t appear as sensor (switch) in the REST API.  If you post some screenshots, showing the node, incl. the enpoints and cluster as well as the details of the basic cluster, I could give it a try.  Alternatively, can you point me to the documentation, the D1’s manual is very comprehensive.

I’ve never updated the firmware of my D1 through deCONZ.  Indeed it would hang a something like 0.77%.  It locked up and I had to send it to ubisys for a factory reset.  When they returned it, it had the latest firmware.

You should add all devices through the web app, not through the deCONZ GUI, or the REST API won’t see them.  The web app uses the REST API to talk to deCONZ, so there’s no difference there.

CDRX2, 2018-01-20:
> The C4 is not yet whitelisted, so it won’t appear as sensor (switch) in the REST API.

Did not know that, explains the fact I can't find it.

> If you post some screenshots, showing the node, incl. the enpoints and cluster as well as the details of the basic cluster, I could give it a try.

Sure, thank you very much! Here are the screenshots:
[image]
[image]
Sorry for the Frankeinstein aspect, resolution issues, would have been unreadable.

> Will you be using it to control lights or window coverings?

I plan on using it to control lights, only using the first and second sensors, 3 and 4 are unused.

> I’ve never updated the firmware of my D1 through deCONZ. Indeed it would hang a something like 0.77%. 

Hope it will work in time, then. I guess it's in Manuel's hands.

> You should add all devices through the web app, not through the deCONZ GUI, or the REST API won’t see them. The web app uses the REST API to talk to deCONZ, so there’s no difference there.

Did not know that either, very interesting piece of information. I did not, however add the device through deCONZ, it juste showed up, figured therefore there might be some way of adding devices through deCONZ GUI. But since then it won't work on the web app, that's of no use.

ebaauw, 2018-01-21:
I think I added support for the C4, but obviously cannot test it.  Could you please compile the REST API plugin from my latest commit (see above) and test it?  The C4 should be exposed as four sensor resources, one for each endpoint / input / button.  The deCONZ web app might support it, sometimes it seems to like my D1.
[image]

>  there might be some way of adding devices through deCONZ GUI

Edit | Permit Join.  In the web app it's Settings | Open Network.

CDRX2, 2018-01-21:
>  Could you please compile the REST API plugin from my latest commit (see above) and test it?

Been trying it out for a short while now. I am able to add the sensor as follows:
[image]
Which gives me:
[image]
But I noticed that there is no mention of the endpoint parameters as in the examples in the REST documentation.

Added rules as well, here's one example:
[image]
But it is not triggering via the sensor. Any chance this is related to the missing ep parameter?

> The deCONZ web app might support it, sometimes it seems to like my D1.

Nope, the web app is not finding the C4. I did hook up my D1 (not yet installed) via 12V to try and add it to the network and it worked flawlessly right away. I should mention I tried it directly in the old web app.

ebaauw, 2018-01-21:
You shouldn’t add any sensor resources, deCONZ should create them, just as for the D1.  Maybe you need to reset the C4 and re-pair it, opening the network from the web app.  Before that, double-check that you’re running the compiled plugin.  It should report `config.apiversion` as 1.0.6 in the REST API.

CDRX2, 2018-01-21:
Progress! I was able to add the C4 via the web app after resetting it. I am seeing 6 endpoints which is puzzling:
[image]
However, when lights is selected, I cannot open the dropdown list to select which light. When I select group, I'm able to assign a group to the button and then apply. Coming back to this screen, everything is blank again.

ebaauw, 2018-01-21:
Can you post what the sensor resources look like?  I’ll probably need to exclude the endpoints for controlling the window blinds.

I haven’t been able to configure my D1 through the web app, but you can program any action in rules based on the sensor resource’s `state.buttonevent`.  Also, I managed to create a binding manually in the deCONZ GUI, from the dimmer’s _OnOff_ and _Level_ clusters to a group.  After that, the groups showed in the web app.

CDRX2, 2018-01-21:
Just did a reset and re-added the C4. Was able to assign lights to a button, but they still won't show in the web app as above.

Went to the REST client and poked around. Found the following for sensors (keeping in mind I only have the C4 on the network):
{
    "1": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "50cfc78edc199f7c3a7680d4fc0c65e0",
        "manufacturername": "ubisys",
        "mode": 1,
        "modelid": "C4 (5504)",
        "name": "Chambre parentale",
        "state": {},
        "swversion": "20170728-DE-FB0",
        "type": "ZHASwitch",
        "uniqueid": "00:1f:ee:00:00:00:2a:fb-01-0006"
    },
    "2": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 2,
        "etag": "5c3b2e5ba1d79d4c605f463511b2ee7b",
        "manufacturername": "ubisys",
        "mode": 1,
        "modelid": "C4 (5504)",
        "name": "C4 (5504) 2",
        "state": {},
        "swversion": "20170728-DE-FB0",
        "type": "ZHASwitch",
        "uniqueid": "00:1f:ee:00:00:00:2a:fb-02-0006"
    },
    "3": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 3,
        "etag": "5c3b2e5ba1d79d4c605f463511b2ee7b",
        "manufacturername": "ubisys",
        "mode": 1,
        "modelid": "C4 (5504)",
        "name": "C4 (5504) 3",
        "state": {},
        "swversion": "20170728-DE-FB0",
        "type": "ZHASwitch",
        "uniqueid": "00:1f:ee:00:00:00:2a:fb-03-0006"
    },
    "4": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 4,
        "etag": "5c3b2e5ba1d79d4c605f463511b2ee7b",
        "manufacturername": "ubisys",
        "mode": 1,
        "modelid": "C4 (5504)",
        "name": "C4 (5504) 4",
        "state": {},
        "swversion": "20170728-DE-FB0",
        "type": "ZHASwitch",
        "uniqueid": "00:1f:ee:00:00:00:2a:fb-04-0006"
    },
    "11": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 5,
        "etag": "5c3b2e5ba1d79d4c605f463511b2ee7b",
        "manufacturername": "ubisys",
        "mode": 1,
        "modelid": "C4 (5504)",
        "name": "C4 (5504) 11",
        "state": {},
        "swversion": "20170728-DE-FB0",
        "type": "ZHASwitch",
        "uniqueid": "00:1f:ee:00:00:00:2a:fb-05"
    },
    "12": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 6,
        "etag": "5c3b2e5ba1d79d4c605f463511b2ee7b",
        "manufacturername": "ubisys",
        "mode": 1,
        "modelid": "C4 (5504)",
        "name": "C4 (5504) 12",
        "state": {},
        "swversion": "20170728-DE-FB0",
        "type": "ZHASwitch",
        "uniqueid": "00:1f:ee:00:00:00:2a:fb-06"
    }
}

Checked on the rules as well:
{
    "1": {
        "actions": [
            {
                "address": "/lights/16/state",
                "body": {
                    "on": true
                },
                "method": "BIND"
            }
        ],
        "conditions": [
            {
                "address": "/sensors/1/state/buttonevent",
                "operator": "eq",
                "value": "1"
            }
        ],
        "created": "2018-01-21T16:51:45",
        "etag": "50cfc78edc199f7c3a7680d4fc0c65e0",
        "lasttriggered": "none",
        "name": "Sensor: 1 EP:01 On/Off",
        "owner": "Censored",
        "periodic": 0,
        "status": "enabled",
        "timestriggered": 0
    },
    "2": {
        "actions": [
            {
                "address": "/lights/14/state",
                "body": {
                    "on": true
                },
                "method": "BIND"
            }
        ],
        "conditions": [
            {
                "address": "/sensors/2/state/buttonevent",
                "operator": "eq",
                "value": "2"
            }
        ],
        "created": "2018-01-21T16:51:53",
        "etag": "f1138f9096c8865992034ecdbc665af3",
        "lasttriggered": "none",
        "name": "Sensor: 2 EP:02 On/Off",
        "owner": "censored",
        "periodic": 0,
        "status": "enabled",
        "timestriggered": 0
    }
}
 
I noticed that, when I added my rules manually earlier, the "owner" was my API key, but this time it isn't. Any chance that plays a role?

CDRX2, 2018-01-21:
Quick update: did the bindings between endpoint 1 & endpoint 2 and the two lights I want to control in deCONZ as you suggested. Nothing shows up in the web app, but pressing on the switch twice will trigger each event. Only weird thing: I have to press twice...

ebaauw, 2018-01-21:
If they work like the D1’s inputs, a press/release should toggle the light and a press/hold should alternate between dim up and dim down.  That is, of course, if the light supports the corresponding commands.  And if you created a binding for the _Level_ cluster for the dim up/dim down.

Could you check (easiest is through the web socket) what buttonevent values are generated?  It should be a x002 for press/release, and a x001 for hold and x003 for release after hold.  X=1,2,3,4 correponding to the input.

I never fully understood or used the BIND rules.  They don’t exist on the Hue bridge.  If you ceate the bindings in the deCONZ GUI, you don’t need them, as far as I can tell.  The rule `owner` is the username (API key) used to create the rule.  If it’s a different value, maybe deCONZ (or the web app) created the rule itself.

CDRX2, 2018-01-21:
> Could you check (easiest is through the web socket) what buttonevent values are generated? It should be a x002 for press/release, and a x001 for hold and x003 for release after hold. X=1,2,3,4 correponding to the input.

Well, the output of the websocket seems to not want to give me the buttonevent values:
[image]
This is the result of pressing and releasing each of the two buttons four times. Twice to turn the light on, twice to turn it off.

Tried binding level control of the lights (FLS-PP3) to the according cluster, but pressing and holding does nothing. This, however is just an FYI, not something I want to pursue in this case since I won't be needing dimming functionalities.

> The rule owner is the username (API key) used to create the rule. If it’s a different value, maybe deCONZ (or the web app) created the rule itself.

Yes, I checked on this and it was indeed the web app that created the rule.

ebaauw, 2018-01-21:
> I’ll probably need to exclude the endpoints for controlling the window blinds.
> Found the following for sensors (keeping in mind I only have the C4 on the network)

Yeah, that's the Windows Covering Controller endpoints alright.  My latest commit should no longer create ZHASwitch sensor resources for these (you probably need to reset and re-pair the C4 again, after recompiling the plugin).

> Well, the output of the websocket seems to not want to give me the buttonevent values

Bummer.  That probably means the C4 sends different ZigBee commands than the D1, at least in its current configuration.  I take it, you didn't try and change the C4 configuration (see #360)?

Without ZigBee sniffing, this is going to be trial and error, I'm afraid.

> Nothing shows up in the web app, but pressing on the switch twice will trigger each event. 

What do you mean by "event"?  The light going on or off?

> Only weird thing: I have to press twice...

You do have a stateless "pulse" switches connected to the inputs, don't you?  If you connect a regular light to the switch, it only turns on while you're pressing/holding the switch?  
Or is it a stateful switch, one press/release to turn a regular light on, another press/release to turn it off.

CDRX2, 2018-01-22:
> My latest commit should no longer create ZHASwitch sensor resources for these (you probably need to reset and re-pair the C4 again, after recompiling the plugin).

Indeed, the Window Covering Controller endpoints are now gone in the web app.

> I take it, you didn't try and change the C4 configuration (see #360)?

No, I did not attempt this. I will give it a try sometime this week and report back.

> What do you mean by "event"? The light going on or off?

Exactly. I don't see that there are lights assigned to buttons in the web app, but pressing the buttons twice turns the light on or of.

> You do have a stateless "pulse" switches connected to the inputs, don't you? If you connect a regular light to the switch, it only turns on while you're pressing/holding the switch?
Or is it a stateful switch, one press/release to turn a regular light on, another press/release to turn it off.

I'm not an expert, but based on what I've read and seen, my switches (pre-existing, I'm upgrading) are stateful. Could that have an influence on this pressing twice business?

ebaauw, 2018-01-22:
Yes, by default, the D1 is configured for pulse inputs - I guess the C4 is as well.  The unit reacts to power dropping from the input.  With a stateful switch, the first press/release puts power on the input and the second press/release removes it, so the unit only reacts to the second press/release.  With a stateless switch, press puts power on the input, and release removes it again.

I think the web app expects the endpoint to be bound to a group and won’t show a single light.  Also, if the light and the switch are close to each other and further from the RaspBee, deCONZ might not see the command, causing no `buttonevent`.  Could you try and bind the C4 to a group instead, and assign the light to that group?  Group commands are broadcasted repeatedly through the network, so the RaspBee will pick them up.

CDRX2, 2018-01-23:
> Yes, by default, the D1 is configured for pulse inputs - I guess the C4 is as well. The unit reacts to power dropping from the input. With a stateful switch, the first press/release puts power on [...]

Ok, so basically, I need to get stateless switches, correct?

> Also, if the light and the switch are close to each other and further from the RaspBee, deCONZ might not see the command, causing no buttonevent.

Definitely the case in my configuration. RaspBee is on the ground level, the C4 and the lights it controls are upstairs.

> Could you try and bind the C4 to a group instead, and assign the light to that group? Group commands are broadcasted repeatedly through the network, so the RaspBee will pick them up.

Indeed, RaspBee did pick the group command up along with `buttonevent`.
[image]

ebaauw, 2018-01-23:
> Ok, so basically, I need to get stateless switches, correct?

That would be the path of least resistance.  Alternatively, you could re-configure the C4 and tell it there's stateful switches connected to the inputs, but we haven't figured out how to do that with deCONZ (other than writing your own plugin), see #360.

> Indeed, RaspBee did pick the group command up along with `buttonevent`.

Cool.  If you bind the _Level Control_ cluster to the group as well, you should also get `buttonevent` values 1001 (hold) and 1003 (release after hold).  I'd expect, with the stateful switch connected, you'll get a 1001 on the first press/release (when power comes on), and a 1003 on the second (when power goes off).  You'll only get a 1002 when pressing/releasing twice in quick succession (so the C4 sees a pulse).

CDRX2, 2018-01-23:
> That would be the path of least resistance. Alternatively, you could re-configure the C4 and tell it there's stateful switches connected to the inputs, but we haven't figured out how to do that with deCONZ (other than writing your own plugin), see #360.

Well, I'm planning on upgrading my entire home and just did a quick calculation. Changing the switches would be more expensive than getting the Ubisys gateway to change this without plugin. I guess I will live with double pressing for now, will see how #360 and/or deCONZ evolves. There are some news in the works, as I've read multiple times.

>  I'd expect, with the stateful switch connected, you'll get a 1001 on the first press/release (when power comes on), and a 1003 on the second (when power goes off).

Unfortunately, I did not have any success with this on the C4. Still only get a 1002 `buttonevent`. Tried the same on the D1, get a 1003 and then a 1001. However, holding the button has no influence on the dimming values, always ends with turning the lights on/off fully.

I should point out that I just upgraded to 2.05.00.

ebaauw, 2018-01-23:
> Unfortunately, I did not have any success with this on the C4. Still only get a 1002 buttonevent.

Did you create a binding to a group for the _Level Control_ cluster?  Note that the (traditional) light(s) connected to the C4 output won't dim.

> However, holding the button has no influence on the dimming values, always ends with turning the lights on/off fully.

You're using stateful switches, they probably only change state only on release - holding the key doesn't do anything.  Hook up a traditional light to switch to understand how it works: one press/release to turn the light on (i.e. put power on the C4 input), another one to turn the light off (i.e. remove power from the C4 input).  
A stateless switch on the other hand only provides power while you hold the button, very much like a doorbell.  A press/release generates a brief power-on/power-off pulse; a hold provides power for a longer time.  You don't use these to control a light directly, but to control a timer (typical staircase use) or a stateful electronic switch that controls the light.  
To simulate a pulse with your stateful switch, you need to press/release once to turn the power on, followed immediately by a second press/release to turn the power back off.  This should generate a 1002.  To simulate a hold, you need to wait after the first press/release (that turns power on) for the 1001, then time the second press/release, like 0.5 to 2 sec after the first, for the 1003.  If you just press/release once, the power remains on too long, causing the C4 to think you're holding forever, so it dims all the way up or down, turning the light on or off.

> Changing the switches would be more expensive than getting the Ubisys gateway to change this without plugin.

I don't think you're looking at the right switches.  You'd need a simple mechanical switch that emits a pulse, typically under EUR 10, not an electronic one that reacts to a pulse, typically well over EUR 50.

CDRX2, 2018-01-24:
> Did you create a binding to a group for the Level Control cluster? Note that the (traditional) light(s) connected to the C4 output won't dim.

Yes I did. No dice on the 1001 and 1003 `buttonevent`.

> To simulate a hold, you need to wait after the first press/release (that turns power on) for the 1001, then time the second press/release, like 0.5 to 2 sec after the first, for the 1003.

Exactly what I attempted and it's pretty much what I got with the D1, but not with the C4. But even when seeing the different `buttonevent`. Holding the second press did not do anything with the dimming either.

> I don't think you're looking at the right switches. You'd need a simple mechanical switch that emits a pulse, typically under EUR 10, not an electronic one that reacts to a pulse, typically well over EUR 50.

Well, the thing is, I live in Switzerland and there's basically a monopoly of one brand for switches (whichever kind you want). Mostly due to our funny Typ 12 & Typ 13 plugs. Add to this the fact that certain products are more expensive to buy here and I end up with a basic, single Schema 3 stateful switch costing around EUR 25 and a 3-way in-wall outlet costing about EUR 42. So it actually does not surprise me that a simple combination of a dual mechanical switch with one power outlet that is compatible with said manufacturer's program (e.g. built by them) is actually quite expensive.

For the reference, here is the link to said switch on the manufacturer's website (sorry, they don't have an englisch website): [https://online-katalog.feller.ch/kat_details.php?fnr=87063.AR63AR.FMI.61](url) and here's the price of it on a reseller's website: [https://www.elektrogut.ch/em/kleinkombinationen/ediziodue/unterputz-up/taster_steckdosen/baukasten/ediziodue-einsatz-kleinkombination22.html](url). To be fair, I should mention that the price is only for the insert module and its small covers, no VAT and no shipping. Add to that the aluminium frame behind it for mounting and the cosmetic frame (already have those, fortunately) and then you've a complete switch. Don't yet know what I will do, but for now I will live with the double-tap on my switches...

CDRX2, 2018-04-08:
Just wanted to give a short update. I have bought a stateless switch and just replaced it today in combination with my C4. Works like a charm, pushing once turns the light on, pushing a second time turns it off. Thanks for the help @ebaauw. Now I just have to invest for the rest of my home...

ebaauw, 2018-04-08:
As we’re giving updates.  I haven’t had any issues with my D1 since ubisys support updated its firmware.  Knock on wood.

closed 2018-04-08 by ebaauw