dresden-elektronik/deconz-rest-plugin #231: Paulmann LED-Controller

AntiHeld889, 2017-10-16:
A friend of me soon wants to change also on the Raspbee module, however, we wanted to look first whether his available LED controller also works. But there are no resource(s) in the REST API.

[image]

manup, 2017-10-16:
Interesting haven't seen these yet :)
Should work in 2.04.83

AntiHeld889, 2017-10-26:
working

beatpaul, 2017-10-28:
Hi there! following this conversation and was really happy to get the new version 2.04.83 (84), but my paulmann SmartHome ZigBee Cephei (https://de.paulmann.com/leuchten/smart-home/zigbee/smarthome-zigbee-cephei-schalt-controller-max.-1000w-ac/50043) is still not recognized in the rest plugin.
I
[image]
Is there a way to fix this ?

thx, Paul

manup, 2017-10-29:
Should be simple to fix, can you please provide a screenshot from the endpoint–cluster list, as seen in the screenshot above (click on the right circle of the node).

beatpaul, 2017-10-29:
Thx for the quick answer, here is the screenshot. if u nee any further info let me know!
[image]

manup, 2017-10-29:
Thanks, should work in next version 2.04.85.

beatpaul, 2017-10-31:
@manup Hi again! I have one more Question: I just realized, that the device i asked to integrate is a switch not the dimmable version, will the dimmable version also work in version 2.04.85  (i'm not e programmer, but i red something like "DEV_ID_Z30_DIMMABLE_PLUGIN_UNIT:" in the code :) ?
I'm thinking of ordering / changing it if this is or would be possibel.
Grezz! and thanx for your support!

manup, 2017-10-31:
I've added both, so they *should* work but it's not tested yet :)

beatpaul, 2017-10-31:
Perfect! I just ordered the dimmable device -> i‘ll give u a feedback for both devices as soon as i can test them :)

beatpaul, 2017-11-02:
@manup: the Paulmann switching device worked flawlessly! (tested with Deconz and Rest Plugin + homebridge) --> thx
As soon as i get and test my dimming device i let u know the result!

beatpaul, 2017-11-04:
... so luckily the dimmable device was shipped fast and I already installed it ... works perfektly ... so i guess this issue is solved !

thx again for this fast support!

closed 2017-11-10 by AntiHeld889

beatpaul, 2019-02-14:
Hi Manup!

Im really happy with the deconz system and thx again for including the paulmann dimmable device (see this issue)
The problem i have is that just this light switches on randomly some times during day or night. That happend since i installed it, but first i didnt care to much ...
It would be very kind if you could check the code once again

Thx in advance !!
Beatpaul

manup, 2019-02-15:
Hard to tell what's going on. deCONZ doesn't turn on lights on it's own.

Can you enable logging and provide that log when the light has turned on by its own:

    deCONZ --dbg-aps=1 --dbg-info=1 --dbg-http=80 > log.txt

beatpaul, 2019-02-17:
Hi @manup !

Here ist the log.  
The Paulmann LED Controller turned on at time

- 20:36 and
- 20:46  
(+/- 1 minute).

Short Log (I deleted some Parts, because of data size > 190 MB):
[https://www.dropbox.com/s/69z2ecbv2kshex8/log%20short.txt?dl=0 
](url)

Full Log:
[https://www.dropbox.com/s/c6fj1rs7zwyvmvi/log.txt?dl=0](url)

Im Using deCONZ in combination with the homebridge-hue (@ebaauw )plugin.

If you need further information let me know.
Thx in Advance fro your help.

Greez Paul

beatpaul, 2019-02-27:
Hi again!
@manup: any idea about the log?