dresden-elektronik/deconz-rest-plugin #232: Request: Lutron Connected Bulb Remote Integration.

mysteryan, 2017-10-16:
Any chance of the Lutron dimmer remote getting better integration?  It doesn't show up properly in the web gui and receives errors to rejoin it when attempting to create a dimmer group for it.  It successfully creates a group under 2.0.35 though.  Despite that, it works just like my Hue Dimmer.

Its model is: Lutron LZL-4B-WH-L01
Cluster info (grabbed from a smartthings binding):
fingerprint profileId: "C05E", deviceId: "0820", inClusters: "0000,1000,FF00,FC44", outClusters: "0000,0003,0004,0005,0006,0008,1000,FF00"

mysteryan, 2017-11-05:
Here's more info.  The device works fine when I bind it to a group, but it doesn't receive any proper state/buttonevents.  It's commonly found in US retail stores like HomeDepot and it's the only zigbee remote that seamlessly fits a 1-gang traditional light switch hole.  Product information can be found here: https://www.amazon.com/Lutron-LZL-4B-WH-L01-Connected-Bulb-Remote/dp/B014STZASK/

Heres my APS log for the remote on deCONZ 2.04.86 on rPi3/Stretch:
no button map for: LZL4BWHL01 Remote cl: 0x0008 cmd: 0x04 pl[0]: 0FE  //On Press
no button map for: LZL4BWHL01 Remote cl: 0x0008 cmd: 0x06 pl[0]: 000  //Dim up?
no button map for: LZL4BWHL01 Remote cl: 0x0008 cmd: 0x03 pl[0]: 000  //Dim up release?
no button map for: LZL4BWHL01 Remote cl: 0x0008 cmd: 0x02 pl[0]: 001  //Dim down?
no button map for: LZL4BWHL01 Remote cl: 0x0008 cmd: 0x03 pl[0]: 000  //Dim down release?
no button map for: LZL4BWHL01 Remote cl: 0x0008 cmd: 0x04 pl[0]: 000  //Off Press

jonathanweinberg, 2018-01-31:
I would absolutely love to see this happen. Is there the need for a developer to have physical equipment to help this along?

manup, 2018-01-31:
Based on the logs and data from @mysteryan we should be able to process some of the button events, I figure on/off cluster should be added too. It would be interesting to see some more screenshots of the device clusters, maybe they implemented some extra stuff like Philips remote does via manufacturer specific cluster.

mysteryan, 2018-01-31:
Bottom two (dining + kitchen) nodes with expanded clusters are the Lutron remote dimmers.  For comparison, the top one is a Phillips Hue dimmer.
[image]

ebaauw, 2018-01-31:
> I figure on/off cluster should be added too.

Don’t think so, looks like they send a move to level with on/off to turn the light on or off.

mysteryan, 2018-02-01:
Don't really have much in programming skills, but I did try to modify the modelid and manufacturer in the database in hopes that the endpoints were similar enough to fire off events, but still nothing.  It did expose them in the web gui as the sensor I changed them to, but only seemed to be aesthetic and non-functional.  I've successfully manually binded every cluster to the gateway and not even a change in "lastupdated" value would show up.  Despite all that, it can still control the group of lights that I've manually binded them to in deconz.

mysteryan, 2018-02-01:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

OzzieJacks, 2018-02-06:
I was about to open a support request for this remote as well but it looks like mysteryan beat me to it (Thanks for posting all the info).  

With a Wink hub it is possible to configure this remote to act as either a dimmer or to assign user defined functions to each of the four buttons.  I don't have a wink hub but if the developers think it would help I could grab one and try to sniff out the communication between the remote and the Wink hub to see if Wink sends anything to the manufacturer specific clusters.

Link to a smartthings thread discussing this remote https://community.smartthings.com/t/lutron-connected-bulb-remote/20825/80

Link to the product manual http://www.lutron.com/TechnicalDocumentLibrary/040421_Zigbee_Programming_Guide.pdf

manup, 2018-02-06:
Thanks guys the data should help to add a first prototype implementation in the next releases.
For the events we'll create a binding from 0x0008 `level control cluster` to a newly created device group, so the remote will also work when the gateway is offline.

mysteryan, 2018-02-17:
Not sure if this is relevant, but the "Model Identifier" is usually reported back as "LZL-4B-WH-L01 Remote", but it's not that uncommon for it to report back as "LZL4BWHL01 Remote" at times.  Not sure why the dashes get dropped sometimes.  Could be interference or the differences the network meshes itself since I tend to update to every new commit as they come in.

jonathanweinberg, 2018-02-17:
@OzzieJacks I have a Wink Hub and Lutron remotes if you think you might be able to get oriented in sniffing the traffic.

mysteryan, 2018-03-17:
@manup Just wanted to give a followup.  Didn't realize that the devices needed to be re-paired to the gateway for (I think) commit #ca381006cdd85987d4f8cf43b34e0a489429ab17 to be effective.

Although, these devices aren't exposed in the web gui (including pwa), I consider this issue resolved.  Rest api now reports buttonevents with their corresponding lastupdated times for these Lutron remotes.  It reports buttonevent "1002" for On, "2001" for DimUp-Press, "2003" for DimUp-Release, "3001" for DimDown-Press, "3003" for DimDown-Release and "4002" for Off.

manup, 2018-03-18:
Cool glad to hear, visual implementation will follow once the new switch editor arrives.

closed 2018-03-18 by manup