dresden-elektronik/deconz-rest-plugin #235: Are Osram Lightify Motion Sensor and Lightify Switch (4 button) supported by the deCONZ REST API plugin ?

oleg77s, 2017-10-19:
Osram Lightify Motion Sensor and Lightify Switch (4 buttons) in deCONZ:
[image]
[image]

Are Osram Lightify Motion Sensor and Lightify Switch (4 button) supported by the deCONZ REST API plugin ?

manup, 2017-10-19:
The motion sensor is supported in the REST API, you might have to 

1) start sensor search (open network in the webapp);
2) power cycle the sensor by pull out battery press side button, put battery in again

It kinda works but in my short tests the sensor didn't work reliable compared to hue one.

The switch isn't supported currently.

oleg77s, 2017-10-20:
Ok, 
Thanks

closed 2017-10-20 by oleg77s