dresden-elektronik/deconz-rest-plugin #237: v2.04.82 WebSocket first event not sent

HerrHofrat, 2017-10-20:
After opening a connection and triggering a ZHAOpenClose sensor event for example, I do not receive a websocket message. 

Tested workflow
1. Precondition: Sensor is closed
2. Opening websocket connection
3. Open sensor
4. No message for the open event is received
5. Close senor
6. Message for the close event is received

However, it seems that the gateway itself receives the event - but does not broadcast it over the websocket:
1. Precondition: Sensor is closed
2. Opening websocket connection
3. Open sensor
4. No message for the open event is received
5. Check Sensor status over REST Api
6. Sensor status is open

All tested with my small config tool that can be found [here](https://github.com/HerrHofrat/deconz-config) and with a python implementation for home-assistant, that I'm currently working on.

Am I missing anything? Can I notify the gateway somehow to reload the websocket connections?

ebaauw, 2017-10-20:
Is this after restarting deCONZ, or every time when you close and reconnect the web socket client?  
What sensor state is reported by the REST API when the sensor is closed (before you open the websocket connection)?  Is `state.lastupdated` changed after you open the sensor?

HerrHofrat, 2017-10-20:
deCONZ is running all the time - only the websocket client is closed and reconnected.

The state in the REST API and lastupdated is always correct. Also, the websocket message are also working well - only the first state change after the connect is missing. 

state when it is closed:

        "2": {
            "config": {
                "on": true,
                "reachable": true
            },
            "ep": 1,
            "etag": "567d08ddb1f824a3a4634ef58d337bac",
            "manufacturername": "LUMI",
            "modelid": "lumi.sensor_magnet.aq2",
            "name": "Xiaomi Door Sensor 1",
            "state": {
                "lastupdated": "2017-10-20T15:00:39",
                "open": false
            },
            "swversion": "3000-0001",
            "type": "ZHAOpenClose",
            "uniqueid": "00:15:8d:00:01:ab:37:91-01-0006"
        },

state when it is opened

        "2": {
            "config": {
                "on": true,
                "reachable": true
            },
            "ep": 1,
            "etag": "9b89cc02d04edac78e2ca0a2428cc658",
            "manufacturername": "LUMI",
            "modelid": "lumi.sensor_magnet.aq2",
            "name": "Xiaomi Door Sensor 1",
            "state": {
                "lastupdated": "2017-10-20T15:02:10",
                "open": true
            },
            "swversion": "3000-0001",
            "type": "ZHAOpenClose",
            "uniqueid": "00:15:8d:00:01:ab:37:91-01-0006"
        },

snozzlebert, 2017-10-20:
I've the same issue since I started using deCONZ on version 2.04.54.
The bug is related to the websocket connection, not a specific sensor (or restarting deCONZ). When is press my Xiaomi button after a new websocket connection I only get one event (1002):

```
> $ wscat --connect http://deconz:8080
connected (press CTRL+C to quit)
< {"e":"changed","id":"91","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2017-10-20T19:04:13"},"t":"event"}
> 
```

When I press the button again, I do get both press and release events:
```
< {"e":"changed","id":"91","r":"sensors","state":{"buttonevent":1000,"lastupdated":"2017-10-20T19:06:17"},"t":"event"}
< {"e":"changed","id":"91","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2017-10-20T19:06:17"},"t":"event"}
```

Kane610, 2017-11-02:
I was just about to write a ticket about this as well. I've noticed this with lights.

manup, 2017-11-02:
I can confirm the issue, the first event won't bubble up. I have some ideas to resolve the problem during the next days.

HerrHofrat, 2017-11-07:
I have tested the latest beta (2.04.86) and it seems that the bug is fixed! Thanks!

snozzlebert, 2017-11-13:
I can confirm that it's fixed in .86 
Thanks!

closed 2017-11-20 by HerrHofrat