dresden-elektronik/deconz-rest-plugin #238: Getting deCONZ Command Line Parameters

tgroeneveldlg, 2017-10-20:
Hi @manup,

Do you have an overview of all Command-Line parameters for deCONZ startup ?
I am particularly interested in, specifying the WS port, which seems to default to 8080.

On my system, port 8080 is already in use, and it seems to fallback to 8088.
It would be nice just to specify the port instead.

CLI options for deCONZ are not in the PDF's I could find so far.

Thanks.

Kind regards, Tom.

IcarusR, 2017-10-22:
Hi
You can specify alternate port like this...
`/usr/bin/deCONZ  --http-port=8090`
Dont know about any other cli perameters.

tgroeneveldlg, 2017-10-22:
Hi @IcarusR ,

That is not for the WebSocket port unfortunately, only for the http port.

androme13, 2017-10-23:
As said here : http://dresden-elektronik.github.io/deconz-rest-doc/websocket/
Deconz is started on an unused proxy friendly port, may be if you run your service on port 8080 first and only after, run Deconz, with chance deconz will choose another port ?

manup, 2017-10-23:
Hi, currently there is no command line switch/config available to specify a fixed WebSocket port.
I'll add this in the next versions (2.04.84?).

The command line options should be documented in the main documentation, I hope to find some time for that in the next weeks.

tgroeneveldlg, 2017-10-23:
Thanks @manup, that would be great!

jo-me, 2017-11-03:
It's working fine for me in the latest beta (85).

I adapted the `/etc/systemd/system/deconz.service` and added  `--ws-port=8087` to the parameter list.

tgroeneveldlg, 2017-12-10:
@jo-me > It's working fine for me in the latest beta (85).
You are correct, it works fine now. 
Not sure if it is actually documented. @manup this issue is closed.

closed 2017-12-10 by tgroeneveldlg

StephanStS, 2024-01-08:
Actually the deCONZ command line parameters are described there: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-command-line-parameters