dresden-elektronik/deconz-rest-plugin #240: unable to compile rest plugin

jsve, 2017-10-23:
I get this:

```
make -f Makefile.Release
make[1]: Entering directory '/home/pi/deconz-rest-plugin'
g++ -c -pipe -Wno-attributes -Wall -Wno-attributes -O2 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DARCH_ARM -DARCH_ARMV7 -DGW_SW_VERSION=\"2.04.82\" -DGW_API_VERSION=\"1.0.4\" -DGIT_COMMMIT=\"ec5c87b0949694aa4cdb0cc031b3da075480b8eb\" -DGW_MIN_RPI_FW_VERSION=0x26190500 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"deCONZ-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++ -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4 -I/usr/include -I../.. -I../../common -Irelease -I. -o release/authentification.o authentification.cpp
g++ -c -pipe -Wno-attributes -Wall -Wno-attributes -O2 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DARCH_ARM -DARCH_ARMV7 -DGW_SW_VERSION=\"2.04.82\" -DGW_API_VERSION=\"1.0.4\" -DGIT_COMMMIT=\"ec5c87b0949694aa4cdb0cc031b3da075480b8eb\" -DGW_MIN_RPI_FW_VERSION=0x26190500 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"deCONZ-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++ -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4 -I/usr/include -I../.. -I../../common -Irelease -I. -o release/bindings.o bindings.cpp
authentification.cpp:12:38: fatal error: QMessageAuthenticationCode: No such file or directory
 #include <QMessageAuthenticationCode>
                                      ^
compilation terminated.
Makefile.Release:315: recipe for target 'release/authentification.o' failed
make[1]: *** [release/authentification.o] Error 1
make[1]: *** Waiting for unfinished jobs....
```

ebaauw, 2017-10-23:
That doesn’t look good, qt4 include files.  They moved to qt5.  Are you sure you installed/updated all dependencies?  And ran `qmake`?

jsve, 2017-10-24:
Seems like he default to using qt4. Forcing qt5 with
```
qmake -qt=5 && make -j2
```

seems to work better, but after clearing a few other errors now I'm stuck with this:
```
Project ERROR: Unknown module(s) in QT: serialport
```

I tried this:
```
sudo apt-get install libqt5serialport5
```
Which installed something, but doesn't clear up the error.

oleg77s, 2017-10-24:
It is necessary to compile a rest plugin? 
Is there a difference between the compiled rest plugin and the built-in one?

ebaauw, 2017-10-24:
When you’re happy with a released (beta) version of the plugin, just install deCONZ with the bundled REST API plugin.  When you want the latest commits, or make changes to the plugin yourself, you need to compile and install the plugin.

jsve, 2017-10-24:
I see, so if I'm at deconz version 2.04.82 I'm automatically on the same (latest) rest API?

ebaauw, 2017-10-24:
Yep.

DOliana, 2017-11-03:
I am also not able to compile the plugin. Especially since the description is missing a few points in my oppinion. (I am running Raspbian stretch)

So far I installed the following commands (maybe you can add them to the readme):

- sudo apt-get install qt5-qmake
- sudo apt-get install qt5-default
- sudo apt-get install libqt5websockets5-dev
- sudo apt-get install libqt5serialport5-dev

Now I get the following error when running qmake && make -j2:
```
In file included from /usr/include/c++/6/bits/stl_algo.h:59:0,
                 from /usr/include/c++/6/algorithm:62,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/qglobal.h:94,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/QtGlobal:1,
                 from de_web_plugin_private.h:13,
                 from atmel_wsndemo_sensor.cpp:11:
/usr/include/c++/6/cstdlib:75:25: fatal error: stdlib.h: No such file or directory
 #include_next <stdlib.h>
                         ^
In file included from /usr/include/c++/6/bits/stl_algo.h:59:0,
                 from /usr/include/c++/6/algorithm:62,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/qglobal.h:94,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/qatomic.h:41,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/qrefcount.h:43,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/qbytearray.h:44,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/qcryptographichash.h:44,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/QCryptographicHash:1,
                 from authentification.cpp:11:
/usr/include/c++/6/cstdlib:75:25: fatal error: stdlib.h: No such file or directory
 #include_next <stdlib.h>
```

ebaauw, 2017-11-04:
Workaround in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/216#issuecomment-335021651.

DOliana, 2017-11-05:
I switched back to a backup using Jessie and everything worked fine. I still had to install qt though, which could be added to the README.

closed 2018-09-17 by manup