dresden-elektronik/deconz-rest-plugin #241: deCONZ on ARM64 system?

JeremyHX, 2017-10-24:
Hi,
I'm running my smart home from an arm64 system (the Odroid C2). It's (mostly) pin-compatible with the rpi so electrically I can connect it to my raspbee.
The thing is, it's running an arm64 OS and libraries, and afaik there's no arm64 deCONZ binary available.
Is there a way to run deCONZ and this plugin on an arm64 system (other than reinstalling all the dependencies in 32bit version - some don't like to be installed twice) ?
Thanks,
Jeremy

manup, 2017-10-29:
Currently there is no other way as we just maintain the common Raspbian versions of deCONZ which are currently 32-bit. The demand for non RPi arm64 distros is just too small at the moment to assign resources to maintain a further platform.

To my understanding Raspbian may become 64-bit one day at least for RPi 3, but I'm afraid this will take a while.

jonatanolofsson, 2017-11-24:
Is there any way one can aid in making this happen?

etique57, 2019-03-17:
@udoewich thanks, very interesting.

I would like to use a tinker board S with 16GB eMMC, can you tell me if you managed to use a Raspbee or a Conbee with your setup?

Thanks!

udoewich, 2019-03-17:
I'm using a ConBee, which appears as /dev/ttyUSB0 to the system. `lsusb` should give you something like this:
`Bus 001 Device 003: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)`

It may at first appear to not being recognized.  If that's the case, leave it connected and reboot, then try lsusb again.

One thing to note though: I have it working fine on Armbian 5.60 - they're now at 5.76.  I've yet to get it to work with the newer build: the docker version that installs with 5.76 doesn't seem to be able to run the deCONZ armhf container anymore.  No idea why.

etique57, 2019-03-22:
Thanks, working very fine on the Tinker Board.

I'm trying to figure out now if I can use my RaspBee on the tinker board.

akinomeroglu, 2019-04-20:
> Thanks, working very fine on the Tinker Board.
> 
> I'm trying to figure out now if I can use my RaspBee on the tinker board.

Hello,

Did you manage to run your raspbee on tinker board? Thanks

etique57, 2019-04-20:
Unfortunately not I lack the skill to understand how to achieve this...

poelsevev, 2019-05-21:
i got the raspbee working with tinkerboard s Using eMMC storage. 

instead of using the /dev/ttyACM0 path.
i used /dev/ttyS1 on the TinkerboardS

Allso i got this working with Both normal deconz add on repository new ZHA(deconz) integration
Hope this helps.

manup, 2019-09-18:
The arm64 port is available via APT repository for Debian Stretch distribution.
Packages can also be found at https://www.dresden-elektronik.de/rpi/deconz/alpha/

closed 2019-09-18 by manup

sagittarius06, 2020-01-05:
I have an Odroid-C2 running archlinuxarm with mainline kernel (Linux Odroid-C2 5.4.6-1-ARCH) and the ConBee II works successfully with a deCONZ Docker Image.

See https://hub.docker.com/r/marthoc/deconz

WelterRocks, 2020-09-19:
Still no arm64 support :( while Ubuntu 20.04 LTS is officially available and running on the RPI 3 and 4. Are there any plans to get the conbee stuff working on Ubuntu arm64 for Raspberry in the next few decades? If not, I will drop that conbee sticks and switch to zigbee mqtt gateway, which works out of the box.

manup, 2020-09-20:
Have you tried the arm64 .deb packages mentioned above?

Stable version are now also online located under:

http://deconz.dresden-elektronik.de/debian/stable/

keesj, 2020-10-31:
On Ubuntu focal (arm64) on a Odroid C4 the packages for arm64 are not available. I downloaded the .deb files above and installed them (works so far) it would be nice to support ubuntu focal on arm64!