dresden-elektronik/deconz-rest-plugin #242: Responding to Dresden scene switch on/off

NathanSweet, 2017-10-25:
I have many [Dresden scene switches](https://www.dresden-elektronik.de/funktechnik/solutions/wireless-light-control/wireless-switch/scene-switch/?L=1) and I want to use the websocket to detect button presses. The switch's group is empty, because I don't need to directly control devices with it. I see events for the 4 scene switch buttons, since they set a group scene. I don't see anything for the on and off buttons. I'd like to have an event for on, off, long press, etc. Is this possible? Might it be possible in the future?

manup, 2017-10-29:
The on/off and scene buttons should have worked before.

* on (1002)
* off (2002)

The fix no also fires events for:

* dimm up (1001)
* dimm up stop (1003)
* dimm down (2001)
* dimm down stop (2003)

NathanSweet, 2017-10-29:
I wasn't seeing websocket events at all for any of the button events. I only saw websocket events for scene changes when pressing the scene buttons. To be clear: I added the Dresden scene switch via the web interface, it created an empty group for the switch, I connected to the websocket and printed all events, pressed buttons, didn't see any websocket events except scene changes.

I'm traveling for 2 weeks now but can try again as soon as I'm back.

manup, 2017-10-29:
Strange, maybe version was too old (albeit these events exist for a while now).
> I'm traveling for 2 weeks now but can try again as soon as I'm back.

Ok, let me know if the recent version works then.

NathanSweet, 2017-11-08:
Sorry for the delay.

I updated DeCONZ to 2.04.86. I do indeed see websocket button events. I don't know what was happening earlier. I forgot to note what the older version was. Oh well, I'm happy now. Sorry to bother!

NathanSweet, 2017-11-09:
Maybe there is an issue here after all. Should DeCONZ report all the `buttonevent` values [here](https://developers.meethue.com/documentation/hue-dimmer-switch-programming)? Here is what I see on a Dresden scene switch:
1) Press and quickly release ON: 1002
2) Press and hold ON for a while, then release: nothing for a bit, then 1001, then 1003 when released.
I never see an "initial press" 1000 event.

Also, I noticed if I press the scene buttons on my scene switch I see only a X002 `buttonevent`. I never see X000, X001, or X003.

ebaauw, 2017-11-09:
The Hue dimmer switch is the only switch that supports the full set of buttonevent values: press (x000), hold (x001), release (after press) x002, (release after hold) x003.  It uses ZigBee attribute reporting on a manufacturer specific attribute to report the corresponding buttonevents to the gateway.

Most switches don't report buttonevents, but instead send the corresponding ZigBee command (On, Off, Toggle, DimUp, DimDown, DimStop, SceneRecall, ...) directly to a group of lights. For these, deCONZ listens in to these group commands and reverse engineers the buttonevent.  The buttonevent values supported for these switches really depends on what commands the switch sends in what situation.

Typically, a scene button only issues a single SceneRecall command (either on press or release), which is translated to x002.  A typical wall switch On button combines the On and DimUp functions: it sends On when pressed/released, DimUp when held, and DimStop when released (after hold).  These are translated into x002, x001, and x003.

NathanSweet, 2017-11-09:
Ah, ok. Thanks for the information! It's a little unfortunate, as I prefer events and coding the behavior myself. I can get pretty far with how it works though, so it's not the end of the world. Cheers!

closed 2017-11-09 by NathanSweet