dresden-elektronik/deconz-rest-plugin #243: Stale devices in deCONZ

wvuyk, 2017-10-26:
I have been experimenting regulary with devices in deCONZ for the REST API. But now I have a few devices in deCONZ I think two IKEA's whic are explainable as I removed them long ago and a Philips device for which I do not know where it came from (neighbours?) 

Each time I delete them with the delete key, but each time they are 'discovered' again, and even exist with green lines (but no cluster ifo). How do I get rid of these as they are really powered of and should not be known anymore by deCONZ?

Wim
[image]

ebaauw, 2017-10-26:
Looking at the mac address, the third device is by dresden elektronik, not by Philips.

If the devices are (were) sensors, you might try deleting the corresponding resources using the REST API.

wvuyk, 2017-10-26:
Ok the third device might have been the ConBee then in router mode maybe? 
The IKEA devices are no sensors, both are lights - one GU10 and an E27. I tried to reattach these 2 weeks ago, without luck, could not connect them and could not reset them. 

Wim

ebaauw, 2017-10-26:
Odd.  If they're stale neighbour entries, power cycling all your routers (lights) and the RaspBee/Conbee should get rid of them.

wvuyk, 2017-10-26:
Will try... does take some crawling here LOL

wvuyk, 2017-10-26:
Had to hunt for the last bulb, all reset and now it seems to be solved. Will test leter again by bringing back on tof the bulb and removing again. A delete through the REST API of lights doesn't work?

wvuyk, 2017-10-26:
Not funny, with a restart of the deCONZ, all lights show not connected anymore....

wvuyk, 2017-10-26:
Raised nwk id and rejoined network... works again!

closed 2017-10-26 by wvuyk