dresden-elektronik/deconz-rest-plugin #245: Give Gateway Address to Beta App?

markuspf, 2017-10-28:
Is it possible to give the gateway IP to the beta app in some way? Autodetection does not work since the gateway is in a different subnet than laptop/tablets...

manup, 2017-10-29:
There is no GUI for this yet, but I've added a workaround so that the host, and port if different from 80/ 8080, can be specified in the URL.

Modify the URL as follows with your gateway IP.

    login.html#host/<ip>
    login.html#host/<ip:port>

    login.html#host/<hostname>
    login.html#host/<hostname:port>

For example:

    login.html#host/192.168.2.1
    login.html#host/192.168.2.1:8066

You may need to refresh the page so that the browser cache picks up the latest version.

In the release version a copy of the app is also available on the raspberry pi itself (like the old webapp), then it will be possible to just open `<raspberry-ip>/pwa` which automatically will connect the respective gateway.

markuspf, 2017-10-31:
Thanks this works great for me now. Playing around with the app for a bit then...