dresden-elektronik/deconz-rest-plugin #248: Open Network via REST API

paedae93, 2017-10-29:
Hello all,

I am playing with deCONz and RaspBee. It is really cool and really easy to use. Now I have a more or less interesting question. To connect the Raspberry with a new lamp the network has to be open, which can be done from the WebGUI for example. But is it possible to do that also via the API? 

Thank you very much for helping me out
Patrick

manup, 2017-10-29:
Yes the related parameter is `config/permitjoin`.

http://dresden-elektronik.github.io/deconz-rest-doc/configuration/#modifyconfiguration

Since the webapp also just uses the REST API you can always see what is being transferred in the browser developer tools network tab (F12 in Chrome).

paedae93, 2017-10-29:
Hi Manup,

thank you very much. I was looking already into this. Sorry for the question. 

Have a nice evening 
Patrick

closed 2017-10-29 by paedae93