dresden-elektronik/deconz-rest-plugin #250: Sensors not recreated after deleting and rejoining

djwlindenaar, 2017-11-01:
Hi all,
During an attempt to get the xiaomi temperature/humidity sensor working, I've deleted the sensors through the websocket interface. I was expecting them to be recreated after deleting the sensor in deCONZ gui and then rejoining it by holding the reset butting while the network is in permit join mode.

Although the device shows up in the deCONZ gui, including the temp and humidity clusters, the sensors are still empty in the websocket interface. Is there anything I need to do to get the sensors recreated for the websocket interface?

I do: GET http://192.168.1.31:80/api/747C417170/sensors
I get:
Headers:
    Status Code: 200 OK
    Access-Control-Allow-Origin: *
    Content-Length: 2
    Content-Type: application/json; charset=utf-8
    ETag: "ccb800375a30622067eb3e533ef017da"
Body: 
    {}

Can anyone help me with this?

best regards
Daniel

djwlindenaar, 2017-11-01:
Oh yeah. probably good to know: I'm using version 2.04.84

ebaauw, 2017-11-01:
The REST sensor resource is created only when the node’s Basic cluster has been read (particularly the Manufacturer and Model attributes) and the network is open.  The Xioami sensors don’t like (and sometimes even don’t implement) the read attributes command sent by deCONZ.   You need to press the sensor’s setup button regularly during pairing, to keep them awake and/or to have them report the Basic cluster attributes.

djwlindenaar, 2017-11-04:
Thanks for the tip. That was actually not the issue, but it pointed towards the mechanism I was missing. I opened the network from the GUI instead of through the web app. Apparently this process only works when opening the network through the web app.

It would be nice though to be able to somehow point the rest API to a device to initiate the creation of sensors or others.

djwlindenaar, 2017-11-04:
Thanks for the tip. That was actually not the issue, but it pointed towards the mechanism I was missing. I opened the network from the GUI instead of through the web app. Apparently this process only works when opening the network through the web app.

It would be nice though to be able to somehow point the rest API to a device to initiate the creation of sensors or others

manup, 2017-11-22:
Since the issue popped up multiple times recently, deCONZ 2.04.91 will contain a info in the permit join dialog for that matter.

[image]

closed 2017-11-22 by manup